betaflight/betaflight #12896: GPS Rescue:  Hover throttle needs to be changed to Hover RPM

mactac, 2023-06-18:
### Is your feature request related to a problem? Please describe

GPS rescue relies somewhat on the hover throttle setting in the GPS rescue tab to work properly.   The importance of this is noted in a few places, but there is a fairly large flaw.

GPS rescue is used  most often for longer distance flights, and during these flights the battery level will vary from 4.2v/cell down to as low as 3.5v/cell.   

The voltage of the battery can greatly affect the hover throttle % - especially when using a li-on battery.   The hover throttle when first taking off at 4.2v/cell is radically different than at 3.0v/cell.

This could result in a lost quad in the worst case.

### Describe the solution you'd like

Since throttle % is not a fixed number, the hover should be based on something that does not change - such as motor RPM.  

### Describe alternatives you've considered

Another way to do it would be to have a battery voltage modifier to the throttle %, but that seems more complicated and more prone to incorrect settings.

### Other information

This is a significant problem for long distance flyers.

haslinghuis, 2023-08-13:
@ctzsnooze

ctzsnooze, 2023-08-14:
If we truly wanted to 'hover', we would use the baro and/or acc data and PIDs to target zero vertical velocity, as we do during the fly-home phase - assuming we had altitude information available.

But the true purpose of 'hover' throttle is not to 'hover', but to apply as an initial 'ballpark' value, a starting point for the PIDs when the rescue first starts to avoid sudden drops from going to a low value, or sudden climbs from an excessively high value.

There were suggestions that we could dynamically set the throttle to use at the start of a rescue based on the throttle values that resulted in stable level flight preceding the rescue.  However in practice it isn't so easy, especially with freestyle pilots.

If the user enables  battery voltage sag compensation, and generally I recommend using it, then there won't be a lot of variation from battery voltage changes for a given throttle value.  

The only other situation in which 'hover' throttle matters is when in 'do nothing' phase, in which case we take reduce throttle below the hover value and go into level mode, giving a descent without a total free-fall.  This is only applicable if we lose the GPS signal completely.  It could be done better by using whatever altitude values we have, eg if we have a baro, but at the moment we haven't done that.

ctzsnooze, 2024-04-06:
It's best to choose a hover throttle that hovers with a fairly low battery.  Then in stage 1 or on initiation of a rescue, the tendency will be to not drop, but climb.  That is the best option.

closed 2024-04-06 by ctzsnooze