betaflight/betaflight #12901: BF 4.4.2 external I2C BMP280 barometer not supported, see #12209

roligmue, 2023-06-20:
### Describe the bug

Upgrading Betaflight from 4.4.0 to 4.4.2. Can not get external BMP280 barometer recognized after upgrading to 4.4.2.
Having the same issue again as we had as worked through on this closed issue. https://github.com/betaflight/betaflight/issues/12209 . 
I was able to get it to working in 4.4.0 using these steps given to me from development.

This flight controller does not have a built-in Barometer, it has an external barometer and it was working up to the point of installing 4.4.2.

_We found an issue with custom defines shown but not being included in the build.
The workaround is to enable expert mode. This has been fixed in the configurator nightly builds.
Enable expert mode on firmware flasher tab
Add BARO_BMP280 to custom defines
Flash the board
Configure resources for external BARO device (note this is for TALONF7FUSION board) in CLI:
resource SERIAL_TX 3 NONE
resource SERIAL_RX 3 NONE
resource I2C_SCL 2 B10
resource I2C_SDA 2 B11
set baro_bustype = I2C
set baro_i2c_device = 2
set baro_hardware = AUTO
save_

Getting the following errors from the CLI when entering the set baro_ commands, thus it seems preventing the Barometer from working.
[image]

### To Reproduce

Install Betaflight 4.4.2
Enable expert mode on firmware flasher tab
Add BARO_BMP280 to custom defines
Flash the board
Configure resources for external BARO device (note this is for TALONF7FUSION board) in CLI:
resource SERIAL_TX 3 NONE
resource SERIAL_RX 3 NONE
resource I2C_SCL 2 B10
resource I2C_SDA 2 B11
set baro_bustype = I2C
set baro_i2c_device = 2
set baro_hardware = AUTO
save_

### Expected behavior

Entering these commands from the CLI will enable the external Barometer to again work.

_We found an issue with custom defines shown but not being included in the build.
The workaround is to enable expert mode. This has been fixed in the configurator nightly builds.
Enable expert mode on firmware flasher tab
Add BARO_BMP280 to custom defines
Flash the board
Configure resources for external BARO device (note this is for TALONF7FUSION board) in CLI:
resource SERIAL_TX 3 NONE
resource SERIAL_RX 3 NONE
resource I2C_SCL 2 B10
resource I2C_SDA 2 B11
set baro_bustype = I2C
set baro_i2c_device = 2
set baro_hardware = AUTO
save_

### Support ID

```txt
Id: "ba0fc71b-9103-4202-94e8-a7cf014ff36d"
```

### Flight controller

TALONF7FUSION

### Add any other context about the problem that you think might be relevant here

[image]

haslinghuis, 2023-06-20:
@roligmue update for step 2:

Add `BARO BARO_BMP280` to custom defines.

roligmue, 2023-06-20:
These Updated steps worked, Barometer is discovered and working. No errors from the CLI commands.
Thank you for the quick response and resolution!

========
Install Betaflight 4.4.2
Enable expert mode on firmware flasher tab
Add BARO BARO_BMP280 to custom defines
Flash the board
Configure resources for external BARO device (note this is for TALONF7FUSION board) in CLI:
resource SERIAL_TX 3 NONE
resource SERIAL_RX 3 NONE
resource I2C_SCL 2 B10
resource I2C_SDA 2 B11
set baro_bustype = I2C
set baro_i2c_device = 2
set baro_hardware = AUTO
save

closed 2023-06-20 by roligmue