betaflight/betaflight #12902: BF 4.4.2 GEPRCF411 target unable to connect serial receiver

dxs94, 2023-06-20:
### Describe the bug

When flashing GEPRCF411 target (MPU6000 board from GEPRC P16HD) with BF 4.4.2, enabling serial rx in port tab at correct uart 2, and selecting serial uart / CRSF Telemetry in receiver tab results in no stick inputs shown.
Flashing BF 4.4.0 and same settings, same hardware, results in correct stick movements. However when enabling "telemetry" results in repeating transcient failsafes, eveything is fine without enabling telemetry.
Flashing BF 4.1.7 and same settings, on same hardware everything works fine with no fault.

### To Reproduce

See above.

### Expected behavior

Stick movements should register no matter which BF version is flashed.

### Support ID

```txt
38eebbe0-3f74-4cf4-af8d-23a7b45343d7
```

### Flight controller

 GEPRCF411 target (MPU6000 board from GEPRC P16HD)

### Other components

Crossfire nano RX on uart 2
Caddx Vista on uart 1 (with Goggles2 compatible FW)

### Add any other context about the problem that you think might be relevant here

To narrow down the problem, I just flash BF, apply custom defaults, set ports and receiver tabs. No other modification is done. Flashing itself throws no error.

haslinghuis, 2023-06-20:
Unable to reproduce. Tried to flash release 4.4.2 in core only mode and cloud build. Also verified working with 4.5.0 pre-release.
Using TX firmware 6.19, hardware version 1.03 using CRSF V2, 25mW power and Dynamic Power enabled.

Still using OpenTX 2.3.14 on Taranis X9D+ as I have moved to RM TX16S with ELRS.

Can you verify CRSF version (with TBS Agent Lite)?

dxs94, 2023-06-20:
Hi, thanks for looking into it, really stumped there. Agent lite gives (micro tx v2): fw 6.19, hw 2.03, bl 2.05, rcinput crsfshot.
And regarding nano Rx: fw 6.19, hw 1.33, bl 2.06. Dynamic power. Tx16s, otx 2.3.15.
Btw even when half working on bf 4.4, the DJI HD OSD gets missing elements when too many are present on screen (and that’s not a lot).
This can’t be a hardware problem, this quad was 100% working bf 4.1.7, no msp DisplayPort, right before I flashed 4.4.2 (which never got me a receiver input as explained above).

haslinghuis, 2023-06-21:
> Side observation: saving and reboot from the ports tab, disables « telemetry « option in the receiver tab, even without changing settings in ports tab.

Please use nightly configurator to solve and also check 4.5.0

haslinghuis, 2023-06-21:
@pitts-mo also test 4.5.0 please.

pitts-mo, 2023-06-21:
ok @haslinghuis I will.

Some extra background:
I did just get in three packs on 4.4.0 and did not have a single RX LOSS... but more positively my flight OSD elements appear to populate much more quickly on 4.4.0 and I did not have any instances of failing to re-arm for normal flight after using turtle mode (I think that is a known 4.4.2 d-shot bug).

I also wonder if this users issue may be similar.
https://youtu.be/n8Yp_0IDUfs?t=4082

haslinghuis, 2023-06-21:
As @dxs94 reports we need to disable TELEMETRY in configurator?

4.4.3-zulu contains a fix regarding dshot but there is still some issue slumbering (rpm errors)

pitts-mo, 2023-06-21:
Well, tried with Configurator: 10.10.0-debug-d143d42
*normal build w/acro trainer &reloading diff: no OSD splash or OSD flight elements and can not arm, but my RX shows channel movement in receiver tab. No OSD even after CLI defaults

*even with core only build my results were the same as above :-(

# Betaflight / STM32F411 (S411) 4.5.0 Jun 21 2023 / 20:38:23 (c243e83de) MSP API: 1.46
# config rev: 723e647

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MATEKF411RX
manufacturer_id MTKS

I also think that flying 4.4.0 felt more connected than 4.4.2... Almost as if the FC is not receiving or not processing some RX packets.

pitts-mo, 2023-06-21:
```
amazing what three "```" can do
```

dxs94, 2023-06-22:
> > Side observation: saving and reboot from the ports tab, disables « telemetry « option in the receiver tab, even without changing settings in ports tab.
> 
> Please use nightly configurator to solve and also check 4.5.0

Tried nightly configurator. The self-disabling "telemetry" bug described in the quote is indeed solved. However I flashed 4.5Z (tried standard and core), no way to get the serial receiver data to be recognized (no channel inputs / rssi data etc).
Flashed back 4.4.0 receiver instantly working.

haslinghuis, 2023-06-22:
If using CRSF_V3 can you try`crsf_use_negotiated_baud`

haslinghuis, 2023-06-22:
Can you save (a diff for normal operation and) a dump - and flash the oldest commit from 4.4.2 and restore the dump.

https://github.com/betaflight/betaflight/compare/4.4.1...4.4.2

If that works you could repeat from the list and find the one which breaks the issue for you.
If not we have to look at the commits in 4.4.0...4.4.1

pitts-mo, 2023-06-22:
I reverted to release configurator 10.9.0 . 
selected 4.4.3-Zulu [pre-release]
-commit 12795 does not appear to lag display of my flight OSD elements after the splash screen.
-commit 12811 appears to introduce a random increase in lag initially displaying my flight OSD elements.

@dxs94 
Because my symptoms are not consistent can you test to see if your RX works with 12795?
If so, please test if your RX fails with 12811.

dxs94, 2023-06-25:
In following post working=can have stick inputs in receiver tab as designed. Not working = impossible to have any data from receiver. Full config details in first post.

@haslinghuis 
Did that, 4.4.1 release was working, so I went through choosing some suspects and I found that 12782=not working (whereas 12779=working).

[image]

@pitts-mo See above, no receiver working for me from 12782.

haslinghuis, 2023-06-25:
@SteveCEvans please investigate changes in #12782 as the issue does not reproduce on my build.

haslinghuis, 2023-08-10:
@dxs94 please retry with 4.4.3-zulu.

haslinghuis, 2023-09-05:
Ordered three new TBS crossfire nano's for testing on new builds

Acterion, 2023-09-05:
Just got the same issue with GEPRCF411_AIO firmware. Was using 4.2.11 for a long time, now decided to upgrade all firmware, started from FC. Flashed 4.4.2. And FC couldn't detect any inputs from RX. Reverted to 4.2.11 for now until this is resolved.

haslinghuis, 2023-09-07:
Taranis X9D+ (OpenTx) with CRSF micro TX module works fine on multiple builds (F4, G4, F7) using 4.4 and 4.5 firmware. Tested with crossfire nano and nano pro receiver. OpenTx, EdgeTx, crossfire all on latest firmware.

However when using same CRSF module on Radiomaster TX16 (EdgeTx flying dutchman) there is no stick input.

dxs94, 2023-09-07:
I updated the first post to include info that I also have the problem with TX16S/micro TX V2, using OTX 2.3.15. I don't have another module-bay capable radio. I'm available should you require any test or information.

haslinghuis, 2023-09-19:
Disabling dynamic power in CRSF TX / RX solved stick input on two bui;lds.

haslinghuis, 2023-09-20:
Please build development version `4.4.3-zulu` using https://github.com/betaflight/betaflight/pull/13084 in select commit field for testing.

dxs94, 2023-09-23:
Tested as per instructed 4.4.3-z with commit 13084: working perfectly on the bench. No problem enabling telemetry, no more transcient failsafe, stick inputs are registered. And this is with dynamic power enabled. Pending actual flight testing but the problem was obvious and I'm confident this is the fix.
Bravo!

yodaco, 2023-10-06:
I have the exact same issue of no RX output in Receiver tab on 4.4.1 on a Matek411, working fine on previous versions.

Prajwal-Koirala, 2023-12-26:
Sure, here is a long summary of the article you requested:

The article discusses a bug in Betaflight 4.4.2 that prevents users from connecting a serial receiver and experiencing transient failsafes when telemetry is enabled. This issue is specific to GEPRCF411 targets and CRSF receivers. Several users have reported that downgrading to Betaflight 4.4.0 or 4.4.1 resolves the problem. The developers are investigating the issue and a fix may be available in Betaflight 4.5.0.

Here are some of the key points from the article:

* The bug prevents users from connecting a serial receiver and experiencing transient failsafes when telemetry is enabled.
* The issue is specific to GEPRCF411 targets and CRSF receivers.
* Downgrading to Betaflight 4.4.0 or 4.4.1 may resolve the problem.
* The developers are investigating the issue and a fix may be available in Betaflight 4.5.0.

I hope this summary is helpful! Let me know if you have any other questions.

dxs94, 2025-03-17:
4.5.1 solves the "no stick input" issue.
A switch of UART between the RX and DJI VTX solved the transcient failsafe issue: from my findings it seems the UART which feeds inverted sbus pad causes trouble (a known issue with ELRS too, where receivers are stuck in bootloader mode upon power-up).

closed 2025-03-17 by dxs94