betaflight/betaflight #12908: ICM426XX gyros not listed in check gyroHasOverflowProtection

tbolin, 2023-06-21:
### Describe the bug

Neither of the supported ICM426XXX gyros is listed in the switch case in gyroInitSensor in gyron_init.c. 
Is suspect that both gyros have overflow protection. Currently the firmware assumes that they do not.### 

### To Reproduce

Look in gyroInitSensor and try to find GYRO_ICM42605 and GYRO_ICM42688P.

### Expected behavior

Both gyros should be evaluated for overflow protection and listed  in gyroInitSensor.

### Support ID

```txt
NA
```

### Flight controller

NA

### Other components

NA

### How are the different components wired up (including port information)

NA

### Add any other context about the problem that you think might be relevant here

NA

tbolin, 2023-06-24:
[ICM42688-overflow-test.zip](https://github.com/betaflight/betaflight/files/11857976/ICM42688-overflow-test.zip)
The ICM42688-P  appears to have overflow protection. The gyro value stays at exactly 2000 dps when saturated.

haslinghuis, 2023-06-24:
Maybe add a comment in the code.

commit efe1825a (2023-12-05): Fix #12908 Add ICM-42688-P IMU to list of IMUs with overflow protection (#13013)

Add ICM-42688-P imu to list of imus with overflow protection
closed 2023-12-05 by haslinghuis