betaflight/betaflight #12919: Are servos supposed to be working on BF 4.4?

Davidsastresas, 2023-06-25:
### Describe the bug

I am using a matek H743. In the firmware build I enabled servo support.

Then I have aux channel 4 set to a radio slider, and I confirm it works in the receiver tab, Flight controller is receiving that channel moving.

Then in the servo tab, for the servo 1, I tick the A4 box. I also click the live update tickbox, but the UI doesn't represent the servo moving, nor I see it on the real servo.

In CLI, typing resources reports SERVO 1 being assigned to E5, which is s11 in matekh743 slim:

http://www.mateksys.com/?portfolio=h743-slim#tab-id-7

I wen't over the documentation and all the videos available about the matter several times, but I can not get it to work. Am I missing some step?

Thank you very much! 

### To Reproduce

.

### Expected behavior

.

### Support ID

```txt
.
```

### Flight controller

.

### Other components

.

### How are the different components wired up (including port information)

.

### Add any other context about the problem that you think might be relevant here

.

haslinghuis, 2023-06-25:
Yes servo's should work. Did you check mixer model ?

https://betaflight.com/docs/development/Mixer

Davidsastresas, 2023-06-25:
Thanks @haslinghuis.

I thought the configurator method would be equivalent. I gave a try to the smix CLI approach, setting it as follows:

smix 0 1 11 100 0 0 100 0

Being:
rule 0
servo 1, being assigned to E5, which is s11 in matekh743 slim:
source 11 ( aux channel 5 )
rate 100
speed 0
min 0
max 100
box 0

The use case is a regular quad, only I need to control a pwm output to manage a led switch which works by PWM. Should it work with the above?

Thank you very much.