betaflight/betaflight #12922: COM Port desappears after save and backup and i can´t reconect to FC

ArthurMureb, 2023-06-25:
### Describe the bug

After the settings and the save and backup, betaflight disconnected the FC and it disappeared from the system. Betaflight and my system does not recognize the device. I manage to put it in boot mode, I used impulse rc, I flashed the FC again, but when applying the default settings for the FC, betaflight disconnects to connect again and the cycle repeats itself. I can no longer perform any configuration on my FC. I tried to use previous versions of betaflight but without success. If I don't apply the default configuration, betaflight opens but when saving any configuration and restarting the same error happens

### To Reproduce

Any way I do a setup and reconect the FC will cause the error of lost communication with the computer

### Expected behavior

I expected that when applying a configuration, saving and restarting, betaflight would disconnect and reconnect to proceed with the configurations

### Configurator version

10.9.0

### Flight controller configuration

```txt
defaults nosave

# version
# Betaflight / STM32F7X2 (S7X2) 4.4.2 Jun  2 2023 / 12:22:27 (23d066d08) MSP API: 1.45

# config: YES

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name SPEEDYBEEF7MINIV2
manufacturer_id SPBE
mcu_id 0017001a4232500620343639
signature 

# name: ExplorerLR4

# feature
feature -TELEMETRY
feature -AIRMODE
feature GPS

# serial
serial 1 2 115200 57600 0 115200
serial 2 2048 115200 57600 0 115200

# beacon
beacon RX_LOST
beacon RX_SET

# aux
aux 0 0 0 1875 2100 0 0
aux 1 1 2 1850 2100 0 0
aux 2 13 6 1900 2100 0 0
aux 3 28 2 1300 2100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BAND_A   A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BAND_B   B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BAND_E   E CUSTOM  5705 5685 5665 5665 5885 5905 5905 5905
vtxtable band 4 AIRWAVE  F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 14 26 29 32
vtxtable powerlabels 25 400 800 MAX

# vtx
vtx 0 5 0 0 1 1750 2050
vtx 1 5 0 0 2 1500 1750
vtx 2 5 0 0 3 1250 1500
vtx 3 5 0 0 4 950 1250

# master
set dyn_notch_count = 1
set dyn_notch_q = 500
set acc_calibration = 10,-16,12,1
set rc_smoothing_auto_factor = 35
set rc_smoothing_auto_factor_throttle = 28
set rc_smoothing_setpoint_cutoff = 12
set rc_smoothing_feedforward_cutoff = 12
set rc_smoothing_throttle_cutoff = 20
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set motor_poles = 12
set vbat_max_cell_voltage = 410
set vbat_min_cell_voltage = 270
set vbat_warning_cell_voltage = 300
set yaw_motors_reversed = ON
set gps_sbas_mode = AUTO
set deadband = 2
set yaw_deadband = 2
set report_cell_voltage = ON
set osd_tim1 = 0
set osd_tim2 = 1
set osd_link_quality_pos = 2168
set osd_link_tx_power_pos = 2200
set osd_rssi_dbm_pos = 2136
set osd_flymode_pos = 2433
set osd_vtx_channel_pos = 18497
set osd_craft_name_pos = 2442
set osd_gps_speed_pos = 2177
set osd_gps_sats_pos = 2127
set osd_home_dir_pos = 2125
set osd_home_dist_pos = 2241
set osd_flight_dist_pos = 2273
set osd_altitude_pos = 2209
set osd_warnings_pos = 14569
set osd_avg_cell_voltage_pos = 2369
set osd_disarmed_pos = 2443
set osd_stat_bitmask = 2373420
set osd_canvas_width = 30
set osd_canvas_height = 13
set vtx_band = 5
set vtx_channel = 3
set vtx_power = 2
set vtx_low_power_disarm = UNTIL_FIRST_ARM
set vtx_freq = 5732
set craft_name = ExplorerLR4

profile 0

# profile 0
set feedforward_smooth_factor = 20
set feedforward_jitter_factor = 15
set feedforward_boost = 5

profile 1

profile 2

profile 3

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set roll_rc_rate = 2
set pitch_rc_rate = 2
set yaw_rc_rate = 2
set roll_srate = 50
set pitch_srate = 50
set yaw_srate = 50

rateprofile 1

rateprofile 2

rateprofile 3

# restore original rateprofile selection
rateprofile 0

# save configuration
```

### Add any other context about the problem that you think might be relevant here

This diff was before the bug happened. I was applying settings in the failsafe tab

The FC is a speedybee F7 mini V2
betaflight version used 4.4.2

ArthurMureb, 2023-06-25:
I flashed with full chip erase. I pressed the send support data button, but I can only enter the CLI tab without applying the default settings for my FC. If I apply these default settings, I have to restart the boot process and flash firmware.

version
# Betaflight / STM32F7X2 (S7X2) 4.4.2 Jun  1 2023 / 08:53:11 (23d066d08) MSP API: 1.45

# config: YES
# board: manufacturer_id: SPBE, board_name: SPEEDYBEEF7MINIV2

# status
MCU F722 Clock=216MHz, Vref=3.31V, Core temp=41degC
Stack size: 2048, Stack address: 0x20010000
Configuration: UNCONFIGURED, size: 3630, max available: 16384
Devices detected: SPI:0, I2C:0
Gyros detected: locked

OSD: NONE (33 x 89)
BUILD KEY: 8e75847218a2781bb2fe05bc388f2c96 (4.4.2)
System Uptime: 38 seconds, Current Time: 2023-06-25T22:55:55.273+00:00
CPU:0%, cycle time: 0, GYRO rate: 0, RX rate: 0, System rate: 10
Voltage: 0 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
Arming disable flags: NOGYRO BOOTGRACE CLI MSP

# flash_info
Flash sectors=0, sectorSize=0, pagesPerSector=0, pageSize=0, totalSize=0 JEDEC ID=0x00000000
Partitions:
FlashFS size=0, usedSize=0

# dump hardware

# version
# Betaflight / STM32F7X2 (S7X2) 4.4.2 Jun  1 2023 / 08:53:11 (23d066d08) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name SPEEDYBEEF7MINIV2
manufacturer_id SPBE

# resources
resource BEEPER 1 NONE
resource MOTOR 1 NONE
resource MOTOR 2 NONE
resource MOTOR 3 NONE
resource MOTOR 4 NONE
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource LED_STRIP 1 NONE
resource SERIAL_TX 1 NONE
resource SERIAL_TX 2 NONE
resource SERIAL_TX 3 NONE
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 NONE
resource SERIAL_RX 2 NONE
resource SERIAL_RX 3 NONE
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 NONE
[... 179 lines cut ...]
set gyro_2_spibus = 0
set gyro_2_i2cBus = 0
set gyro_2_i2c_address = 0
set gyro_2_sensor_align = CW0
set gyro_2_align_roll = 0
set gyro_2_align_pitch = 0
set gyro_2_align_yaw = 0
set i2c1_pullup = OFF
set i2c1_clockspeed_khz = 800
set i2c2_pullup = OFF
set i2c2_clockspeed_khz = 800
set i2c3_pullup = OFF
set i2c3_clockspeed_khz = 800
set mco2_on_pc9 = OFF
set scheduler_relax_rx = 25
set scheduler_relax_osd = 25
set box_user_1_name = -
set box_user_2_name = -
set box_user_3_name = -
set box_user_4_name = -

# end the command batch
batch end

#

haslinghuis, 2023-06-25:
Try SPEEDYBEEF7MINI instead. Have both V1 and V2. Only difference is gyro:

[image]

ArthurMureb, 2023-06-25:
The same error happened, the FC disappears from my pc without being able to reconnect

haslinghuis, 2023-06-25:
What happens if detaching GPS.
What RC link (receiver) is used?

ArthurMureb, 2023-06-25:
Disconnect gps? Welds? I'm using a happymodel 2.4 diversity express LRS receiver

haslinghuis, 2023-06-25:
Welds: https://www.bobvila.com/articles/types-of-welds/

Does the GPS not have a plug?

ArthurMureb, 2023-06-26:
I disconnected the gps and the same error

xXREDEEMEDXx, 2023-06-26:
@haslinghuis in my thread you mentioned you had the v1 and v2 upgrades to 4.4.2. Can we get an export of your FC's and use those to complete a local build?

ArthurMureb, 2023-06-26:
> @haslinghuis in my thread you mentioned you had the v1 and v2 upgrades to 4.4.2. Can we get an export of your FC's and use those to complete a local build?

And if you try to install BF 4.3 or earlier with the 10.8 configurator??  I'm on the street, when I get home I'm going to do the test

xXREDEEMEDXx, 2023-06-26:
If I downgrade to the original 4.2.11 I end up with no Gyro.

ArthurMureb, 2023-06-26:
> If I downgrade to 4.3, I get the same result. And the original 4.2.11 I end up with no Gyro.

Are you used the 10.9 configurador? Or  10.8? For me, the 4.2.11 hasn't work

xXREDEEMEDXx, 2023-06-26:
10.9

ArthurMureb, 2023-06-26:
Yes, the 10.8 have the same problema... The FC only work with 4.2.11 version but without gyro

haslinghuis, 2023-06-27:
>  Can we get an export of your FC's and use those to complete a local build?

I'm not a dump factory 😂 

Just do a cloud build and you can produce your own

ArthurMureb, 2023-06-27:
> >  Can we get an export of your FC's and use those to complete a local build?
> 
> I'm not a dump factory 😂 
> 
> Just do a cloud build and you can produce your own

do you think we can test this tool?  Won't that make the situation worse?  I don't know what else to do, I'm desperate hahaha

https://www.st.com/en/development-tools/stm32cubeprog.html

https://youtu.be/SBJQCZKvf68

xXREDEEMEDXx, 2023-06-27:
I know your not a dump factory, lol. You presently have two working FC's with upgraded builds. I would like to compare those to the output from a cloud build. It may give us some info on the issue we're having.

haslinghuis, 2023-06-27:
[BTFL_cli_20230627_133358_SPEEDYBEEF7MINI.txt](https://github.com/betaflight/betaflight/files/11881433/BTFL_cli_20230627_133358_SPEEDYBEEF7MINI.txt)
[BTFL_cli_20230627_133232_SPEEDYBEEF7MINIV2.txt](https://github.com/betaflight/betaflight/files/11881434/BTFL_cli_20230627_133232_SPEEDYBEEF7MINIV2.txt)

ArthurMureb, 2023-06-27:
> [BTFL_cli_20230627_133358_SPEEDYBEEF7MINI.txt](https://github.com/betaflight/betaflight/files/11881433/BTFL_cli_20230627_133358_SPEEDYBEEF7MINI.txt) [...]

I used this tool (https://www.st.com/en/development-tools/stm32cubeprog.html), I managed to force the flash with .HEX from betaflight 4.4.2 but I couldn't apply the default settings. Any idea how I can get a .HEX file with the default settings for this FC?

ArthurMureb, 2023-07-05:
> Thanks for sending this along @haslinghuis. it looks like I need a hex file to be able to load from local. I believe it would need to be a backup of your FC?! Does that make sense?
> 
> Thanks again

https://youtu.be/ZmZoE9bajzY

My friend, I'm traveling and I can't perform this test.  Could you by any chance try to do these steps to see if we resolve our situation?

haslinghuis, 2023-07-05:
Seems you just need to update configurator to latest nightly as I don't have any issues with V1 or V2. V2 is needed when using BMI270 gyro.

https://github.com/betaflight/betaflight-configurator-nightlies/releases

ArthurMureb, 2023-07-07:
> Parece que você só precisa atualizar o configurador para as últimas noites, pois não tenho problemas com V1 ou V2. V2 é necessário ao usar o giroscópio BMI270.
> 
> https://github.com/betaflight/betaflight-configurator-nightlies/releases

I installed the latest version of the debug configurator and when plugging in the FC I was able to access Betaflight, but the same error happened when saving a configuration in the ports tab. I tried to update to all versions of betaflight and all to no avail. Now my FC is not entering DFU mode anymore

ArthurMureb, 2023-07-07:
I managed to get FC to enter DFU mode again. I don't know what else to do, I removed all the accessories (RX, GPS) only the buzzer is connected...

xXREDEEMEDXx, 2023-07-07:
Thanks for the additional info @haslinghuis, I will look into the debug configurator this weekend.

Cheers!

xXREDEEMEDXx, 2023-07-08:
> Seems you just need to update configurator to latest nightly as I don't have any issues with V1 or V2. V2 is needed when using BMI270 gyro.
> 
> https://github.com/betaflight/betaflight-configurator-nightlies/releases

Still no go. I also tried what was suggested in the youtube link about flashing a local file, but that did not work either. apply custom defaults and FC doesn't reboot and no com port. @haslinghuis any chance your willing to back up and share the Hex from your stack?

haslinghuis, 2023-07-08:
- save dump from working 4.5.0 cloud build
- flash [hex](https://github.com/betaflight/betaflight-nightlies/releases/download/v20230705.589/betaflight_4.5.0_STM32F7X2_9186d05.hex)
- restore dump

xXREDEEMEDXx, 2023-07-08:
> * save dump from working 4.5.0 cloud build
> * flash [hex](https://github.com/betaflight/betaflight-nightlies/releases/download/v20230705.589/betaflight_4.5.0_STM32F7X2_9186d05.hex)
> * restore dump

thank you. What do you mean by "restore dump"?

haslinghuis, 2023-07-08:
1. in cli type dump and save to file
2. flash the hex using local
3. in cli use load from file you saved in step 1.

xXREDEEMEDXx, 2023-07-08:
Thank you. Successful update but I have no Gyro. :( I have the V2, so I know it is the BMI270 Gyro

haslinghuis, 2023-07-08:
Cloud build should produce a working build. Recommend using latest nightly configurator.

xXREDEEMEDXx, 2023-07-08:
> Cloud build should produce a working build. Recommend using latest nightly configurator.

Yes, That is what I used.

themeadery, 2023-07-12:
I have found this to be the case with my FC and 4.4.2. I had to go back to 4.4.0. I will spare the usual insane details I provide and just say that you should probably downgrade. Newer than 4.4.0 breaks (incrementally) my gyro/accelerometer and then my USB communications outside of DFU mode. Sad I got booted out of BF on the newest versions, my FC is only a year old I think.

haslinghuis, 2023-07-12:
YMMV. Afaik most if not all ACC/GYRO issues are resolved. Also when having issues try 4.4.3 zulu and 4.5.0 zulu builds.

Always flash with full chip erase and do not rely on backups.

It might also help to use latest [configurator nightly build](https://github.com/betaflight/betaflight-configurator-nightlies/releases)

**Please consider joining our discord - where a lot of volunteers will look into the issue as it's in the details.**

_As a reminder, this issue is about SPEEDYBEEF7MINI and SPEEDYBEEF7MINIV2 and I can't reproduce the issues mentioned in this room._

xXREDEEMEDXx, 2023-07-12:
Thanks @haslinghuis and @themeadery. I have done everything that has been suggested in this string to date. I may end up just replacing my FC. I am from Northern Canada and want to get flying ASAP as my season is short.

Cheers,

themeadery, 2023-07-12:
FWIW: I was forgetting to do full chip erase, so I tried it multiple times on multiple builds and USB/COM port comms was still broken. I didn't try 4.4.3 zulu or newer because I just wanted to get a working setup after diagnosing all day. Might try them out when I get some more motivation, lol.

haslinghuis, 2023-07-12:
Using latest configurator ....

themeadery, 2023-07-12:
@xXREDEEMEDXx I forgot to mention. If you are still broken go back to 4.4.0 cloud build with a full chip erase. That should fix the COM port issue. Windows will then stop saying "Unknown USB Device (Device Descriptor Request Failed)" in the device manager and be happy with the right driver again. What's weird is no matter how hard I poked it with Impulse RC driver fixer or zadig I could not force the correct drive into it. As always, DFU mode still works and lets you go back to a working build/config.

> Using latest configurator .... nightly build

Good advice, however the problem I had was at a lower level than that; at the Windows driver level. Unless the latest nightly configurator somehow includes some new magic driver that gets install on the back end? I can see the FC in device manager, it's just that nothing could connect to it. I'm back on a working 4.4.0 setup now, so I'm gonna sit tight for a while.

closed 2023-08-19 by github-actions[bot]