betaflight/betaflight #12923: Flight controller - no boot after upgrade to any firmware after 4.3

xXREDEEMEDXx, 2023-06-25:
### Describe the bug

Updated my Speedy Bee F7 mini v2 to 4.4. after applying custom defaults controller will not boot. When downgrading I loose gyro 4.2.11

### To Reproduce

Plug in FC
Connect battery
Connect USB 
power up in DFU
set to upgrade to 4.4
when restarted/ connect
"apply Custom Defaults"

flight controller gives a brief dim blue light for BT connect, then 1 green flash then no response

### Expected behavior

Plug in FC
Connect battery
Connect USB 
power up in DFU
set to upgrade to 4.4
when restarted/ connect
"apply Custom Defaults"

flight controller gives a flashing blue light for BT connectivity, then 1 green flash then BF opens to set up FC

### Support ID

```txt
Id: "2b8d39b6-8123-4e3e-83a5-2146b592b7af"

in the above ID, I have not applied Custom Defaults. when I do the FC becomes unresponsive.
```

### Flight controller

Speedy F7 mini v2 stack

### How are the different components wired up (including port information)

all components removed

### Add any other context about the problem that you think might be relevant here

downgraded the FC to the original 4.2.11 and get no Gyro

xXREDEEMEDXx, 2023-06-25:
I am aware this has been discussed. after trying Dev releases 4.3 and 4.5 I get the same result

haslinghuis, 2023-06-25:
Duplicate of #12922

closed 2023-06-25 by haslinghuis

xXREDEEMEDXx, 2023-06-25:
Still no response from my FC. After choosing board SPEEDYBEEF7MINI 4.2.2

haslinghuis, 2023-06-25:
Having two board and they work for me. Let's keep conversation going in 12922

xXREDEEMEDXx, 2023-06-25:
Could I get a copy of your cli dump from both boards so I can make a comparison? I wonder if I can import the cli before applying custom Defaults?!