betaflight/betaflight #12924: BF 4.4.2 - drone fell out of the sky

Nilssonfpv, 2023-06-26:
### Describe the bug

Drone fell out of the sky as if a motor or esc died. I disarmed before hitting the ground, Once i got near the drone it did not arm when i tried.
O3 air unit kept recording until i disconnected the battery, as if it never registered the disarm command?

Bench testing afterwards shows no issues with a esc or motor, except that two motors now need cleaning/possibly new motor bells.
Unfortunately i had osd recording disabled in the goggles, and no blackbox to go by.

If it was a failsafe, it would have had no issues rearming once i got close. And i was only flying ~200m out with crossfire set to 250mw.
If it was a desync that should not have had any effect on arming or O3 recording.

The only thing i can think of is the fc crashing/hanging?

https://youtu.be/XpedM7zS984

### To Reproduce

I have no idea on reproduction steps, and id rather not try to reproduce this either.
Dropping out of the sky at these kinds of speeds isn't exactly something id like doing.

### Expected behavior

Not falling out of the sky.

### Support ID

```txt
Id: "23d445e5-d185-43b3-b0bd-1fc428774358"
```

### Flight controller

Mamba F7 App MK1

### Other components

Mamba F55 ESC
Xing 2506 1850kv
Crossfire Nano
DJI O3

### Add any other context about the problem that you think might be relevant here

https://www.youtube.com/watch?v=XpedM7zS984

jawz101, 2023-06-30:
I broke a new Runcam Thumb Pro last weekend because one of my quads randomly fell  out of the sky.  I don't know if it was due to 4.4.2 necessarily but it was odd

petercabra, 2023-07-01:
The same thing happened to me, I never had this problem and suddenly it disarmed while hovering in place, after motor 1 cutout. I could not arm, so I checked disarm flags in configurator right after failure without powering the FC off, and RPMFILTER disarm flag was there. After power cycling, everything was working again. I later checked wiring from fc to esc, motors and everything was ok... Happened on IFLIGHT_BLITZ_F722 mini stack with DJI O3

keri-s, 2023-07-01:
Same. 

Running STM32F411 on BF 4.4.2 with CRF-N (150Hz fixed) and Vista.

All looks fine then suddenly it indicates RXloss, freezes and goes down.

Never had a issue before.

wwddd39, 2023-07-01:
My 6in drone had the same problem. I'm using GF30F722-ICM with 4.4.2. It just randomly fell from the sky but I can't find anything wrong when testing even after the crash.

FloFPV, 2023-07-10:
It was the same for me, except that no roll or flip is necessary....

MJ666, 2023-07-18:
I have same the same thing now already 2 times after upgrading to 4.4.2. The quad was flying well with 4.4.1 before. It starts spinning as on motor would be failing and than its falling like a stone. I also disarmed when i see the problem. I have ELRS telemetry recordings but can not see anything there: 
[AF_ELRS-2023-07-16.zip](https://github.com/betaflight/betaflight/files/12081823/AF_ELRS-2023-07-16.zip)

Also i can hear some motor twitching during flight. The crash was at the end of the second flight (also end of the telemetry recording). The flash memory for BlackBox recording was already full. So no useful data from there. The FC is an OmnibusF4 target and the ESC is and BLHeli 35A.

nerdCopter, 2023-07-18:
another account (Discord), possibly the same: https://discord.com/channels/868013470023548938/1130576925698572439/1130576925698572439

herokme, 2023-07-18:
Hey! I'm the one having the issues that was just linked on discord in the previous message

So I recently had a perfect working quad on 4.4.2. I swapped out the old FC (Span F722 BT HD [V2)]([url](url)) to a speedybee f405 v3 (also on 4.4.2) to upgrade my fc (wanted a baro and 4v5 pads on it). Ever since the first flight, my quad would just death roll out of nowhere. Sometimes it would happen on the first throttle punch, sometimes i would do 10 rolls and throttle punches and have no issues, and then be gently cruising and just have it death roll from there.

My problem seems to be consistent with the fc rebooting. Whenever the death roll happens, I get "sensor loss" from my radio, and i lose all telemetry readings on my radio from the quad except rssi (which should be from the rx itself, not the fc). Im not able to arm, turtle mode, activate buzzer, etc unless i power cycle the quad. I've tried getting logs of it, and it seems like my fc also stops recording blackbox right when the death roll happens, so I'm not getting logs of it either. My cap should be sufficient, i checked wiring harness between the fc and esc, none of those seem to be a problem.
[](url)
I ended up swapping my old fc back in, and used a whole pack doing los rolls and throttle punches, and it seems to be working perfectly again, i couldnt get it to death roll. I'm thinking at this point, it doesn't seem like it has something to do with BF 4.4.2 and more so with a defective FC, although I'm still having trouble pinpointing what exactly is the problem. Hoping speedybee can warranty this FC

Here's the [video and log file](https://drive.google.com/drive/folders/1OGUI03ZNeVgQniaA8I4rt6_LkJWujWrX?usp=share_link) if anyone wants to take a look (log51 is the one that corresponds to the video).

KarateBrot, 2023-07-18:
What is your ESC firmware and version, guys?

herokme, 2023-07-18:
I'm running BLHeli32, Rev 32.9

MJ666, 2023-07-18:
BLHeli 16.7

haslinghuis, 2023-07-18:
Please test with 4.4.3-zulu as there are some improvements made.
See #12912 for info about new `dshot_telemetry_start_margin` cli parameter

haslinghuis, 2023-07-18:
SPEEDYBEEF405F4 has known [fatal] issues - so we cannot recommend using this target before issue being resolved.

@MJ666 Please test with 4.4.3-zulu as there are some improvements.
See #12912 for info about new `dshot_telemetry_start_margin` cli parameter

EDIT: we need to have some working values for different ESC firmware being used.

herokme, 2023-07-18:
> SPEEDYBEEF405F4 has known [fatal] issues - so we cannot recommend using this target before issue being resolved.

what issues are there with the f405v3? And is it with only a specific BF firmware?

MJ666, 2023-07-19:
Will now take some time to fly again. The quad needs some repair (broken arm) before I can proceed.

ly2707, 2023-07-24:
I have the same problem，
F722 +VISTA+ELRS 3.2 +BES Bluejay 19.0

herokme, 2023-07-24:
> SPEEDYBEEF405V3 has known [fatal] issues - so we cannot recommend using this target before issue being resolved.
> 
> @MJ666 Please test with 4.4.3-zulu as there are some improvements. See #12912 for info about new `dshot_telemetry_start_margin` cli parameter
> 
> EDIT: we need to have some working values for different ESC firmware being used.

does the speedybee F7 have known issues as well? or just the f405v3?

ly2707, 2023-07-24:
> I have the same problem， F722 +VISTA+ELRS 3.2 +BES Bluejay 19.0
board_name HGLRCF722
manufacturer_id HGLR
mcu_id 002500145631500220303937
signature

haslinghuis, 2023-07-24:
@herokme afaik only speedybeef405v3 has an issue with shared SPI used for OSD and SDCard.

MJ666, 2023-07-30:
After rebuilding the copter i had two successful flight today with 4.4.2. During the rebuild i updated the ESC's to BlueJay 0.20 and enabled bidirectional DShot. I have not heard the twitching anymore. More testing may come including going to 4.4.3-zulu. But i have not seen any RPM telemetry errors during my bench tests. Even not with high motor RPM's. Not sure what to tune here? Running 8/4 loop and DShot300.

robertrullyp, 2023-08-07:
I'm experiencing the same issue here with bf 4.4.2 on BetaFPV F722 35A toothpick aio V2. Now the GPS has become useless 😂

SinuXVR, 2023-08-08:
Same with Cinebot30 (F722-45A AIO V2 + Bluejay 0.16 96kHz), after some flight time quad just turns by Yaw and then falls uncontrollably. This started after I installed the GPS module and it happens regardless of the throttle value and flight style,

After some investigation I figured out that on-ground disarmed CPU usage is always around 70%. When I lowered the PID loop from 8 to 4kHz and switched to DSHOT300 the CPU usage dropped to 30%. After that I was able to spend 6 batteries without crashes, but I can still hear motors twitches on sharp turns and quad is very wobbly overall

haslinghuis, 2023-08-08:
We have three different issues:

- serial - should be fixed with 4.4.3-zulu
- dshot telemetry - should be fixed with latest esc firmware and 4.4.3-zulu
- gps - especially with M10 modules we need to lower pid loop and dshot rate - even on F7 MCU

haslinghuis, 2023-08-08:
@herokme Sharing SPI (OSD / Blackbox) is the root cause of the problem with the SPEEDYBEEF405V3.

herokme, 2023-08-08:
> @herokme Sharing SPI (OSD / Blackbox) is the root cause of the problem with the SPEEDYBEEF405V3.

was that the thing with the sd card sharing spi? because i didnt have an sd card in my quad the first few times i flew it with the reboot before i decided to try and get a log

haslinghuis, 2023-08-08:
Yes I think the recommendation is to use an openlager for logging instead of the onboard logging capabilities.

herokme, 2023-08-08:
> Yes I think the recommendation is to use an openlager for logging instead of the onboard logging capabilities.

yes, buti meant, i was getting fc reboots even without an sd card in my fc, so i dont see how my issue is related to sharing spi. any suggestions on that?

nerdCopter, 2023-08-08:
**Discord for support. Github for bug-reports.**
@herokme , `set pid_process_denom = 2` if not already done; disable any features not "needed" (rpm filters, dynamic filters, gps, blackbox) -- i.e. anything to keep CPU from 100% saturation and resource from conflict.  If instead its not CPU, then check power wires from ESC's ribbon-cable. (sometimes copper can be broken inside silicone sheathing and not noticed, yet still make contact until momentum separates them.)

haslinghuis, 2023-08-10:
Please flash using [#13022](https://github.com/betaflight/betaflight/pull/13022) in the select commit field

fabi9163, 2023-08-12:
Same problem for me, fc freezing in middle of flight. Happened 2 times, now 2 more arms are broken and still no clue whats happening...
OSD disappears; O3 still recording; ELRS Telemetry available; Not able to communicate with FC (arm, beeper, nothing) anymore

Target: Diat/MambaF722_2022B(STM32F7X2)
BF: 4.4.2 Stable
O3-Unit
ELRS
(GPS M10)
DSHOT 600

Configurator:
CPU Load: 52%
Run ~20 mins, no freezes until now

Unfortunately no blackbox-log, sorry.

@haslinghuis Do you have any ideas so far why the FC freezing in flight? The related/referenced issues seems to have no impact on running fc? Maybe you could explain it in one sentence whats your theorie?

Thanks a lot and happy flying

haslinghuis, 2023-08-12:
> BF: 4.4.2 Stable

We need to test 4.4.3-zulu and reproduce (with blackbox log) any failure which seems to need the quad being inflight.
While testing keep flying low and avoid concrete. 

In 4.4.2 the UART driver has been changed due to FC's going into DFU randomly. Further changes were made after some issue reports and now are part of 4.4.3-zulu. As with STM32 development we are using different middleware for specific MCU's.

gonebdg, 2023-08-12:
I'm grateful that I saw the problem when still on my desk and yet testing on the field.
I've similar issues with my 5inch quad after upgrade FW to 4.4.2
### **FC Matek F722SE + LS40 SPEDIX 40A Slim F4 3-6 S Blheli_32 M4-32bit**
**_BTFL configurator freeze many times_** during check/setting or saving the changes (first saw when checking OSD elements) and I had to connect/disconnect USB several times before it can normaly connected again. 
Even after downgrading the FW to BTFL 4.4.0 the problem still persist, and I didn't see 4.4.1 as an options anymore.
Thus I decide to use 4.3.2 and everything seems back normal, usb connection went normal and no freezing anymore during check and setting in BTFL configurator.

fabi9163, 2023-08-12:
> We need to test 4.4.3-zulu and reproduce (with blackbox log) any failure which seems to need the quad being inflight. While testing keep flying low and avoid concrete.
>
> In 4.4.2 the UART driver has [...]

Seems legit, thanks.

Tried one more time on configurator-usb-connection over 30 minutes, with success!
->Blue LED which was blinking is now permanently on and no response in Configurator anymore; ELRS transmitter tells "Sensor Lost!"
Does blue-led means BL-Mode?

I will flash to 4.4.3-zulu, try again and update my comment...

haslinghuis, 2023-08-12:
> Blue LED which was blinking is now permanently on

This indicates a hard fault in firmware. Can you provide me with buildkey (after flashing there is a log button) or support id (cli) ?
I want to build you a version for testing.

fabi9163, 2023-08-12:
@haslinghuis CLI-Support-Id: "20a10243-e5b0-4bbb-9037-bae45a2bbdb2"

herokme, 2023-08-12:
> **Discord for support. Github for bug-reports.**
> @herokme , `set pid_process_denom = 2` if not already done; disable any features not "needed" (rpm filters, dynamic filters, gps, blackbox) -- i.e. [...]

my cpu usage in BF is around 38%, so i feel like its not cpu usage. unless something is causing it to spike to 100 as soon as i take off. regardless, i havent tested the broken fc again, im kind of wary of putting that back on a quad

haslinghuis, 2023-08-12:
@fabi9163 

```
make STM32F7X2 EXTRA_FLAGS="-D'BUILD_KEY=62f36631bb01512e2609fac7156cd9bf' -D'RELEASE_NAME=4.4.3-zulu' -D'BOARD_NAME=MAMBAF722_2022B' -D'MANUFACTURER_ID=DIAT' -DCLOUD_BUILD -DUSE_ACC -DUSE_ACC_SPI_ICM42688P -DUSE_ACC_SPI_MPU6000 -DUSE_BARO -DUSE_BARO_BMP280 -DUSE_BARO_DPS310 -DUSE_DASHBOARD -DUSE_DSHOT -DUSE_FLASH -DUSE_FLASH_M25P16 -DUSE_GPS -DUSE_GPS_PLUS_CODES -DUSE_GYRO -DUSE_GYRO_SPI_ICM42688P -DUSE_GYRO_SPI_MPU6000 -DUSE_MAG -DUSE_OSD_HD -DUSE_PINIO -DUSE_SERIALRX -DUSE_SERIALRX_CRSF -DUSE_TELEMETRY -DUSE_TELEMETRY_CRSF"
```
[betaflight_4.4.3_STM32F7X2.zip](https://github.com/betaflight/betaflight/files/12327779/betaflight_4.4.3_STM32F7X2.zip)

fabi9163, 2023-08-12:
@haslinghuis installed your hex-file via "load Firmware (local) without full chip erase.

Powercycled and let it connected to configurator via USB. It took a while (maybe 40 minutes), but now the blue led lights permanent again. Switching tabs in Configurator not possible anymore. (There is no hard time-limit in configurator, right?)

haslinghuis, 2023-08-12:
How is the CPU temp ?
Lately crashed a MSP VTX flashing on a F722 after thermal shutdown.

fabi9163, 2023-08-13:
@haslinghuis Since your hex-file is installed it only freezed the first time I reported yesterday. I wasn't able to reproduce it again until now.

I covered the stack with a towel trying to increase temperature and got it to 64degree without freezing, see:
# status
MCU F722 Clock=216MHz, Vref=3.27V, **Core temp=64degC**
Stack size: 2048, Stack address: 0x20010000
Configuration: CONFIGURED, size: 3017, max available: 16384
Devices detected: SPI:1, I2C:2
Gyros detected: gyro 1 locked dma
GYRO=ICM42688P, ACC=ICM42688P, BARO=DPS310, MAG=QMC5883
OSD: MSP (53 x 20)
BUILD KEY: 62f36631bb01512e2609fac7156cd9bf (4.4.3-zulu)
System Uptime: 3058 seconds, Current Time: 2023-08-13T09:00:20.370+00:00
CPU:49%, cycle time: 123, GYRO rate: 8130, RX rate: 15, System rate: 9
Voltage: 2 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 5
FLASH: JEDEC ID=0x00ef4018 16M
Arming disable flags: RXLOSS THROTTLE NOPREARM CLI MSP RPMFILTER

Will try to reproduce again. Maybe fly 1-2 packs today with logging enabled

haslinghuis, 2023-08-13:
Use SCHEDULER_DETERMINISM for logging.

fabi9163, 2023-08-13:
@haslinghuis Today I flew ~8 packs cleaning blackbox 2 times per flight hoping to reproduct the failure with the hex-file you gave me (8 Packs just slow cruising waiting for freeze can be veeery long, hehe).
Result: No failure until now, sorry.

Last log appended (Can't open it in blackbox-explorer, maybe corrupted, or due to the SCHEDULER_DETERMINISM setting something special)

[4.4.3-zulu_manualHexWithSCHEDULER_DETERMINISM.zip](https://github.com/betaflight/betaflight/files/12330114/4.4.3-zulu_manualHexWithSCHEDULER_DETERMINISM.zip)

haslinghuis, 2023-08-13:
[image]

Too bad blackbox log is not working (tried with latest master of the viewer) - `od -x ` seems to miss EOF

fabi9163, 2023-08-13:
…Hope I will make another short flight tomorrow

haslinghuis, 2023-08-13:
Others having the exact same issue should provide a support ID so I can build a version for testing.

ctzsnooze, 2023-08-14:
If using GPS Rescue, I recommend to not use a PID loop speed above 4k, ie no higher than 8k4k.

herokme, 2023-08-14:
> If using GPS Rescue, I recommend to not use a PID loop speed above 4k, ie no higher than 8k4k.

this is even with a f7 board?

fabi9163, 2023-08-14:
> @fabi9163 maybe this fixes blackbox drivers:
> 
> [betaflight_4.4.3_STM32F7X2_W25Q128FV.zip](https://github.com/betaflight/betaflight/files/12330231/betaflight_4.4.3_STM32F7X2_W25Q128FV.zip)

Flew another pack with the "original" hex file you gave me [here](https://github.com/betaflight/betaflight/issues/12924#issuecomment-1675984809)

Now it's readable, maybe you can use something of it:
[4.4.3-zulu_manualHexWithSCHEDULER_DETERMINISM_secondTry.zip](https://github.com/betaflight/betaflight/files/12337147/4.4.3-zulu_manualHexWithSCHEDULER_DETERMINISM_secondTry.zip)

Did you do anything besides compile it with another sdk? Is 4.5 also compiled with this change?

haslinghuis, 2023-08-14:
No only different ARM SDK.

[betaflight_4.5.0_STM32F7X2_MAMBAF722_2022B.zip](https://github.com/betaflight/betaflight/files/12337489/betaflight_4.5.0_STM32F7X2_MAMBAF722_2022B.zip)

haslinghuis, 2023-08-17:
@fabi9163 did you test the build for 4.5?

We need to be sure using ARM SDK 9.3.1 fixes the freezing issue.

fabi9163, 2023-08-17:
@haslinghuis I've flashed 4.5-zulu cloud-build (https://build.betaflight.com/api/builds/1f38dc0567242478c76cf31b6e1b3d69/log) to test, if the Bug in principle exists in 4.5 (cause I can't find any message showing this)

Sadly I can't fly these days, so the quad sitting on desk connected to configurator and waiting to freeze->running 3 hours without problems until now.

Will flash your hex-version in 1-2 hours and try the same. But still not sure, if configurator and armed situation are the same

haslinghuis, 2023-08-17:
Yes afraid this requires careful A - B flight tests to try to reveal the error.

fabi9163, 2023-08-17:
I think I will go back to 4.4.2 and reproduce the failure first.
How many times is it possible to reflash the f7-chip before it gets wasted? 50/100/1000? Don't want to destroy the flash with to much reflashing.
Any ideas how to get to a conclusion

@haslinghuis Were there any matching release-note-entries in the new arm-version which seems legit or was it just a try?

fabi9163, 2023-08-17:
Legit, but why does the problem not happened in 4.5 which is build on SDK 10.3.1, right?
Maybe only luck until now...

realThirdpartycookie, 2023-08-19:
Yep my freshly new built drone fell out of the sky and hit a windmill :/ 
Is Betaflight 4.3 safe?

fabi9163, 2023-08-19:
> Yep my freshly new built drone fell out of the sky and hit a windmill :/
> Is Betaflight 4.3 safe?

Could you list some infos how you flashed the firmware (with or without erase flash) from which version to which and which fc you have? Does the fc freeze if you let it connected to configurator for some hours?

nerdCopter, 2023-08-31:
@haslinghuis, is the SDK v9 reversion merged to 4.4.3-zulu?  and 4.5 is not affected?
i.e. both branches "fixed"?  jsut asking for clarification so as to answer pilots in discord.

haslinghuis, 2023-08-31:
Cloud build needs modification to use SDK v9. Until we decide we should downgrade builds are provided on request for testing.

realThirdpartycookie, 2023-09-13:
> Could you list some infos how you flashed the firmware (with or without erase flash) from which version to which and which fc you have? Does the fc freeze if you let it connected to configurator for [...]

I can only offer the footage. The Quad itself isn't in flying condition right now

Antislag, 2023-09-16:
Hi, I have the same issue with Cinebot30. FC: GEP-F722-45A AIO V2

Can I generate some date to support solving this issue?

[10 more replies]

closed 2023-12-05 by haslinghuis