betaflight/betaflight #12935: BF 4.4.2 T-Motor F7 35A AIO no barometer and flash chip showing up

wyldrlorenz, 2023-07-01:
### Describe the bug

Flashing from 4.3.2 to 4.4.0 or 4.4.2 no barometer or flash chip are showing. On 4.3.2, which the board came originally everything works. Barometer is a BMI280
https://store.tmotor.com/goods-1293-F7+35A+AIO.html

### To Reproduce

Flashing to 4.3.2 - barometer and flash chip are there. Flashed again with 4.4.2, even added BARO BARO_BMP280 to custom defines  - no barometer is showing. 

Here is a cli-dump:
# 

# dump

# version
# Betaflight / STM32F7X2 (S7X2) 4.4.2 Jun  3 2023 / 20:45:40 (23d066d08) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name TMOTORF7_AIO
manufacturer_id TMTR

# name: -

# resources
resource BEEPER 1 C13
resource MOTOR 1 C06
resource MOTOR 2 C07
resource MOTOR 3 B00
resource MOTOR 4 B01
resource MOTOR 5 B06
resource MOTOR 6 B08
resource MOTOR 7 B07
resource MOTOR 8 C09
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 C10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 C12
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 C11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 D02
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource SERIAL_RX 12 NONE
resource I2C_SCL 1 NONE
resource I2C_SCL 2 B10
resource I2C_SCL 3 NONE
resource I2C_SCL 4 NONE
resource I2C_SDA 1 NONE
resource I2C_SDA 2 B11
resource I2C_SDA 3 NONE
resource I2C_SDA 4 NONE
[... 968 lines cut ...]

Barometer and flash chip should be there

### Support ID

```txt
Id: "01096714-d602-40f8-9e37-ba5e80207662"
```

### Flight controller

T-Motor F7 35A AIO

### Other components

ELRS RX on UART3 but not configured
Bluejay ESC Firmware

### How are the different components wired up (including port information)

ELRS RX on UART3 but not configured

### Add any other context about the problem that you think might be relevant here

[image]

wyldrlorenz, 2023-07-02:
Thanks for the fast work! Will the change to the BF code be available already within the cloud build or how can I test it?

haslinghuis, 2023-07-02:
Until merged add `BARO BARO_SPI_BMP280 FLASH_W25Q128FV` to custom defines. (first enable show release candidates and enable expert mode).

closed 2023-07-02 by haslinghuis

wyldrlorenz, 2023-07-02:
Just tried with the custom defines - works! Thanks!