betaflight/betaflight #12938: skystars F7HD mini pro and Pro3 GPS not working and it will not go to Bootloader mode.

Ls4life, 2023-07-04:
### Describe the bug

I'm having the same issue as many other people already described regarding GPS not working on Betaflight 4.3 and 4.4.  It was working before upgrading but now regardless the UART I get not satellite lock. Also when trying to upgrade either FC, I have to change the FC to boot loader mode manually. I checked and watch as many youtube video to make sure the issue is not a physical connection, I replace the GPS at least twice. I did all that event thought, the GPS was working on the previous version of Betaflight (4.2).

### To Reproduce

Take a Skystars F7HD mini pro or pro3 and configure GPS as it is done with any quad.

### Expected behavior

After powering on the goggle and radio transmitter, see telemetry information on the OSD and radio master boxer or Betafight GPS tap.

### Support ID

```txt
Unique device ID: 0x27000a5746501620303431
```

### Flight controller

Skystars F7hd mini pro and pro3

haslinghuis, 2023-07-04:
Please try 4.5.0 pre-release while adding `#12799` in commit field. (Need to enable show release candidates and enable expert mode and select development)

Ls4life, 2023-07-05:
Hello!
@haslinghuis, I did as you indicated. Just to be on the safe side, I got a message that I was using an old version of betaflight configurator, but I downloaded the latest one for a mac. Do I need to download the betaflight that has the .dev at the end. The developer version?

haslinghuis, 2023-07-05:
https://github.com/betaflight/betaflight-configurator-nightlies/releases

OSX uses the dmg file.

[image]

Ls4life, 2023-07-05:
Great news. It worked. I did not fly the quad but had it tested and I was able to get satellite lock and received telemetry on the radio master controller display as I used to get previously. Thanks so much. Betaflight 10.10 looks good. Thanks again.

closed 2023-07-05 by Ls4life

Ls4life, 2023-07-24:
Mark,

The fixed you provided worked for me the first time I tested it, but it
stopped working the few other times I used the quad. It has not work again.
Nothing changed with the quad from the time I replied to you that the issue
was corrected.

haslinghuis, 2023-07-24:
Please retry and use 57K6 for GPS baudrate in ports tab.

Ls4life, 2023-07-25:
Thanks for your reply. I will try and let you know. Thanks & Regards.

haslinghuis, 2023-07-25:
With latest changes 115K2 seems to work also but be aware this is still in development phase.

Ls4life, 2023-07-27:
I had it set to 57k, but was not working after the very first time. I will
try the new values and let you know.

Ls4life, 2023-07-29:
I try the new setting on the port tap and it still do not pickup any
satellites. It did the first time I flashed the betaflight 4.5 and added
the especial code, but it has not worked since again. Last time they work
without issue was on betaflight 4.2 and maybe 4.3.

haslinghuis, 2023-08-08:
@Ls4life `#12799` is still in development. Latest edition seems to work better. Confirm need to use boot button for DFU with GPS attached on different FC's tested. You need to set pidloop frequency to 4K in configuration tab.

Ls4life, 2023-08-09:
My flight controller, "skystarsfhdpro" it seem it do not support 4k pid
loop. On the betaflight system configuration it only goes up to 3.20khz.
What it is strange, it is that it did work one time after I had moved to
betaflight 4.5, but it stopped working. It worked after I used your
instructions. I am using an HDzero system, do you think this system is
causing some kind of interference. Yet GPS did work on betaflight 4.2.

Ls4life, 2023-08-09:
Mark, I flashed the flight controller again and went through all the steps
again and it is working. I am not using 4k Pid loop. Is not available. At
the port tap I have it setup to the default value and at the GPS tap, I set
everything to auto. Not sure if the issue will come up again but look good
so far. I tried several times and it is still not working. Thanks for your
help. I let you know if something changes.