betaflight/betaflight #12939: KakuteF7HDV Resource Conflict

jihlein, 2023-07-04:
### Describe the bug

With the KakuteF7HDV, selecting the tricopter mixer and the redefining the motor 5 resource as none and the servo 1 resource as C09 leads to the following error msg:

[image]

Bug seems to have been introduced in 4.3.0, the above steps work fine in 4.2.11.

### To Reproduce

Follow steps above with any firmware from 4.3.0 and above.  I've personally tried 4.2.11, 4.3.0, and 4.4.2.

### Expected behavior

Since I see no evidence of resource (pins, dma, timers) conflicts, the listed steps should produce a functional mixer.

### Support ID

```txt
# dump hardware

# version
# Betaflight / STM32F745 (S745) 4.4.2 Jul  4 2023 / 23:02:42 (23d066d08) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name KAKUTEF7HDV
manufacturer_id HBRO

# resources
resource BEEPER 1 D15
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 E09
resource MOTOR 4 E11
resource MOTOR 5 NONE
resource MOTOR 6 A03
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 C09
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource LED_STRIP 1 D12
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 D05
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 D06
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 E07
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource SERIAL_RX 12 NONE
resource I2C_SCL 1 B06
resource I2C_SCL 2 NONE
resource I2C_SCL 3 NONE
resource I2C_SCL 4 NONE
resource I2C_SDA 1 B07
resource I2C_SDA 2 NONE
resource I2C_SDA 3 NONE
resource I2C_SDA 4 NONE
resource LED 1 A02
resource LED 2 NONE
resource LED 3 NONE
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 NONE
resource SPI_SCK 4 E02
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 NONE
resource SPI_MISO 4 E05
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 NONE
resource SPI_MOSI 4 E06
resource ADC_BATT 1 C03
resource ADC_RSSI 1 C05
resource ADC_CURR 1 C02
resource ADC_EXT 1 NONE
resource BARO_CS 1 NONE
resource BARO_EOC 1 NONE
resource BARO_XCLR 1 NONE
resource SDCARD_CS 1 A04
resource SDCARD_DETECT 1 D08
resource PINIO 1 NONE
resource PINIO 2 NONE
resource PINIO 3 NONE
resource PINIO 4 NONE
resource USB_MSC_PIN 1 NONE
resource GYRO_EXTI 1 E01
resource GYRO_EXTI 2 NONE
resource GYRO_CS 1 E04
resource GYRO_CS 2 NONE
resource USB_DETECT 1 A08
resource PULLUP 1 NONE
resource PULLUP 2 NONE
resource PULLUP 3 NONE
resource PULLUP 4 NONE
resource PULLDOWN 1 NONE
resource PULLDOWN 2 NONE
resource PULLDOWN 3 NONE
resource PULLDOWN 4 NONE

# timer
timer E13 AF1
# pin E13: TIM1 CH3 (AF1)
timer B00 AF2
# pin B00: TIM3 CH3 (AF2)
timer B01 AF2
# pin B01: TIM3 CH4 (AF2)
timer E09 AF1
# pin E09: TIM1 CH1 (AF1)
timer E11 AF1
# pin E11: TIM1 CH2 (AF1)
timer C09 AF3
# pin C09: TIM8 CH4 (AF3)
timer A03 AF2
# pin A03: TIM5 CH4 (AF2)
timer D12 AF2
# pin D12: TIM4 CH1 (AF2)

# dma
dma SPI_MOSI 1 1
# SPI_MOSI 1: DMA2 Stream 3 Channel 3
dma SPI_MOSI 2 NONE
dma SPI_MOSI 3 NONE
dma SPI_MOSI 4 NONE
dma SPI_MISO 1 NONE
dma SPI_MISO 2 NONE
dma SPI_MISO 3 NONE
dma SPI_MISO 4 NONE
dma SPI_TX 1 1
# SPI_TX 1: DMA2 Stream 3 Channel 3
dma SPI_TX 2 NONE
dma SPI_TX 3 NONE
dma SPI_TX 4 NONE
dma SPI_RX 1 NONE
dma SPI_RX 2 NONE
dma SPI_RX 3 NONE
dma SPI_RX 4 NONE
dma ADC 1 1
# ADC 1: DMA2 Stream 4 Channel 0
dma ADC 2 NONE
dma ADC 3 NONE
dma UART_TX 1 NONE
dma UART_TX 2 NONE
dma UART_TX 3 NONE
dma UART_TX 4 NONE
dma UART_TX 5 NONE
dma UART_TX 6 NONE
dma UART_TX 7 NONE
dma UART_TX 8 NONE
dma UART_RX 1 NONE
dma UART_RX 2 NONE
dma UART_RX 3 NONE
dma UART_RX 4 NONE
dma UART_RX 5 NONE
dma UART_RX 6 NONE
dma UART_RX 7 NONE
dma UART_RX 8 NONE
dma pin E13 1
# pin E13: DMA2 Stream 6 Channel 6
dma pin B00 0
# pin B00: DMA1 Stream 7 Channel 5
dma pin B01 0
# pin B01: DMA1 Stream 2 Channel 5
dma pin E09 2
# pin E09: DMA2 Stream 3 Channel 6
dma pin E11 1
# pin E11: DMA2 Stream 2 Channel 6
dma pin C09 0
# pin C09: DMA2 Stream 7 Channel 7
dma pin A03 0
# pin A03: DMA1 Stream 1 Channel 6
dma pin D12 0
# pin D12: DMA1 Stream 0 Channel 2

# feature
feature -RX_PPM
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -RANGEFINDER
feature -TELEMETRY
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -OSD
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -RX_SPI
feature -ESC_SENSOR
feature -ANTI_GRAVITY
feature RX_SERIAL
feature TELEMETRY
feature OSD
feature AIRMODE
feature ANTI_GRAVITY

# serial
serial 20 1 115200 57600 0 115200
serial 0 1 115200 57600 0 115200
serial 1 0 115200 57600 0 115200
serial 2 0 115200 57600 0 115200
serial 3 0 115200 57600 0 115200
serial 5 64 115200 57600 0 115200
serial 6 0 115200 57600 0 115200

# master
set gyro_to_use = FIRST
set baro_bustype = I2C
set baro_spi_device = 0
set baro_i2c_device = 1
set baro_i2c_address = 0
set adc_device = 1
set blackbox_device = SDCARD
set dshot_burst = OFF
set dshot_bitbang = AUTO
set dshot_bitbang_timer = AUTO
set current_meter = ADC
set battery_meter = ADC
set vbat_scale = 110
set ibata_scale = 275
set beeper_inversion = ON
set beeper_od = OFF
set beeper_frequency = 0
set sdcard_detect_inverted = ON
set sdcard_mode = SPI
set sdcard_spi_bus = 1
set sdio_clk_bypass = OFF
set sdio_use_cache = OFF
set sdio_use_4bit_width = OFF
set led_inversion = 0
set pinio_config = 1,1,1,1
set pinio_box = 255,255,255,255
set usb_msc_pin_pullup = ON
set gyro_1_bustype = SPI
set gyro_1_spibus = 4
set gyro_1_i2cBus = 0
set gyro_1_i2c_address = 0
set gyro_1_sensor_align = CW270
set gyro_1_align_roll = 0
set gyro_1_align_pitch = 0
set gyro_1_align_yaw = 2700
set gyro_2_bustype = SPI
set gyro_2_spibus = 4
set gyro_2_i2cBus = 0
set gyro_2_i2c_address = 0
set gyro_2_sensor_align = CW0
set gyro_2_align_roll = 0
set gyro_2_align_pitch = 0
set gyro_2_align_yaw = 0
set i2c1_pullup = OFF
set i2c1_clockspeed_khz = 800
set i2c2_pullup = OFF
set i2c2_clockspeed_khz = 800
set i2c3_pullup = OFF
set i2c3_clockspeed_khz = 800
set i2c4_pullup = OFF
set i2c4_clockspeed_khz = 800
set mco2_on_pc9 = OFF
set scheduler_relax_rx = 25
set scheduler_relax_osd = 25
set box_user_1_name = -
set box_user_2_name = -
set box_user_3_name = -
set box_user_4_name = -

# end the command batch
batch end

*** Support data submitted *** 
Id: "5592643e-49d2-4939-b54f-bd6cce66039c"
```

### Flight controller

KakuteF7HDV

### Other components

Serial GPS, I2C mag, fport RX.  Not sure this matters as the copter hardware config works in 4.2.11

### How are the different components wired up (including port information)

N/A

haslinghuis, 2023-07-05:
Only have the KakuteH7[MINI]

[image]

Can we compare output of `timer show` and `dma show` before and after mapping resource [motor5 -> servo1]?

jihlein, 2023-07-05:
Sure, here's the results on 4.4.2.  Will do the same on 4.2.11 for another comparison.  Don't understand why there is bit bang dshot when dma channels are defined, bo no expert on the inner workings of resource management here.

[noServoDefined_4p4p2.txt](https://github.com/betaflight/betaflight/files/11961308/noServoDefined_4p4p2.txt)
[servo1Defined_4p4p2.txt](https://github.com/betaflight/betaflight/files/11961310/servo1Defined_4p4p2.txt)

haslinghuis, 2023-07-05:
Can you try with bitbang on and off.

haslinghuis, 2023-07-05:
Right. Seems 4.2.11 defaults to bitbang disabled as > 4.4 defaults to bitbang enabled. Try disabling dshot_bitbang.

jihlein, 2023-07-05:
Okay, dshot bitbang was set to AUTO.  Setting it to OFF solves the problem.  With it set to off, able to set motor 5 to none and servo 1 to C09, and have a working configuration.  Screen grabs look identical to 4.2.11.

Thanks for looking into this!

-John

closed 2023-07-05 by haslinghuis

haslinghuis, 2023-07-06:
FYI: https://github.com/betaflight/betaflight/pull/11278