betaflight/betaflight #12940: SITL - Can't set the PID loop frequency changing the rates from actual to betaflight

Lanzo98, 2023-07-05:
### Describe the bug

On the SITL version of betaflight, when I change the rates from actual to betaflight in the rates tab, then I cannot change the PID loop frequency on the Configuration tab, and it displays it as empty
[image]

### To Reproduce

Change the the rates from actual to betaflight in the rates tab

### Expected behavior

I should be able to change the PID loop frequency

### Support ID

```txt
1d12cb1b-922c-4d91-89bd-857674aa5799
```

### Flight controller

SITL

haslinghuis, 2023-07-05:
Same happens on a [normal] build. Pid loop frequency shows nothing. Selecting frequency and saving does not sustain.

ctzsnooze, 2023-12-18:
With nightly configurator, I get `pid_process_denom = 7`, and I cannot save to 1, and consequently a very strange PID rate.
[image]

ctzsnooze, 2023-12-18:
I found that if I set `use_unsynced_pwm = ON`, then the PID loop control returns to normal.

Could someone please verify that PWM controlled motors work normally in 4.5 with  `use_unsynced_pwm = ON` ?

Thanks

haslinghuis, 2023-12-18:
[image]

haslinghuis, 2023-12-24:
Fixed in #13236

closed 2023-12-24 by haslinghuis

demvlad, 2024-11-17:
> Could someone please verify that PWM controlled motors work normally in 4.5 with `use_unsynced_pwm = ON` ?
> 
> Thanks

I have same sitl issue today.
Thank's. Will try

demvlad, 2024-11-18:
@haslinghuis Please, could you help.
I have this issue in SITL 4.6 with PWM motors now.
The pid_process_denom = 16, i can not change it. The pid loop time in configurator status line is changed from 2000 as minimum to maximum up to 30000. 
The use_unsynced_pwm settings does not help.

haslinghuis, 2024-11-18:
Works on a real FC:
```bash
# master
set motor_pwm_protocol = PWM
set pid_process_denom = 2

# get use_unsynced_pwm
use_unsynced_pwm = ON
Allowed values: OFF, ON
```

demvlad, 2024-11-18:
> Works on a real FC:

@haslinghuis Thank you!
The pid frequency shows as 4kHz at the configuration page (the clear field was before).
The pid loop time at the status line is around 250