betaflight/betaflight #12943: GPS M10 icon not showing up depending on board and power

Pancronos, 2023-07-05:
### Describe the bug

BFC version 10.10 (nightly) workbench testing 4.5 zulu on a cinelog 30 with a flywoo M10 micro gps, after a warm up (30 seconds on ) it gets satellite and 3d fix. Pretty good. 
[image]
The cli shows no gps ? Speed not set ?
[image]
the port is set as reccomended at 115200
[image]
After few minutes (nearly 10) BF stops responding, disconnect doesn't work, Please note that on other builds with M8 gps it all works fine.
Changing tab just crash BF
[image]
Disconnect / reconnect ends to same behaviour, only solution is to physically remove usb from pc port.
[image]

### To Reproduce

Set an fc with above setup. and follow same steps.

### Expected behavior

BF configurator  after changing gps setting should not  go crazy and block USB.

### Configurator version

2023-06-09 @22:13:01 -- OS: Windows  2023-06-09 @22:13:01 -- Configurator: 10.10.0-debug-84a0d6b

### Flight controller configuration

```txt
Entering CLI Mode, type 'exit' to return, or 'help'

# 
# Building AutoComplete Cache ... Done!
# 
# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.5.0 Jun  9 2023 / 15:16:04 (7d7b6596b) MSP API: 1.46
# config rev: 547d467

# start the command batch
batch start

board_name GEPRC_F722_AIO
manufacturer_id GEPR

# name: Cinebot30

# resources
resource MOTOR 1 C07
resource MOTOR 2 C06
resource MOTOR 3 C09
resource MOTOR 4 C08
resource PINIO 2 A01

# feature
feature -AIRMODE
feature GPS
feature TELEMETRY
feature OSD

# serial
serial 4 2 115200 115200 0 115200

# beacon
beacon RX_LOST
beacon RX_SET

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 900 1700 0 0
aux 2 2 1 1700 2100 0 0
aux 3 13 2 1700 2100 0 0
aux 4 28 1 1700 2100 0 0
aux 5 35 3 1700 2100 0 0

# master
set gyro_lpf1_static_hz = 0
set dyn_notch_count = 1
set dyn_notch_q = 600
set dyn_notch_min_hz = 150
set gyro_lpf1_dyn_min_hz = 0
set acc_lpf_hz = 10
set acc_trim_pitch = -1
set acc_trim_roll = -1
set acc_calibration = -69,-45,-20,1
set rc_smoothing_auto_factor = 250
set rc_smoothing_auto_factor_throttle = 250
set rc_smoothing_setpoint_cutoff = 6
set rc_smoothing_feedforward_cutoff = 6
set rc_smoothing_throttle_cutoff = 20
set dshot_bidir = ON
set motor_output_reordering = 2,0,3,1,4,5,6,7
set align_board_roll = 180
set align_board_yaw = 0
set vbat_max_cell_voltage = 435
set beeper_dshot_beacon_tone = 3
set yaw_motors_reversed = ON
set small_angle = 180
set gps_ublox_use_galileo = ON
set gps_rescue_min_start_dist = 30
set gps_rescue_ascend_rate = 500
set gps_rescue_ground_speed = 2000
set gps_rescue_descent_dist = 200
set gps_rescue_landing_alt = 5
set gps_rescue_throttle_max = 1600
set gps_rescue_throttle_hover = 1280
set gps_rescue_sanity_checks = RESCUE_SANITY_ON
set gps_rescue_min_sats = 5
set gps_rescue_allow_arming_without_fix = ON
set gps_rescue_throttle_p = 150
set gps_rescue_throttle_i = 20
set gps_rescue_throttle_d = 50
set gps_rescue_velocity_p = 80
set gps_rescue_velocity_i = 20
set gps_rescue_velocity_d = 15
set gps_rescue_yaw_p = 40
set osd_vbat_pos = 2658
set osd_rssi_pos = 3126
set osd_rsnr_pos = 3158
set osd_tim_1_pos = 3663
set osd_tim_2_pos = 3631
set osd_flymode_pos = 2110
set osd_vtx_channel_pos = 404
set osd_crosshairs_pos = 2462
set osd_ah_pos = 2335
set osd_current_pos = 259
set osd_craft_name_pos = 2076
set osd_gps_sats_pos = 2081
set osd_warnings_pos = 2681
set osd_avg_cell_voltage_pos = 2626
set osd_battery_usage_pos = 2744
set osd_displayport_device = MSP
set osd_canvas_width = 60
set osd_canvas_height = 22
set debug_mode = GYRO_SCALED
set vcd_video_system = HD
set pinio_config = 129,1,1,1
set pinio_box = 39,40,41,42
set gyro_2_bustype = SPI
set gyro_2_sensor_align = CW0
set craft_name = Cinebot30
set pilot_name = Mat

profile 1

# profile 1
set vbat_sag_compensation = 100
set horizon_level_strength = 50
set thrust_linear = 20
set feedforward_averaging = 2_POINT
set feedforward_smooth_factor = 75
set feedforward_jitter_factor = 13

rateprofile 0

# rateprofile 0
set roll_rc_rate = 2
set pitch_rc_rate = 2
set yaw_rc_rate = 2
set roll_srate = 50
set pitch_srate = 50
set yaw_srate = 50

# end the command batch
batch end
```

### Add any other context about the problem that you think might be relevant here

Other build with M8 gps works fine and smooth. Will try to set M10 on a working build and report.

adam-ah, 2023-07-16:
If that matters, I have the `GEPRC GEP-F722-45A AIO v2`, and I wired a `HGLRC M100 MINI GPS` to the recommended UART (R3, T3, UART3) and the GPS is powered up when a USB cable is connected to the FC.

I have very frequent disconnects at 115200 and intermittent ones at 57600 baud from the GPS (not losing satellites, but BF thinking GPS module disappeared) during flights, which I'm trying to debug, but that seems unrelated to the issue here.

Pancronos, 2023-07-16:
As haslinghuis said uart should not matter, the difference is that with some fc and gps it works just with USB power. With pr#12799 they all work with battery plugged in. Btw I think my geprc aio is v1. 
On all the test fc firmware had the pr#12799 and i runned u-center2 with configuration from rabbit-ambulance (iflight gps mod). The best gps so far is that cheap A-SKY AK413, ultrafast in getting huge number of satellites and extremely precise.

haslinghuis, 2023-07-16:
Issues with `BZGNSS 181` not getting icon. Get's a fix in a couple of minutes but even does not update GPS Signal Strength column in configurator. After flashing #12799 all seems to work.

Pancronos, 2023-07-31:
tested to be ok with pr#12799
walksnail WS-M181 (COPJ-18GPS) 
site: https://caddxfpv.com/products/walksnail-ws-m181gps
foxeer m10q-120 
site: https://www.foxeer.com/foxeer-m10q-120-gps-5883-compass-g-517

jonathanoeijoeng, 2023-08-06:
anyone has same experience using FC speedybee F7 v3? I have double checked in wiring but gps and magnetometer still not light up. I'm using flywoo m10 mini with compass

haslinghuis, 2023-08-06:
Add `#12799` in commit field while flashing.

closed 2023-08-12 by blckmn

P-I-Engineer, 2023-09-19:
confirmed with 2 fc's now. 

walksnail WS-M181 is not working with F405 fc's. I flashed pr 12799 test as well. 

to repeat issue, flash betaflight, no gps icon, no connection. flash inav, gps works as desired.

2 boards tested 
- diatone mamba f405 mk4 "B"
- speedybee f405 mini

confirmed gps does work with flywoo gps and betaflight.

"status" in cli shows uart cycling speeds

haslinghuis, 2023-09-20:
WS-M181 works on FOXEERF745AIO (firmware 4.5)

P-I-Engineer, 2023-09-20:
anyone have a f405 to test with? i don't have a f7 to test with

Pancronos, 2023-09-27:
> confirmed with 2 fc's now.
>
> walksnail WS-M181 is not working with F405 fc's. I flashed pr 12799 test as well.
>
> to repeat issue, flash betaflight, no gps icon, no connection. flash inav, gps works as [...]

I did test with PR#12799 and diatone mamba F405 mini stack, was working fine, make sure you use nightly configurator and of course plug battery. I suspect that this behaviour is caused by the selected uart, moreover the power pad used. If could try another uart would be great

Killerbeans, 2024-11-17:
Please check the wiring of the GPS to FC. Most wiring diagrams of the GPS to the FC online to not take into account the color codes are not the same. SCL, SDA, Tx and Rx should be checked thoroughly not just by color of cable.