betaflight/betaflight #12945: BF 4.4 : Battery average cell voltage show always 0.0V and flash

moorfpv, 2023-07-06:
### Describe the bug

In BF 4.4, For "Battery average cell voltage" I get always get 0.0V in my OSD.
Waht I use:
- BF 4.4.2
- BlueJay 19.2
- Foxeer F745 AIO V4
- TBS Sixty9 Tracer

In BF 4.3.2 works all fine. I get the correct "Battery average cell voltage".

### To Reproduce

Install BF 4.4.2 on Foxeer F745 AIO V4.

### Expected behavior

See the correct "Battery average cell voltage".

### Support ID

```txt
Sorry, I flashed the Board to 4.3.2 before I write this issue.
```

### Flight controller

Foxeer F745 AIO V4

moorfpv, 2023-07-08:
LiPo is plugged in. In BF 4.4 no voltmeter!
The voltmeter in BF 4.3 works fine.

BF 4.4.2
[image]

haslinghuis, 2023-07-08:
Works on 4.4.2 and 4.5.0 on speedybeef7mini.
Please check if using [portable nightly build](https://github.com/betaflight/betaflight-configurator-nightlies/releases) resolves the issue.

haslinghuis, 2023-07-08:
Works for me on 4.4.2 and 4.5.0 with speedybeef7mini. Did not check OSD.

moorfpv, 2023-07-08:
In BF 4.5 works the voltmeter also fine on the Foxeer Reaper V4 AIO.
I have this issue only on 4.4.2 ... hm ?!

haslinghuis, 2023-07-08:
Did you try 4.4.3 zulu pre-release?

matyas-brenner, 2023-08-12:
I can confirm this issue. I have the same problem with BF 4.4.2 using Foxeer Reaper F745 AIO

matyas-brenner, 2023-08-14:
A member of my local community was able to help me with this.
This fixes the problem.
`set adc_device = 1
save`