betaflight/betaflight #12946: Support for Formationflight/iNAV radar

sevet, 2023-07-07:
### Is your feature request related to a problem? Please describe

Flyying in formation with your fpv buddies on quads is tricky, once you first lose them its game over.

### Describe the solution you'd like

Please add support for formationfligh (fork of iNAV radar)
This shows you your friends location on the OSD and helps track them.

This is the official website:
https://formationflight.org/getting-started/

They actually have a betaflight fork with the feature already:
https://github.com/MUSTARDTIGERFPV/Betaflight

### Describe alternatives you've considered

iNAV but i dont want to

### Other information

Nope

Prajwal-Koirala, 2023-12-27:
Thanks for suggesting support for FormationFlight! I'll share your feedback with the development team.

I understand your frustration, but unfortunately, I'm not able to assist with technical modifications like this. It's best to reach out to the developers directly for further support.

sevet, 2025-02-28:
Any chance to see this icorperated?
It will be amazing, 
currently flying long or even close with friends is a one miss deal, once you loose them its very hard to find them....

sevet, 2025-10-15:
Bumping, is there any chance this will enter the roadmap? 
finding my friends after take off is hard, flying together is amazing especially long range and there finiding a friend is very difficult and once you loose him its nearly impossible to join together.

closed 2026-09-26 by blckmn