betaflight/betaflight #12950: rx uart pads not working in 4.4.2 for flywoo goku gn745 AIO v1

kakashi2018, 2023-07-11:
### Describe the bug

updated from BF 4.3.2 to 4.4.2 and noticed that rx pads are not working with dji o3( tried with uart number 2 and 5) and hence the arm signal is not being picked up and so unable to auto record footage when quad is armed. the tx pads are working as I see the osd elements showing up in the goggles

### To Reproduce

connect dji 03 to uart 2 or uart 5

### Expected behavior

arm signal should be picked up by o3 air unit and also the voltage should be coming up in the goggle osd

### Support ID

```txt
9e7fdfdb-c874-471c-b819-db7d9ce78412
```

### Flight controller

Flywoo Goku GN745 AIO version 1

### How are the different components wired up (including port information)

power from direct main voltage pads, tx and rx connected to either uart 2 or uart 5

closed 2023-07-17 by kakashi2018

haslinghuis, 2023-07-17:
What was the issue?

kakashi2018, 2023-07-17:
Used diff from betaflight 4.3.2 to 4.4.2 and some msp related config got
messed up. Did a clean preset install and it is working fine now.