betaflight/betaflight #12954: GPS settings: updating online documentation to match code

adam-ah, 2023-07-14:
### Is your feature request related to a problem? Please describe

Many (including myself :( ) struggle with intermittent GPS disconnects (not signal lost, rather BF not recognising the module in-flight, then recognising again, then losing the module again...)

Going through the possible explanations include setting up the GPS manually based on the online documentation at https://betaflight.com/docs/development/Gps

This is outdated and does not match the code any more.

### Describe the solution you'd like

I recommend changing the documentation so it's in-line with the code (and ublox recommendations):

### NAV-SOL

## From UBLOX documentation

The UBX-NAV-SOL and UBX-NAV-STATUS messages also report whether a fix is valid in their gpsFixOK and GPSfixOk flags. These messages have only been retained for backwards compatibility and users are recommended to use the UBX-NAV-PVT message
in preference.

## Documentation online
Says use NAV-SOL but code tries NAV-PVT primarily
https://betaflight.com/docs/development/Gps

## Current code
```
src/main/io/gps.c:692
gpsData.ubloxUsePVT = true;
```

## Fix

Change doc to recommend NAV-PVT only (not the messages that are outdated/not supported on some devices).
---------------------------------------------------------------------------------

### NAV-PVT

Similarly, if PVT is used, neither VELNED nor POSLLH is not used, but the documentation asks for it to be turned on
https://betaflight.com/docs/development/Gps

## Current code
```
src/main/io/gps.c:721
case UBLOX_MSG_POSLLH:
if (gpsData.ubloxUsePVT) {
    gpsData.state_position++;
} else {
   ubloxSetMessageRate(CLASS_NAV, MSG_POSLLH, 1); // set POSLLH MSG rate
}
```

```
src/main/io/gps.c:735

case UBLOX_MSG_VELNED:
if (gpsData.ubloxUsePVT) {
    gpsData.state_position++;
} else {
    ubloxSetMessageRate(CLASS_NAV, MSG_VELNED, 1); // set VELNED MSG rate
}
break;
```

## Fix

Change doc to recommend NAV-PVT only (not the messages that are outdated/not supported on some devices).

---------------------------------------------------------------------------------

### NAV-SVINFO

## From UBLOX documentation
Comment Information about satellites used or visible This message has only been retained for backwards compatibility; users are recommended to use the UBX-NAV-SAT message in preference.

## Documentation online
Ssays use NAV-SVINFO but code tries NAV-SAT primarily
https://betaflight.com/docs/development/Gps

## Current code
```
src/main/io/gps.c:693
gpsData.ubloxUseSAT = true;
```

## Fix

Change doc to recommend NAV-SAT only (not the messages that are outdated/not supported on some devices).

---------------------------------------------------------------------------------

### GPS NAV5 mode

## Documentation online
Says to use Pedestrian GPS model, but code uses _stationary_ to lock then _airborne 4G_ to fly
https://betaflight.com/docs/development/Gps

## Current code
```
src/main/pg/gps.c:39
.gps_ublox_acquire_model = UBLOX_MODEL_STATIONARY,
.gps_ublox_flight_model = UBLOX_MODEL_AIRBORNE_4G,
```

## Fix

Change online to recommend Airborne 1G (or 4G), never recommend pedestrian as the position will be invalidated often (measurements are often outside of sanity check limits for pedestrian mode, see below)

[image]

### Describe alternatives you've considered

Manually reading the 460p long ublox protocol specification and matching it up with the C code of BF.

haslinghuis, 2023-07-14:
@adam-ah Please note that GPS code is still under revision.

adam-ah, 2023-07-14:
That's all and well @haslinghuis but it does not explain the discrepancy between the current code and current documentation (which, by the sound of it, will just get bigger after the *revision*?)
Again, I'm happy to submit a PR for the doc change if that's any help?

haslinghuis, 2023-07-14:
Thanks. Always happy for contributions. Just wanted to make you aware new GPS code might be totally different. As 4.3 and 4.4 introduced major changes, 4.5 could potential change much more before final release.

haslinghuis, 2023-08-15:
Fixed in https://github.com/betaflight/betaflight.com/pull/201

closed 2023-08-15 by haslinghuis