betaflight/betaflight #12957: Please make ACRO_TRAINER a standard flight mode again.

jawz101, 2023-07-15:
### Is your feature request related to a problem? Please describe

When cloud builds were imlemented, ACRO TRAINER was taken out of a standard build.  It's just another flight mode.
Personally, it is great for tiny whoops when flying outside yet still not necessarily wanting to do flips.

I recently bought a Flywoo Firefly 1s Nano Baby v2 and they use the Core Only build option.  Consequently, the ACRO TRAINER flight mode is unavailable and the quad won't fly with what is supposed to be its build target- FLYWOOF411_5IN1_AIO

### Describe the solution you'd like

Reintroduce ACRO TRAINER flight mode to the standard build.

### Describe alternatives you've considered

I tried compiling the FLYWOOF411_5IN1_AIO build target while including the ACRO TRAINER flight mode and the quad won't run it.
[1S Firefly V2 WALKSNAIL ELRS ICM42688 BF441 230616.txt](https://github.com/betaflight/betaflight/files/12060197/1S.Firefly.V2.WALKSNAIL.ELRS.ICM42688.BF441.230616.txt)

haslinghuis, 2023-07-15:
The dump points to target `FLYWOOF411ELRS` which we don't have in store.

<details>
  <summary>FLYWOOF411ELRS dump</summary>

```
# 

# dump

# version
# Betaflight / STM32F411 (S411) 4.4.1 Apr  6 2023 / 04:23:14 (e43d591) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name FLYWOOF411ELRS
manufacturer_id FLWO

# name: FLYWOO

# resources
resource BEEPER 1 C14
resource MOTOR 1 B10
resource MOTOR 2 A08
resource MOTOR 3 B07
resource MOTOR 4 B06
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A02
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 NONE
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 B00
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 NONE
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
[... 902 lines cut ...]
set quickrates_rc_expo = OFF
set roll_rc_rate = 108
set pitch_rc_rate = 108
set yaw_rc_rate = 100
set roll_expo = 0
set pitch_expo = 0
set yaw_expo = 0
set roll_srate = 73
set pitch_srate = 73
set yaw_srate = 70
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998
set roll_level_expo = 0
set pitch_level_expo = 0

# end the command batch
batch end

# 
SAVE
```
</details>

jawz101, 2023-07-15:
According to their support page, to update to 4.4.x firmwares you use the FLYWOOF411_5IN1_AIO build target and use the Core Only build option.

https://flywoo.tawk.help/article/firefly-1s-dc16fr16-nano-baby-v2

haslinghuis, 2023-07-15:
Core build is only for support purposes and not meant for normal operation.

- Choose `CRSF` as radio protocol
- Add `ACRO_TRAINER` in build options
- Add `GYRO_SPI_ICM42688P ACC_SPI_ICM42688P` to custom defines.
- Flash and see if anything is still missing.

Recommend using latest nightly configurator or be sure to enable show release candidates and enable expert mode on the firmware flasher tab.

jawz101, 2023-07-15:
Thank you for taking a look @ this, btw.  Since you're changing it to a config issue I'll collect the standard troubleshooting info.  It may possibly take a new build target or something since this is a new Flywoo model.  [Here is their product page, btw](https://flywoo.net/products/firefly-1s-dc16-nano-baby-quad-walksnail-v2.0-bnf?sku=18059402334187804914872494).

The gyro isn't detected & the ExpressLRS receiver is not detected

support data:

bff27cdc-e891-48f4-b916-d89150b09b1d

[image]

haslinghuis, 2023-07-15:
Have you added ICM42688P drivers in custom defines? As they don't appear in the build log.

Otherwise do a core build on 4.5 and submit the support ID from there so we can see what's actual on the board.

jawz101, 2023-07-15:
hm.  I guess I didn't do it correctly the 1st time because now it worked. Can these drivers be added to that build target?

Thank you again!

haslinghuis, 2023-07-15:
Is this with or without restoring the dump?

jawz101, 2023-07-15:
For this one specifically, the other deviations from the default.  The quad would spin out of control without the motors resource changes but once I added those from the stock configuration and did yaw motors reversed it would hover in place.

`GYRO_SPI_ICM42688P ACC_SPI_ICM42688P`

```
# resources
resource MOTOR 1 B10
resource MOTOR 2 A08
resource MOTOR 3 B07
resource MOTOR 4 B06
resource SERIAL_TX 11 B00
resource SERIAL_RX 11 B04
resource I2C_SCL 1 B08
resource I2C_SDA 1 B09

# feature
feature -RX_SPI
feature SOFTSERIAL

# serial
serial 0 131073 115200 57600 0 115200
serial 1 64 115200 57600 0 115200
```

KarateBrot, 2023-07-15:
@jawz101 
> Can these drivers be added to that build target?

Problem is that `FLYWOOF411ELRS` is not an official target. Flywoo never added it to the list of targets to make it official. I think we need to contact Flywoo because right now this is a mess.

closed 2023-08-26 by github-actions[bot]