betaflight/betaflight #12961: Dynamic Idle min. to 15% Throttle with RPM Filter Disabled

spatzengr, 2023-07-16:
### Describe the bug

Dynamic Idle has two different behaviors depending if the RPM filter is enabled or disabled.  If the RPM filter is enabled along with Dynamic Idle, all is normal.  If the RPM filter is DISABLED while Dynamic Idle is enabled, the minimum motor command will be 15.01% regardless of Dynamic Idle settings.

[image]

In the zipped logs, I have tested BF 4.4.0 vs. 4.4.2 vs. 4.3.2.
[LOGS of RPM on vs. OFF.zip](https://github.com/betaflight/betaflight/files/12064795/LOGS.of.RPM.on.vs.OFF.zip)

### To Reproduce

Set Dynamic Idle in BF 4.4.x and turn ON or OFF the RPM Filter.

### Expected behavior

Minimum motor command will be 15.01% when the RPM filter is disabled.

### Support ID

```txt
*** Support data submitted *** 
Id: "a5aa9f09-7947-4a85-b51d-3797506e4430"
```

### Flight controller

BetaFPVF411

SupaflyFPV, 2023-08-01:
this is the same behaviour I noticed, which I think is related to bidir DSHOT. I noticed it not when switching on RPM filtering but dynamic idle on or off. Dynamic Idle off I had normal 0 throttle. I will get reproducible info asap...

https://github.com/betaflight/betaflight/issues/12828

ctzsnooze, 2023-08-13:
I'm not able to reproduce this issue with 4.4.2 [01-Jun-2023], or in any 4.5 development branch.

ctzsnooze, 2023-08-13:
Thanks for the report, apologies for being slow to look into it, have been busy with the GPS module interface code.

I can confirm that there is a problem in 4.4.2.  I don't know exactly when this issue was introduced.

The problem doesn't exist in 4.5 including with the refactoring in PR#13012.  4.5 brings significant GPS improvements and a number of bug fixes.

I would be grateful if users who wish to use RPM filtering 'off', and DShot telemetry and Dynamic Idle `on`, would confirm that the bug is fixed in 4.5.

4.5 is entering feature freeze and is ready for flight testing.

ExplodingFist, 2023-09-06:
This issue is probably not fixed, as when it is enabled, but breaks in 4.5-Zues the symptoms described above happen. Now the "RPM Filter" no longer shows when it prevents arming (after disarm), where in 4.4.2 you would see that error. Note that 15.01% isn't the case when idle on the ground, but seems to be the case in the air. 

See: https://github.com/betaflight/betaflight/issues/13068

ctzsnooze, 2023-10-29:
@ExplodingFist 
Please see if https://github.com/betaflight/betaflight/pull/13140 improves the situation.

We had some residual code that goes back to when dShot telemetry was originally put in place, it was only provided on the basis that it was for RPM filtering; the two were locked together as one function.  A check existed that said, "if RPM filtering is enabled, then if DShot Telemetry is not working, do not arm".  

With dynamic idle, however, some people wanted to use dynamic idle without needing to use RPM filtering.  This resulted in a number of code changes to make that possible.  

Unfortunately, however, some checks for the functional status of DShot Telemetry remained focused around RPM telemetry.  The arming block only checked DShot Telemetry if RPM filtering was enabled.  As a result, if DShot Telemetry was broken, or not working for any reason, and if RPM filtering *was not* enabled, but Dynamic Idle *was* enabled, it would be possible to arm but not have any useful RPM data for dynamic idle.  That resulted in Dynamic Idle being sent 'zero' RPM values, so it would lift idle to 15%.

The changes in PR13140 should result in an arming block whenever DShot Telemetry is not working.  It won't matter what DShot Telemetry is used for; even if used for 'nothing', you still won't be able to arm unless it is working.  A `DSHOT_TELEM` warning should show in the OSD.  At present in Configurator the old `RPM_FILTER` warning may be there until that is updated also.

The changes in PR13140 will prevent people from unexpectedly finding a high Idle value from Dynamic idle when their RPM telemetry is broken and they aren't using RPM filtering.  This should make it easier for you to figure out what is going on.

A separate issue affecting DShot Telemetry on some builds caused the underlying failure of the telemetry that led to your issues.  Fixes have been implemented to make DShot Telemetry more reliable in our code, but most of those problems appeared to be due to ESC firmware errors.

Note that if DShot Telemetry fails mid-flight, then RPM filtering will fail, and Dynamic idle will fail mid-flight..  The increase in idle value should alert the user.  We should see a `DSHOT_TELEM` warning.  It is best to not keep flying if RPM filtering or Dynamic idle is enabled but DShot Telemetry is broken.

haslinghuis, 2023-11-12:
@ExplodingFist please log with DSHOT_TELEMETRY_COUNTS debug mode enabled.

ctzsnooze, 2023-12-04:
@ExplodingFist unless we hear back from your soon, we will close this issue.

haslinghuis, 2023-12-05:
Duplicate of #12828

closed 2023-12-05 by haslinghuis