betaflight/betaflight #12970: Artefacts in ICM-42688-P gyro output

tbolin, 2023-07-19:
### Describe the bug

The angular rates from ICM-42688-P IMUs can contain artefacts in the form of short sharp spikes.
I have tested with two different FCs, a Matek H743-SLIM V3, and a GEPRC GEP-F722-45A AIO V2. Both FCs have similar artefacts in the output. 
I'm not sure that the artefacts causes a problem in flight, but they could be an indicator of something else being slightly wrong.

The FCs was powered over USB during testing, and not connected to anything else during testing.
The tests used 4.5-zulu at commit [1485438](https://github.com/betaflight/betaflight/commit/1485438c40176520eabadca46cb2509d478b3069). I have seen similar behavior with earlier builds, so I suspect these artefacts aren't something new. 

**Typical artefacts**
[image]
[image]

**Logs**
[ICM42688-P artefact.zip](https://github.com/betaflight/betaflight/files/12094934/ICM42688-P.artefact.zip)
[image]

### To Reproduce

* Take FC that is capable of logging at 8 kHz and has a ICM-42688-P gyro.
* Flash FC with [1485438](https://github.com/betaflight/betaflight/commit/1485438c40176520eabadca46cb2509d478b3069
* Set black box logging rate to 8 kHz, and turn off all fields except unfiltered gyro and accelerometer (the later is probably not necessary).
* Set `blackbox_mode = ALWAYS` and save.
* Keep the FC still for a few seconds to let the gyro to calibrate,
* Wave FC around for 10-30 seconds
*  Set `blackbox_mode = NORMAL`
* Extract and view log, small, spike shaped artefacts should now appear in the unfiltered gyro data.

### Expected behavior

There should not be any small spikes in the gyro data.

### Support ID

```txt
NA
```

### Flight controller

Matek H743-SLIM V3, and GEPRC GEP-F722-45A AIO V2

### Other components

None

nerdCopter, 2023-08-09:
rather surprising :(
foxeerf722 icm42688p
[image]

nerdCopter, 2023-08-09:
i think they all have pos & neg spikes. i just selected some areas easiest to see.  i wonder if speed of movement or momentum is a factor.  maybe momentum as seemingly spike are triggering near slow-down before change of direction :man_shrugging:  (highly unscientific cognition :joy: ). sometime i moved wildly, and sometimes not.
[icm42688p-spikes.zip](https://github.com/betaflight/betaflight/files/12304500/icm42688p-spikes.zip)

Quick-Flash, 2023-08-09:
@nerdCopter Could you try different internal lowpass filter settings? Just set `gyro_hardware_lpf` to some different values.

nerdCopter, 2023-08-09:
as @Quick-Flash requested: [icm42688p_spikes_with_gyro_hardware_lpf_options.zip](https://github.com/betaflight/betaflight/files/12306883/icm42688p_spikes_with_gyro_hardware_lpf_options.zip)

here is Emu log, but could not limit data to only raw: (ported Betaflight driver; pre-BF-SPI-changes) [EMUF_BLACKBOX_LOG_20230809_152753_8K_debug-eq-gyroRaw.BBL.zip](https://github.com/betaflight/betaflight/files/12306901/EMUF_BLACKBOX_LOG_20230809_152753_8K_debug-eq-gyroRaw.BBL.zip)

it occurred to that me as humans, we want perfection, but does this problem even matter considering gyro_filtered would not contain the spikes?

hydra, 2023-08-10:
I see the same issue on a very early version of the H7RF BF firmware (git rev: 56b8bbca6, which has a H7RF code + a BF parent of f25642cc56) and on a more recent one.

[image]

The height of the flat spots on the upward transition is the same and the height of the flat spots on the downwards transition is the same too.  happens on both negative and positive values.

However, I will say that in practice the H7RF still flies great, clearly the filtering is working well.

hydra, 2023-08-10:
Also note that the H7RF's have additional gyro power supply filtering and separate supplied to VDD and VDDIO on the gyro, whereas on the Matek board, IIRC, VDD and VDDIO are tied together, so we can probably rule out gyro power supply design being the anything to do with the issue.

For those wanting to experiment, the H7RF's gyro (ICM42688P) has it's FSYNC pin wired up to PD14 on the H730 and INT wired up to PD15 which are on TIM4-CH3/CH3 respectively, so you can try generating the gyro's clock signal from BF to see if that makes any difference...

Quick-Flash, 2023-08-11:
> @nerdCopter : Well ... filtering in BF was downgraded to Q=0.5 filters just because it looks better on screen ;-)
> 
> 

What are you talking about? The biquad filter still uses a Q of 1/sqrt(2.0) which is approximately 0.7071. The pt1-pt3 filters are not filters with a Q of 0.5.

hydra, 2023-08-11:
> @hydra: Maybe the gyro is switching resolution (or channel or something) internally and this is resulting artefact? Gain error between different signal paths may cause this.

Interesting thought, but probably someone should contact Invensense TDK to ask them what's going on.  Anyone got a good technical contact there?

> It would be interesting to test it on dual gyro board ...

Indeed, I did make one a couple of years ago with an H730 and 2x ICM42688P's that never got released (similar spec to H7RF, but dual gyro instead of Gyro + RF), I supposed I could make an updated BF target for it...

hydra, 2023-09-19:
@tbolin Did you contact Invensense/TDK about this?

tbolin, 2023-09-20:
@hydra No, I haven't contacted them.

tridge, 2023-09-25:
We've been seeing this in ArduPilot too. The severity of it varies, but on two occasions this forced ArduPilot to change IMUs (what we call a "lane change") in flight. If we hadn't switched lanes then it could have triggered a crash due to an DC bias artefact on the gyro as the flat portions are sometimes not symmetric. The estimated roll/pitch can go off a long way.
https://github.com/ArduPilot/ardupilot/issues/25025
[image]
[image]
the flat portions are bitwise identical, even if you enable the "20 bit" (actually 19 bit) resolution in the FIFO transfer

c3r63ru5, 2023-10-17:
I've purchased a Mamba F740 (AIO) which also uses the ICM-42688 P. I checked the gyro with the firmware originally flashed (4.3.0) and vendor defaults, then thought it might help to update to the latest BF (4.4.2)

This is with 4.3.0 and vendor defaults:
[image]

And this with 4.4.2 and defaults:
[image]

So apparently this is still an issue, does anybody have an idea if this is fixable in BF? I might send the FC's back and find alternatives if this cannot be addressed.

tbolin, 2023-10-17:
@c3r63ru5 
As far as I'm aware it's not fixable, at least not with any setting currently in BF. That said, take a look at the data after filtering too before you come to any drastic conclusions. Those 2 ms blips tend to be a lot less visible after a lpf and even less after the d-term filters.

c3r63ru5, 2023-10-18:
@tbolin thanks for your response, thats bad news.

With 4.4.2, these artefacts are more "spiky", and default filters are not getting rid of them, see below.

[image]

In normal cruising this might not affect anything, but this one blip for every time rates are "inverted" would cause a spike in d-term in every snappy manoeuvre.

Maybe I should see how the quad behaves in practice, but I'd rather try sending the FC back before I waste my time on the build.

tridge, 2023-10-22:
@tbolin TDK got back to us. The problem is an undocumented feature "AFSR". It auto-switches between low-noise low range and higher noise high range, when it switches it gets stuck for a while
the fix for ArduPilot is here: https://github.com/ArduPilot/ardupilot/pull/25332
I flew it today and confirmed the fix. No more stuck gyro on ICM42688
TDB says it impacts any IxM-42xxx, but we have only seen it on ICM42688

SteveCEvans, 2023-10-22:
We don’t use the FIFO, so just setting bit 6 in config register 1 may fix it. Do we have any documentation from TDK that describes this that could be added to a PR for reference? I’ll happily raise a PR for this.