betaflight/betaflight #12973: STMH7 ledstrip bug

gabruzzzz, 2023-07-20:
### Describe the bug

me and one of my friend have the same issue with H7's. I have a kakute mini H7 and he has a foxeer H7 and does the same thing. we tried to wire ledstrips on the fc, configured everything correctly (led changing color based on vtx channel for racing) and leds sometimes have one color per led while power up. i have vtx pitmode & led low and even with those deactivated leds dont keep the same color, and keeps changing color if u do on/off led low switch.
I have tried even to assign a color from tab but still some leds have different color than the chosen one.

### To Reproduce

- grab a H7 FC and solder leds on it
- flash 4.4.x
- configure leds from the tab
- (non mandatory) configure vtx pitmode & led low from tab
- power up the FC and try to change color or on/off pitmode and led low
- check led color

### Expected behavior

- led are not all the same color
- led dont change color if u change settings
- if using colors by vtx channel try to change channel but leds are not changing

### Support ID

```txt
defaults nosave

# version
# Betaflight / STM32H743 (SH74) 4.4.3.RPM_KAACK_V4 Jun 10 2023 / 11:06:24 (67643deec) MSP API: 1.45

# config: YES

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name KAKUTEH7MINI
manufacturer_id HBRO
mcu_id 004600433430510638303334
signature 

# name: ', gabruzz

# feature
feature -AIRMODE
feature LED_STRIP
feature ESC_SENSOR

# serial
serial 0 0 115200 57600 0 115200
serial 1 8192 115200 57600 0 115200

# beacon
beacon RX_SET

# led
led 0 0,0::CV:0
led 1 1,0::CV:0
led 2 2,0::CV:0
led 3 3,0::CV:0
led 4 4,0::CV:0
led 5 5,0::CV:0
led 6 6,0::CV:0
led 7 7,0::CV:0
led 8 8,0::CV:0
led 9 9,0::CV:0

# color
color 1 240,0,255
color 2 240,0,255
color 3 0,0,255
color 4 0,0,255
color 5 0,0,0
color 7 0,0,0
color 8 0,0,0
color 9 0,0,0
color 10 20,10,0
color 11 330,0,255
color 12 0,0,0
color 13 255,255,255

# mode_color
mode_color 7 0 4

# aux
aux 0 0 0 1925 2075 0 0
aux 1 0 1 1250 1300 0 0
aux 2 13 1 1425 1575 0 0
aux 3 15 2 1700 2100 0 0
aux 4 35 1 1250 1325 0 0
aux 5 39 2 1950 2050 0 0

# adjrange
adjrange 0 0 0 1950 2050 29 2 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BAND_A   A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BAND_B   B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BAND_E   E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 AIRWAVE  F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 LOWRACE  L CUSTOM  5362 5399 5436 5473 5510 5547 5584 5621
vtxtable powerlevels 4
vtxtable powervalues 1 2 3 4
vtxtable powerlabels 25 100 200 MAX

# rxfail
rxfail 3 s 875
rxfail 4 s 975
rxfail 5 s 1500
rxfail 6 s 2000

# master
set gyro_lpf1_static_hz = 0
set dyn_notch_count = 2
set dyn_notch_q = 450
set dyn_notch_min_hz = 125
set dyn_notch_max_hz = 650
set gyro_lpf1_dyn_min_hz = 0
set gyro_lpf1_dyn_max_hz = 0
set acc_hardware = NONE
set acc_lpf_hz = 10
set acc_calibration = -17,15,-411,1
set baro_hardware = NONE
set min_check = 1010
set max_check = 1995
set rc_smoothing_auto_factor = 25
set rc_smoothing_auto_factor_throttle = 25
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set motor_output_reordering = 2,3,0,1,4,5,6,7
set ibata_scale = 200
set yaw_motors_reversed = ON
set small_angle = 180
set simplified_gyro_filter = OFF
set report_cell_voltage = ON
set ledstrip_grb_rgb = RGB
set ledstrip_profile = RACE
set osd_logo_on_arming = ON
set osd_vbat_pos = 385
set osd_rssi_pos = 33
set osd_link_quality_pos = 65
set osd_rssi_dbm_pos = 134
set osd_tim_1_pos = 406
set osd_flymode_pos = 44
set osd_throttle_pos = 14391
set osd_vtx_channel_pos = 8298
set osd_crosshairs_pos = 14574
set osd_current_pos = 8566
set osd_mah_drawn_pos = 12695
set osd_craft_name_pos = 14370
set osd_pilot_name_pos = 14402
set osd_warnings_pos = 14634
set osd_avg_cell_voltage_pos = 12673
set osd_disarmed_pos = 14508
set osd_esc_tmp_pos = 196
set osd_flip_arrow_pos = 14543
set osd_core_temp_pos = 14733
set osd_stat_bitmask = 522192000
set osd_profile = 3
set osd_canvas_height = 13
set vtx_band = 5
set vtx_channel = 8
set vtx_power = 1
set vtx_freq = 5917
set rpm_filter_fade_range_hz = 100
set craft_name = ', gabruzz
set pilot_name = ![
set extra_100_throttle = A CIODO!
set extra_turtle_mode_warning = FEMO INVERSION
set extra_armed_warning = DAGHE GAS

profile 0

# profile 0
set dterm_lpf1_dyn_expo = 7
set iterm_relax_cutoff = 20
set iterm_limit = 500
set pidsum_limit_yaw = 1000
set throttle_boost = 2
set p_pitch = 41
set i_pitch = 89
set d_pitch = 23
set f_pitch = 137
set p_roll = 38
set i_roll = 81
set d_roll = 21
set f_roll = 125
set d_min_roll = 21
set d_min_pitch = 23
set motor_output_limit = 94
set feedforward_averaging = 2_POINT
set feedforward_smooth_factor = 65
set feedforward_jitter_factor = 3
set feedforward_boost = 18
set feedforward_max_rate_limit = 100
set dyn_idle_min_rpm = 35
set dyn_idle_p_gain = 35
set simplified_pids_mode = RP
set simplified_i_gain = 120
set simplified_d_gain = 70
set simplified_pi_gain = 85
set simplified_dmax_gain = 0
set simplified_feedforward_gain = 105
set simplified_pitch_pi_gain = 105
set tpa_rate = 70
set tpa_breakpoint = 1250

profile 1

# profile 1
set d_pitch = 34
set d_roll = 30
set d_max_advance = 0
set simplified_dmax_gain = 0
set tpa_rate = 70
set tpa_breakpoint = 1250

profile 2

profile 3

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set roll_rc_rate = 15
set pitch_rc_rate = 20
set yaw_rc_rate = 10
set roll_srate = 42
set pitch_srate = 37
set yaw_srate = 18
set throttle_limit_type = SCALE
set throttle_limit_percent = 80

rateprofile 1

rateprofile 2

rateprofile 3

# restore original rateprofile selection
rateprofile 0

# save configuration
```

### Flight controller

KAKUTE H7 MINI

### Other components

SKYSTARS KO60 (BLHELI_32)
LED ZEEZ (2X STRIPES)
TMOTOR PRO IV F40 
FOXEER PREDATOR V5
MAMBA ULTRA TINY VTX
FOXEER 4+ LOLLIPOP
BETAFPV LITE RX ELRS 2.4 (SW 3.2.1)

### How are the different components wired up (including port information)

I DONT REMEMBER

haslinghuis, 2023-07-20:
KAACK version spotted. @limonspb

SteveCEvans, 2023-07-20:
Please try flashing master to see if that fixes it. A change was made in #12890  which probably fixes this.

limonspb, 2023-07-20:
> KAACK version spotted. @limonspb

do we need the #12890 fix for 4.4-maintenance also?

haslinghuis, 2023-07-20:
Done. #12976

haslinghuis, 2023-08-22:
Resolves #12890, resolves #12976

closed 2023-08-22 by haslinghuis