betaflight/betaflight #12979: Battery Voltage issue or Gyro/Accel issue (Darwin ELRS2.0 F411 15A Bluejay AIO)

CustardCakes, 2023-07-21:
### Describe the bug

I am trying to configure my flight controller and am running into issues when flashing firmware. 

I am using the Darwin ELRS2.0 F411 15A Bluejay AIO flight controller and am running into different issues depending on what firmware I load. 

If I set the FC target as "DarwinF411", everything works except the battery voltage reading constantly reads around 8.9-9volts regardless of whether or not I have a 1s battery connected with usb or just have it connected via USB or just have it powered by 1s battery. The gyro and acc sensor are detected and work poroperly.

If I set the FC target as "BetaFPVF4SX1280" using the online firmware builder, The voltage reading works perfectly however it no longer detects the gyro or accelerometer. I have tried loading this firmware with the custom defines "GYRO_SPI_MPU6500 ACC_SPI_MPU6500" but it still does not detect the gyro or acc sensors. 

The flight controller originally had Firmware target:betaflight_4.3.0_BETAFPVF4SX1280. If I load the .hex firmware from the DarwinFPV website, the battery voltage reads correctly however no gyro or acc sensors are detected. 
DarwinFPV official firmware https://darwinfpv.com/pages/darwin-baby-ape

If I load the "BetaFPVF4SX1280" 4.4.2 in "Core only" mode, then the voltage reading is correct, Gyro and ACC are detected and everything seems to be working.

This is my first time building a drone and using github so please let me know if I need to clarify anything.

Thank you for your help.

### To Reproduce

Load either "BetaFPVF4SX1280" or "DarwinF411" firmware target.

### Expected behavior

Gyro, accelerometer, and voltage sensor to work correctly.

### Support ID

```txt
version
# Betaflight / STM32F411 (S411) 4.4.2 Jun  1 2023 / 17:18:18 (23d066d08) MSP API: 1.45

# config: YES
# board: manufacturer_id: BEFH, board_name: BETAFPVF4SX1280

# status
MCU F411 Clock=108MHz (PLLP-HSE), Vref=3.30V, Core temp=43degC
Stack size: 2048, Stack address: 0x2001fff0
Configuration: CONFIGURED, size: 3752, max available: 16384
Devices detected: SPI:1, I2C:0
Gyros detected: gyro 1 locked dma
GYRO=MPU6500, ACC=MPU6500
OSD: MAX7456 (30 x 13)
BUILD KEY: 2fa15dba5143e8607868b1627e939f54 (4.4.2)
System Uptime: 46 seconds, Current Time: 2023-07-21T09:08:51.075+00:00
CPU:24%, cycle time: 125, GYRO rate: 8000, RX rate: 15, System rate: 9
Voltage: 12 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
SD card: Not configured
Arming disable flags: RXLOSS CLI MSP MOTOR_PROTO

# flash_info
Flash sectors=0, sectorSize=0, pagesPerSector=0, pageSize=0, totalSize=0 JEDEC ID=0x00000000
Partitions:
FlashFS size=0, usedSize=0

# dump hardware

# version
# Betaflight / STM32F411 (S411) 4.4.2 Jun  1 2023 / 17:18:18 (23d066d08) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name BETAFPVF4SX1280
manufacturer_id BEFH

[... 284 lines cut ...]
set gyro_2_align_pitch = 0
set gyro_2_align_yaw = 0
set i2c1_pullup = OFF
set i2c1_clockspeed_khz = 800
set i2c2_pullup = OFF
set i2c2_clockspeed_khz = 800
set i2c3_pullup = OFF
set i2c3_clockspeed_khz = 800
set mco2_on_pc9 = OFF
set scheduler_relax_rx = 1
set scheduler_relax_osd = 1
set box_user_1_name = -
set box_user_2_name = -
set box_user_3_name = -
set box_user_4_name = -

# end the command batch
batch end

#
```

### Flight controller

Darwin ELRS2.0 F411 15A Bluejay AIO

nerdCopter, 2023-07-21:
you seem to have an mpu6500 version.
please try custom defines `ACC_SPI_MPU6500 GYRO_SPI_MPU6500` and report back.
**_MUST_** enable expert mode if you are using configurator 10.9; otherwise it ignores the custom defines.
[image]

CustardCakes, 2023-07-21:
I was able to figure it out. I had to use "DARWINF4SX1280HD" as the board firmware. Thank you for the help.

closed 2023-07-21 by CustardCakes