betaflight/betaflight #12980: BF 4.4.2: Crossfire telemetry, Sensor FM (Flightmode) shows !ERR*

micsafi, 2023-07-22:
### Describe the bug

While using TBS Crossfire receiver protocol with Betaflight telemetry enabled, the sensor value FM (Flightmode) shows value !ERR. All other Betaflight provided sensor values work OK, including GPS location, battery voltage, pitch/roll/yaw angles etc.

This same issue was reported earlier as issue #9860, without any clear fix.

### To Reproduce

Connect TBS Crossfire receiver with telemetry enabled. Discover all sensors from the transmitter.

### Expected behavior

The FM sensor should show some meaningfull value describing flight mode (Acro, Angle etc)...

### Support ID

```txt
*** Support data submitted *** 
Id: "9956d126-afd4-49c8-984e-5b5237ef8214"
```

### Flight controller

SPEEDYBEEF7V2

### Other components

- Radiomaster TX16S transmitter with TBS Crossfire Micro TX V2 with TBS V 4.10 firmware
- TBS Crossfire Nano Receiver with TBS V 4.10 firmware

micsafi, 2023-07-22:
I think this FM sensor bug might not be limited only to TBS but probably also occurs with ELRS transmitters. Check out this Joshua Bardwell tutorial where he has the same issue (FM = !ERR) with ELRS radio link (video time 2:51).

https://youtu.be/SAQwi6QomRg?t=171

haslinghuis, 2023-07-22:
Quad needs to be armed to display something else then `!ERR`

micsafi, 2023-07-22:
After testing again the FM sensor value with quad armed, I can confirm that the value is sent to transmitter correctly.

My personal use case for the FM sensor value is mainly when flying line-of-sight without Goggles and OSD. When preparing to begin LOS flight, it would be nice to check from transmitter telemetry display that the flight mode is what you want to use (in my case mainly Angle mode). 

Now with this approach with the value only sent when armed, it kinds of make it somewhat useless. When flying LOS and you are in the air, there is no time to check the transmitter screen anymore.

Secondly, the FM sensor value behaves inconsistently when compared to OSD element "Fly mode". This OSD element shows the flight mode once the quad and VTX is powered, regardless of whether the quad is armed or not.

haslinghuis, 2023-07-22:
Updated the PR. You can test in on 4.5.0 zulu by adding #12983 in the commit field before flashing.

klutvott123, 2023-07-22:
@micsafi I too fly LOS, and I agree that it would be nice to see the flight mode before arming. For some quads I have flight mode on a switch, and have ended up arming and flying with the "wrong" flight mode lol.

haslinghuis, 2023-07-22:
@klutvott123 see https://github.com/betaflight/betaflight/pull/7500

closed 2023-08-01 by haslinghuis

AlexOrlo, 2024-01-04:
> @micsafi I too fly LOS, and I agree that it would be nice to see the flight mode before arming. For some quads I have flight mode on a switch, and have ended up arming and flying with the "wrong" flight mode lol. I can't remember what the reason was for showing "!ERR" in the first place, but I'm sure there was one 😁

In my example it's usefule to know if quad/ plane is armed or not. I need it for altitude calculation in my ground system. Becouse before arm altitude show like absolute value and after arm is relative to launch altitude.

PR #12983 (2023-08-01): Rename flight mode
Fixes: #12980 

When unarmed FM telemetry element shows `!ERR`.

This PR renames the element to show the selected flight mode.