betaflight/betaflight #12984: Gyro 1 locked dma, Gyro rate 8k+

sieenia, 2023-07-23:
### Is your feature request related to a problem? Please describe

My gyro not working. 

*** Support data submitted *** 
Id: "2f129a53-b11f-4ecf-8ea2-022716344e60"

### Describe the solution you'd like

i dont now

### Describe alternatives you've considered

Flash to 4.4, reflash, fabric config

### Other information

I dont know how fix this

sieenia, 2023-07-23:
# dma show

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: LED_STRIP
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

sieenia, 2023-07-23:
# status
MCU F722 Clock=216MHz, Vref=3.28V, Core temp=58degC
Stack size: 2048, Stack address: 0x20010000
Configuration: CONFIGURED, size: 3641, max available: 16384
Devices detected: SPI:1, I2C:1
Gyros detected: gyro 1 locked dma
GYRO=ICM42688P, ACC=ICM42688P, BARO=DPS310
OSD: MAX7456 (30 x 13)
BUILD KEY: 4a045b91384883f092a92c44c6598352 (4.4.0)
System Uptime: 78 seconds, Current Time: 2023-07-23T10:02:24.436+00:00
CPU:29%, cycle time: 122, GYRO rate: 8196, RX rate: 15, System rate: 9
Voltage: 2 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
FLASH: JEDEC ID=0x00ef4018 16M
Arming disable flags: RXLOSS CLI MSP

SteveCEvans, 2023-07-23:
According to what you’ve posted the gyro is detected, sending interrupts, albeit a bit fast with a gyro period of 122us, and DMA is enabled. What do you see on the sensors tab when you move it? Both gyro and acc traces dead?

sieenia, 2023-07-24:
Acc sensor works. Gyro sensor not work.

sieenia, 2023-07-24:
I have flashed every version since 4.2. up to 4.4. I did factory reset. 
Drone flips on takeoff.

closed 2023-08-31 by github-actions[bot]