betaflight/betaflight #12985: MAMBAF405_2022A don't see baro DPS310

maximborodkin, 2023-07-23:
### Describe the bug

Board: Diatone Mamba F405 MINI version B ([link](https://aliexpress.ru/item/1005004815812821.html?sku_id=12000030597031178)).
Target: MAMBAF405_2022A

After flashing Core Only CLI found it: `BARO=DPS310`.
In version 4.3.2 it works fine.

I try to add Custom Defines like `BARO BARO_DPS310 BARO_SPI_DPS310`, but there is nothing about baro in the CLI.
Just `INVALID NAME: baro_bustype`.

### To Reproduce

Betaflight configurator 10.9.0 (697943b)
Target: MAMBAF405_2022A
Version: 4.4.2

Radio Protocol: CRSF
Other Options: GPS OSD(SD) PinIO VTX Magnetometers
Custom Defines: BARO BARO_DPS310 BARO_SPI_DPS310

### Expected behavior

Ability to set up DPS310 baro via I2C

### Support ID

```txt
2655714f-e71c-4aa8-8371-f4b93ed21b37
```

### Flight controller

Diatone Mamba MK4 F405 MINI with ICM42688P gyro. Baro is soldered on the board (built in)

haslinghuis, 2023-07-23:
Please try B version instead

maximborodkin, 2023-07-23:
@haslinghuis, yes, it works. But on sticker on the board was MAMBAF405_2022A. And it came with this firmware. Very strange. But thank you so much

closed 2023-07-23 by maximborodkin

haslinghuis, 2024-03-09:
FYI

Added the DPS310 baro driver to the 2022A variant (which uses MPU6000) as found in support ID 0cf2f55c-0f84-41d2-b88f-5dbb09d307c9