betaflight/betaflight #12988: Incorrect accelerometer scaling in MSP

flaviopinzarrone, 2023-07-25:
### Describe the bug

If I look at the accelerometer data received through MSP I notice a huge quantization error. While the FC is laying on a flat surface the least increase or decrease in the value received is of 0.019 m/s^2.

When looking at this [line](https://github.com/betaflight/betaflight/blob/293e14e8bd4b87019780ed0598cb7cfaa636415a/src/main/msp/msp.c#L1150C62-L1150C62) I wonder if that scaling is what you really meant to do there. 

If for example my accel reads 2048 at 1g that line would divide that value by 4 and cast it to uInt16. When I am to read that value I will need to divide it by 512 (and multiply it by 9.8 if I want m/s^2). But this leads to a loss of information due to the rounding error of the division by 4 you perform in that line.

### To Reproduce

With any board that has an accelerometer that has 16-bit precision and 16g range (reading 2048 at 1g) observe the accel values coming through msp while the FC is static. 

### Expected behavior

The correct resolution of the accelerometer is reflected into the data read through msp. Recalling the previous example, this would happen the reading is not divided by 4 at that line and then I divide it by 2048 in my msp client.

### Support ID

```txt
ba97d3d3-c5c3-4cf7-a381-9c0168e56047
```

### Flight controller

Holybro KAKUTE H7 V2

haslinghuis, 2023-07-25:
https://github.com/betaflight/betaflight/blob/293e14e8bd4b87019780ed0598cb7cfaa636415a/src/main/msp/msp.c#L1133-L1146

flaviopinzarrone, 2023-07-28:
Could anyone elaborate a little bit more on this?

> https://github.com/betaflight/betaflight/blob/293e14e8bd4b87019780ed0598cb7cfaa636415a/src/main/msp/msp.c#L1133-L1146

haslinghuis, 2023-08-10:
> The correct resolution of the accelerometer is reflected into the data read through msp. Recalling the previous example, this would happen: the reading is not divided by 4 at that line and then I divide it by 2048 in my msp client.

@KarateBrot

nerdCopter, 2023-08-11:
https://github.com/betaflight/betaflight/pull/13026#issuecomment-1675211746

PR #13026 (2023-12-04): Fixed incorrect accelerometer scaling in MSP
Fixes #12988
closed 2023-12-04 by haslinghuis