betaflight/betaflight #12989: Delay Disarm flag sent to O3 Air Unit over MSP

Nilssonfpv, 2023-07-26:
### Is your feature request related to a problem? Please describe

The O3 Air unit stops recording as soon as a disarm is detected, It would be very nice if the disarmed flag sent over msp could be delayed so that the Post Flight Statistics screen gets recorded in the Goggle DVR.

### Describe the solution you'd like

Delay the Disarm flag sent over msp by 5-15s, so that the post flight screen gets recorded.

### Describe alternatives you've considered

There really isnt any other option, dji themselves seem to have forgotten that the O3 Air units exist.

ctzsnooze, 2023-08-14:
We will not have a delay on disarm, so you're asking for the OSD/MSP message to not reflect the true arm/disarm state.  

I doubt we would do that.

Definitely a DJI issue; they could easily record a few seconds after the disarm message.

closed 2023-09-21 by github-actions[bot]

ASDosjani, 2025-01-15:
@ctzsnooze DJI won't do anything useful their users, especially not on older systems like O3. I think it should be an optional cli command.

nerdCopter, 2025-01-15:
@ASDosjani , a PR was proposed (https://github.com/betaflight/betaflight/pull/14090), but devs were in agreement that such could be a safety issue in general.

ASDosjani, 2025-01-15:
Alright, let's hope Steve can achieve something at DJI and the other manufacturers.