betaflight/betaflight #12990: Dshot problems shown by ESCape32

Quick-Flash, 2023-07-26:
### Describe the bug

When trying to get an ESCape32 esc working there were plenty of problems with high dshot errors, and the motor wizard and dshotprog cli command both failed to work. By reverting to BF 4.3 the issue went away. Dshot300 or Dshot600 makes no difference.

On BF 4.4
[image]

On BF 4.3 
[image]

### To Reproduce

Get an f051 based esc that runs ESCape32 and dshot will be buggy and the motor wizard and dshotprog cli will not work.

### Expected behavior

BF 4.4 dshot works just as well as BF 4.3 dshot.

### Support ID

```txt
Not relevant as no settings have been changed and it is the board default.
```

### Flight controller

MatekH743 and FuryF4OSD

### Other components

Literally just a bare FC.

howels, 2023-07-27:
Have you tried the latest 4.4.3 master or 4.5 with PR https://github.com/betaflight/betaflight/pull/12912 ?
With the new code try `set dshot_telemetry_start_margin = 30`

howels, 2023-07-29:
Please check if the author of Escape32 has followed the timings laid out in https://brushlesswhoop.com/dshot-and-bidirectional-dshot/ 
These are what the BlueJay team has used, which is effectively the reference dshot bidirectional and EDT implementation at this point.

neoxic, 2023-07-30:
@howels Author here.😊 Since the beginning, ESCape32 has been following the DSHOT spec very strictly, but the delay for telemetry frames has always been 25us. I didn't invent this value, I took it from one of the "DSHOT spec" links like the one you mentioned above. I can't (and won't) prove it because it appears most of them have been corrected in favour of the 30us value. It's not a problem to switch to a new value, of course, and it has already been done.

neoxic, 2023-07-30:
There's one thing that is not clear for me though. How is it possible that the new tightened DSHOT timings for telemetry frames could cause problems for DSHOT commands. These two things seem to be completely unrelated, but see here:
https://github.com/betaflight/betaflight/issues/12860

Quick-Flash, 2023-07-31:
This problem has been fixed now that ESCscape32 has changed its telemetry frames to a 30us time instead of 25us.

closed 2023-07-31 by haslinghuis