betaflight/betaflight #12993: DPS310 I2C baro not working with GEPRCF722_BT_HD

ftc2, 2023-07-27:
### Describe the bug

hi, i'm having problems with my I2C `DPS310` baro on `GEPRCF722_BT_HD`
core_only 4.4.2 support data: `7d229915-f599-4957-a69c-5d48f4ce4cea`

using configurator 10.9.0

it's on a GEPRC `GEP-M10-DQ` GPS module with mag and baro.
https://geprc.com/product/gep-m10-series-gps-module/
the mag (`QMC5883L`) seems to be working but not the baro (`DPS310`).

i have the mag/baro SDA wired to FC's R3 and SCL wired to FC's T3.

- i have tried cloud build with `BARO_DPS310` custom define. cloud build log: https://build.betaflight.com/api/builds/cdd69af1a125dab246f5206796de4d08/log
- i have also tried with these custom defines (**see attached screenshot**): `BARO_MS5611 BARO_SPI_MS5611 BARO_BMP280 BARO_SPI_BMP280 BARO_BMP388 BARO_SPI_BMP388 BARO_LPS BARO_SPI_LPS BARO_QMP6988 BARO_SPI_QMP6988 BARO_DPS310 BARO_SPI_DPS310 BARO_BMP085 BARO_2SMBP_02B BARO_SPI_2SMBP_02B` cloud build log: https://build.betaflight.com/api/builds/27ee3a7a310611097f191efdc2bb44dc/log

[image]

**in both cases, i get `INVALID NAME: baro_bustype` error when i `get baro_bustype`**
i also can't enable baro in the Configuration tab.

baro seems to be working in core_only (i'm blowing air onto it):

[image]

thanks in advance.

### To Reproduce

1. flash 4.4.2 with `BARO_DPS310`
2. enable baro in Configuration tab; observe it doesn't stay enabled after saving.
3. try `get baro_bustype`, get error: `INVALID NAME: baro_bustype`

### Expected behavior

working baro

### Support ID

```txt
7d229915-f599-4957-a69c-5d48f4ce4cea
```

### Flight controller

GEPRCF722_BT_HD

### Other components

FC/ESC stack: https://geprc.com/product/span-f722-bt-hd-v2-stack/
GPS: https://geprc.com/product/gep-m10-series-gps-module/

### How are the different components wired up (including port information)

T1: DJI O3
R1: DJI O3
(MSP 115200)

T2: ELRS GEPRC NANO 2.4GHz
R2: ELRS GEPRC NANO 2.4GHz
(Serial RX)

T3: GPS GEP-M10-DQ: SCL
R3: GPS GEP-M10-DQ: SDA
(GPS compass & barometer, I2C)

T4: onboard FC bluetooth
R4: onboard FC bluetooth
(MSP 115200 ?)

T5: GPS GEP-M10-DQ: Rx
R5: GPS GEP-M10-DQ: Tx
(gps sensor 115200)
R5: ESC Tx

haslinghuis, 2023-07-27:
Try using custom define `BARO BARO_DPS310`

ftc2, 2023-07-27:
hey, that worked, thanks a lot.

would be nice if someone could add `BARO` to the list of baro defines in https://betaflight.com/docs/release/Betaflight-4.4-Release-Notes

and maybe even on https://betaflight.com/docs/wiki/archive/Barometer-Configuration

closed 2023-07-27 by ftc2