betaflight/betaflight #12994: ESC temperatures work on first arm but are sporadic after second arm

hyp0dermik-code, 2023-07-28:
### Describe the bug

ESC temperatures do not display until first arm, sometimes display after disarming, and are sporadic (most times showing 0 across all 4 escs) from second arm

### To Reproduce

Enable EDT in console and OSD ESC temps.

Observe temps through VTX before, on, after first arm, and on second arm.

### Expected behavior

ESC temps work second arm.

Sporadic ESC errors should not be present second arm

### Support ID

```txt
# Betaflight / STM32F7X2 (S7X2) 4.4.2 Jun  9 2023 / 17:35:03 (23d066d08) MSP API: 1.45
```

### Flight controller

GEPRC_F722_AIO v2

### Other components

Tramp TX only UART1 
ELRS via UART2 - v3.2.1 both ends

### How are the different components wired up (including port information)

resource SERIAL_TX 5 NONE
# serial
serial 0 8192 115200 57600 0 115200
serial 3 0 115200 57600 0 115200
serial 4 0 115200 57600 0 115200

### Add any other context about the problem that you think might be relevant here

[BTFL_cli_backup_20230728_133947.txt](https://github.com/betaflight/betaflight/files/12190565/BTFL_cli_backup_20230728_133947.txt)
[BTFL_cli_20230728_153357.txt](https://github.com/betaflight/betaflight/files/12190566/BTFL_cli_20230728_153357.txt)
Debug DShot RPM logging @ 500Hz ( note spikes to 10000 in second arm of each file )
[debug_dshot_rpm_error.zip](https://github.com/betaflight/betaflight/files/12190594/debug_dshot_rpm_error.zip)
 (unrelated: also 300 seemed to be cleaner than 600 on first arm)

hyp0dermik-code, 2023-07-29:
I noted the LEDSTRIP was using the same timer as the DSHOT bitbang. Running 'resource LED_STRIP none' freed up the timer
The update rate on the second arm was doubled from 1 value every 4 zeros to 1 in 2 approx.

Disabling EDT seems to have removed the spikes to 10000 in the second arm
[dshot_442_noedt.zip](https://github.com/betaflight/betaflight/files/12206240/dshot_442_noedt.zip)

haslinghuis, 2023-08-08:
Duplicate of #12617

hyp0dermik-code, 2023-08-08:
> Duplicate of #12617

Don't think so, that sounds like a configuration issue as it never works i.e. set dshot_edt = ON has not been performed

hyp0dermik-code, 2023-08-10:
#12617 is intended behaviour if set dshot_edt = off, which is default. The motors tab of the configurator enables this, hence his behaviour.

The issue described occurs with the quad disconnected from a PC during all phases.

I've tried 4.4.0 and have the same issue described.

It still occurs with the blackbox logging set to none.

Ideally I'd power on/off the ESC between arms to isolate the problem, but it is an AiO.

damosvil, 2023-08-10:
It also happens on my side

hyp0dermik-code, 2023-08-30:
Furher to the above this disappears after RPMs increase or turtle mode is activated/deactivated.

hyp0dermik-code, 2023-08-31:
May relate https://github.com/betaflight/betaflight/issues/12828#issuecomment-1604046582

TBC - my Dshot margin setting in 4.5z, and latest release
I was also experiencing the <1% error

nerdCopter, 2023-09-06:
see also https://github.com/betaflight/betaflight/issues/13068

hyp0dermik-code, 2023-09-11:
dma show / diff all / timer show / dshot_telemetry_info (usb power only)
# version
# Betaflight / STM32F7X2 (S7X2) 4.5.0 Aug  9 2023 / 14:01:25 (807a7229c) MSP API: 1.46
# config rev: 84cbef4
[BTFL_cli_20230912_001323_GEPRC_F722_AIO.txt](https://github.com/betaflight/betaflight/files/12575891/BTFL_cli_20230912_001323_GEPRC_F722_AIO.txt)

board_name GEPRC_F722_AIO

hyp0dermik-code, 2023-09-13:
From cloud build @ commit 983c04d846de75a8570c191d14972726a0aa4f4f
[dmatimer.txt](https://github.com/betaflight/betaflight/files/12592058/dmatimer.txt)
[dmatimer_nobitbang.txt](https://github.com/betaflight/betaflight/files/12592059/dmatimer_nobitbang.txt)
[dump.txt](https://github.com/betaflight/betaflight/files/12592060/dump.txt)

[dump_nobitbang.txt](https://github.com/betaflight/betaflight/files/12592061/dump_nobitbang.txt)

perhaps of note, when i try to use 'defaults nosave' i get the following errors
[defaults_error.txt](https://github.com/betaflight/betaflight/files/12592069/defaults_error.txt)

hyp0dermik-code, 2023-09-27:
v4.3.1
[dshot_dma_timer.txt](https://github.com/betaflight/betaflight/files/12735246/dshot_dma_timer.txt)
[dshot_dma_timer_nobitbang.txt](https://github.com/betaflight/betaflight/files/12735247/dshot_dma_timer_nobitbang.txt)
[dshot_dma_timer_nobitbang_dmatim31.txt](https://github.com/betaflight/betaflight/files/12735248/dshot_dma_timer_nobitbang_dmatim31.txt)
[dump.txt](https://github.com/betaflight/betaflight/files/12735249/dump.txt)
[dump_nobitbang.txt](https://github.com/betaflight/betaflight/files/12735250/dump_nobitbang.txt)

hyp0dermik-code, 2023-11-02:
Potentially relates to: https://github.com/betaflight/betaflight/pull/11851#issuecomment-1266137629

Prajwal-Koirala, 2023-12-26:
Sure, here is a long summary of the article you requested:

This article discusses a bug in Betaflight software that causes sporadic ESC temperature readings after the second arm of a quadcopter. The bug is present in Betaflight 4.4.0 and 4.5.0. Disabling DShot EDT seems to fix the problem. The issue is marked as closed and won’t be fixed.

Here are some of the key points from the article:

* The bug causes sporadic ESC temperature readings after the second arm of a quadcopter.
* The bug is present in Betaflight 4.4.0 and 4.5.0.
* Disabling DShot EDT seems to fix the problem.
* The issue is marked as closed and won’t be fixed.

I hope this summary is helpful! Let me know if you have any other questions about this article.

CapnBry, 2024-06-06:
Betaflight 4.5.0 I am also encountering this on second+ arm.

Bluejay v0.19.1 and a cheap STM32F405, although a Speedybee F405V3 does it as well (also on Bluejay). I have a BLHeli_32 ESC with dshot_edt on as well (telemetry uart wire disconnected) and it behaves as expected, so I feel like we're looking in Bluejay's direction for this, but obviously it could be that bluejay is choking on something coming from betaflight so I mention it here. All 3 quads are using bitbang, dshot_burst on/off makes no difference.

```
# Betaflight / STM32F405 (S405) 4.5.0 Apr 28 2024 / 05:33:01 (c155f5830) MSP API: 1.46
# config rev: 3068e6e
# board: manufacturer_id: AIRB, board_name: OMNIBUSF4SD

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1 : DSHOT_BITBANG 1
    CH3 : DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_SDI 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_SDI 2
DMA1 Stream 4: SPI_SDO 2
DMA1 Stream 5: SPI_SDO 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 1
DMA2 Stream 3: ADC 2
DMA2 Stream 4: DSHOT_BITBANG 2
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

I am also seeing misreported RPM information when `dshot_edt = ON`.  9 times out of 10 the ESC temperatures work on the first arm, but on the 10th one ESC's temperature is 0 and I get this in my RPM data for that motor. Note that I always get ESC temps going nuts (sporadic 0/value) on the second and subsequent arms.
[image]

hyp0dermik-code, 2024-06-08:
Intersting @CapnBry , so with dshot_edt = on and Blheli32 ESC temps are always reported correctly on first and subsequent arms?

CapnBry, 2024-06-08:
> Intersting @CapnBry , so with dshot_edt = on and Blheli32 ESC temps are always reported correctly on first and subsequent arms?

Yup. I even tried arming/disarming like 10x and power cycling once and the dshot_edt ESC temps are fine every time. So either blheli_32 and Betaflight have the exact same bug in their EDT implementations (that breaks them on standard implementations) or Bluejay has an issue.

I looked at the RPM in blackbox when the ESC temps are flashing on / off and the RPM looked fine there, so it does not seem the random weird RPMs I see are a side effect of this temperature reporting, but when I see a 0 ESC temp on first arm, I **know** the RPM data is being misreported as well for that ESC (and it is a random ESC on a 4-in-1, not the same one).

hyp0dermik-code, 2024-06-09:
> Yup. I even tried arming/disarming like 10x and power cycling once and the dshot_edt ESC temps are fine every time. So either blheli_32 and Betaflight have the exact same bug in their EDT [...]

Could you chuck a logic analyzer on one of the Blheli32 ESCs and capture a first and second arm please?

hyp0dermik-code, 2024-06-09:
> Could you chuck a logic analyzer on one of the Blheli32 ESCs and capture a first and second arm please?

Hold fire, forgot I've got a single ESC to convert to AM32

CapnBry, 2024-06-10:
> Could you chuck a logic analyzer on one of the Blheli32 ESCs and capture a first and second arm please?

I can not, it is in an assembled race quad so there's not much room to get at anything without disassembling it entirely.

I saw [this comment on the bluejay issue tracker](https://github.com/bird-sanctuary/bluejay/issues/163#issuecomment-1683787093) and just for funsies I swapped betaflight over to 4.4.0 to see if it had any different behavior. Same behavior as 4.5.0: ESC temp works on first arm, zeroes flashing on subsequent arms.

blckmn, 2026-09-26:
We're closing this as part of a clean-up of issues more than two years old. Sorry it sat here so long without a resolution, that's on us. Betaflight has changed a lot since this was opened, so if it's still relevant on the latest release, please open a new issue with current details and link back here.

closed 2026-09-26 by blckmn

CapnBry, 2026-09-27:
I have re-enabled DSHOT EDT on 2026.6 with a bluejay 0.21 and it works across arms. Still get RPM telemetry and ESC "temperatures". So, yeah, no problem with closing this as the issue seems resolved with all the changes along the way.