betaflight/betaflight #13002: Failsafe crash, help!

lyhnsdgtdh, 2023-08-02:
### Describe the bug

Friends, can anyone help, the quad dropped today (in FS mode, I checked the return on the toggle switch) flew home as usual and went out over my head, head over heels. managed to intercept, but the reason is unknown, who can see the log? I still can't. where to attach the log file?

### To Reproduce

turned on the failsafe toggle switch, the drone flew to me and turned off at a height of 50 meters and began to fall, although before that everything was working properly on the other battery.

### Expected behavior

expected to land at a given speed and turn off the motors when it touches the ground, zero altitude

### Support ID

```txt
Unique device ID: 0x3800215931500c20383539
```

### Flight controller

 SPBE/SPEEDYBEEF7V3(STM32F7X2), version: 0

### Other components

[image]
[image]

### How are the different components wired up (including port information)

[image]

### Add any other context about the problem that you think might be relevant here

[image]
[image]

haslinghuis, 2023-08-02:
- GPS and MAG module used?
- Was the magnetometer calibrated? Our magnetometer code is very old and hence it's reliability is not established yet.
- In the screenshots see Failsafe switch is set to stage 2 and GPS rescue was allowed to arm without fix and sanity check was set to ON. In this case when GPS does not have a fix it will abort and drop.
- There are some issue with M10 modules which are being addressed in the next release.

lyhnsdgtdh, 2023-08-02:
I greet you. thanks for helping!
GPS installed
Beitian M9140 BK-880Q, magnetometer calibrated three flights ago, it
was no more than 15 minutes. when I turned on the RESC toggle switch,
15 satellites were recorded.
I'm attaching a few more files, maybe they will clarify the situation.
and I also attach the LOG 10 file, I think there you can find the
reason for disarm.Perhaps the barometer readings were incorrect?   log
11 is the arm in the air when falling when I managed to take control.
Thank you! I am hope for your help!

lyhnsdgtdh, 2023-08-02:
[get gps.txt](https://github.com/betaflight/betaflight/files/12245494/get.gps.txt)
[image]
[get.txt](https://github.com/betaflight/betaflight/files/12245495/get.txt)

lyhnsdgtdh, 2023-08-02:
why can't I upload log.BFL file unsupported file format?

haslinghuis, 2023-08-02:
compress to zip file

lyhnsdgtdh, 2023-08-02:
[LOG00011.BFL.zip](https://github.com/betaflight/betaflight/files/12245513/LOG00011.BFL.zip)

haslinghuis, 2023-08-03:
Recommend Joining our discord - there is more people able to look at the log.

lyhnsdgtdh, 2023-08-03:
DISCORD SET UP WHERE TO WRITE AND WHAT FILES TO ATTACH?

haslinghuis, 2023-08-03:
https://discord.com/invite/gV4XSq3fus

Just go to #support and point to this issue.

lyhnsdgtdh, 2023-08-03:
меня почему то блокируют, что я делаю не  так?

haslinghuis, 2023-08-03:
[image]

[image]

lyhnsdgtdh, 2023-08-03:
thanks for your answer, but I don't understand how to read these screenshots, can you comment on why the drone was disarmed at altitude when it arrived at the takeoff point?

haslinghuis, 2023-08-04:
Failsafe Phase == FAILSAFE_GPS_RESCUE (6) so that's correct.

Not sure why we see CALIBRATE_MAG in stateFlags. (seems to be GPS_FIX_EVER)

In BBL code I see
```
FLIGHT_LOG_FLIGHT_STATE_NAME = makeReadOnly([
        "GPS_FIX_HOME",
        "GPS_FIX",
        "CALIBRATE_MAG",
        "SMALL_ANGLE",
        "FIXED_WING",
    ]),
```
And in firmware only:
```
typedef enum {
    GPS_FIX_HOME   = (1 << 0),
    GPS_FIX        = (1 << 1),
    GPS_FIX_EVER   = (1 << 2),
} stateFlags_t;
```

Looks like an altitude measurement problem.

Did you log with `RTH` debug enabled?
Not sure which one we need, there is also DEBUG_GPS_RESCUE_TRACKING and DEBUG_GPS_RESCUE_HEADING

lyhnsdgtdh, 2023-08-04:
[image]

lyhnsdgtdh, 2023-08-04:
hello, if you ask in what mode the log is written, then I made a screen. I still have these files, are they needed? in one of them I saw that the barometer figures in this flight are unchanged. That is how it should be?

lyhnsdgtdh, 2023-08-04:
[image]

haslinghuis, 2023-08-04:
> Did you log with RTH debug enabled? (No - gyro scaled)

Not sure which one we need, there is also DEBUG_GPS_RESCUE_TRACKING and DEBUG_GPS_RESCUE_HEADING
These debugs enabled would have more information about why the drone thought it was already on the ground, as you say it was pretty close when it happened.

lyhnsdgtdh, 2023-08-05:
Hello my friend, if I understand you correctly, then I need to select the recording mode here so that in the following logs there will be more information about the parameters of the drone in gps resque mode? But it's not possible to select two modes here?

lyhnsdgtdh, 2023-08-05:
[image]

lyhnsdgtdh, 2023-08-05:
I have a guess, maybe I found the reason. I activated the failsafe mode not with a toggle switch, but turned off the equipment. when he saw an approaching drone, he turned it on again. perhaps the gas was at a minimum for more than 2 seconds and when the connection was restored, the drone turned off after 2 seconds. Could it be?
[image]

lyhnsdgtdh, 2023-08-05:
when I saw the drone, I turned on the control equipment again

haslinghuis, 2023-08-05:
Yes only one debug mode at the time. (Now we have 8 debug values - we maybe should look into combining some).
Also with 4.5 firmware GYRO_SCALED is automatically logged using unfiltered gyro.

[image]

Back to the topic: https://youtu.be/umKfROrX0aE

lyhnsdgtdh, 2023-08-05:
Thanks, I'll think about updating the firmware. What do you think of my assumption about the cause of the drone crash? am i right or not?

haslinghuis, 2023-08-05:
No, this setting is about entering failsafe - as we were already in failsafe. Not sure exactly what happened. Was the radio armed when turning it back on?

@lyhnsdgtdh you might want to wait for `#12799 ` to be ready. You already can use latest nightly configurator.

lyhnsdgtdh, 2023-08-05:
I don’t remember when the radio was turned on was in the arm or not. most likely yes, because when I intercepted the falling drone, I flipped the toggle switch to disarm and again to arm. If my memory serves me, that was it.

lyhnsdgtdh, 2023-08-05:
please tell me how to upgrade to 4.5, I have the newest 4.4.2 reflected
[image]

haslinghuis, 2023-08-05:
- You will need latest configurator for 4.5: https://github.com/betaflight/betaflight-configurator-nightlies/releases
- Use the portable until final release for easy upgrade.
- Enable expert mode
- Enable show release candidates
- Now you are able to install 4.5

haslinghuis, 2023-08-05:
Closing the issue, as without the gps rescue debug mode it's hard to tell what went wrong.

closed 2023-08-05 by haslinghuis

lyhnsdgtdh, 2023-08-05:
Thanks, you've been very helpful anyway! I got a lot of new knowledge!