betaflight/betaflight #13003: No RC link preset notification

ekap12, 2023-08-02:
### Is your feature request related to a problem? Please describe

We have a Telegram group to which Betflight users send their logs in case of any problems with the flight. So in more than 50 percent of cases, the problem is the lack of a preset on the RC link. Perhaps it is worth displaying some kind of warning on the OSD or a very noticeable message in the configurator when the preset is not installed?

### Describe the solution you'd like

OSD warning - No Rc Link preset
Red message in Configurator

### Describe alternatives you've considered

Default setting of some preset that will reduce performance but will not knock on turns with motors

ekap12, 2023-08-02:
[image]

ctzsnooze, 2023-08-14:
We don't know what presets have, or have not, been installed.
Currently it is up to the user to choose the correct settings for their chosen RC link frequency.
Perhaps in future we can automatically adjust these when the link speed is detected, but unfortunately that detection is not always reliable.

closed 2023-09-21 by github-actions[bot]