betaflight/betaflight #13015: MAMBAF722_2022A 4.4.2 freezes when connected to Betaflight Configurator

francis916, 2023-08-07:
### Describe the bug

When it's connected to Betaflight Configurator, Diatone MK4 F722 Mini MPU6000 FC freezes randomly. It happens both with Mac and PC. BF 4.4.0 or lower doesn't have this issue. The LED on the FC stops blinking. Got help from Diatone engineer and tries many things including cleaning the FC, removing all devices but lowering the BF version solved the issue. 

https://github.com/betaflight/betaflight/assets/18643755/5d1d5c7e-e5d1-4149-a0b7-c8e4e2756df9

### To Reproduce

Connect the FC to Betaflight Configurator, provide enough air flow to cool it, and wait. Displaying the Receiver or Sensors tab helps it to notice it quickly. It happens randomly taking less than a few seconds to more than 10 minutes but it happens 100%. Once, it happened during blackbox erase process. The configuration doesn't matter; it happens just after reflashing it with all defaults values and no devices attached. 

### Expected behavior

FC not freeze up.

### Support ID

```txt
49bb6052-610a-45ec-9a08-bb848e12eef8
```

### Flight controller

Diatone MK4 F722 Mini MPU6000 version, bought from Diatone Official Store on Aliexpress in July, 2023

### Other components

None.

### Add any other context about the problem that you think might be relevant here

I didn't get the Support ID when it had 4.4.2 and the ID above is from 4.4.0. Because the issue happens with no device attached and all default values, I guess it's okay not to include the configuration?

haslinghuis, 2023-08-07:
Please try 4.4.3 and 4.5 zulu

francis916, 2023-08-08:
> Please try 4.4.3 and 4.5 zulu
> 
> Also try latest [nightly configurator](https://github.com/betaflight/betaflight-configurator-nightlies/releases)

Thanks for your comment. I just checked and 4.4.3 isn't available for MambaF722_2022A yet. I'll try it when it's available. Thanks.

haslinghuis, 2023-08-08:
@francis916 it's here:

[image]

francis916, 2023-08-08:
> @francis916 it's here:
> 
> [image]

Oh, I see. Thanks!

haslinghuis, 2023-08-10:
Duplicate of #12924

closed 2023-08-10 by haslinghuis