betaflight/betaflight #13016: Unintentional dangorous GPS Rescue activation

SunjunKim, 2023-08-07:
### Describe the bug

Hi, I'm `sunjun` from the ELRS dev team, and we've got at least two case reports where the arming action caused the motor to spin unstoppably.  

Case 1)
From Darren, "Kwad dawk":
"My ELRS set up does not disarm every time perfectly as it should. About one in ten times or so, when I arm my drones the idle speed is about double the normal idle and no switch inputs will then work. Sticks will work but I can have the disarm switch in the off position and the drones will keep idling for an extended period of time. The fastest way to then disarm is to turn the transmitter off and they will always shut down then.

I have included a black box log where I managed to capture this happening on the last log where it kept running for while I moved the sticks around gently then eventually it realised its supposed to be disarmed and it finally stopped after about 19 second."

Case 2)
From a member of my local club:
"like 1/5 probability, the arming action causes a unusual motor spins at like 17% speed about 10+ seconds, and then disarmed by itself. Disarming with the switch doesn't work in the meantime."

He found that this was related to the GPS rescue setting, and I was able to find a reproducible step for the problem situation.

### To Reproduce

1. Prepare a quad with an ELRS RX attached.
1. PROPS OFF!!!
1. Set a GPS enabled.
1. Set a failsafe mode to GPS Rescue, with sanity check `Off` or `Failsafe only`. Allow `Arming without Fix`.
1. Set your ELRS RF mode low enough, like 150 Hz or 250 Hz (to ensure the RX connection is made after the BF startup).
1. Turn on your radio, and turn off the quad.
1. Turn on the quad. Make sure ELRS RX is not connected (ELRS RX is blinking) before the FC is fully booted.
1. As soon as you see the ELRS connection is made (carefully look at the RX LED to be solid), flick the arm switch (You should be quick! within like 0.5s).
1. Watch the idle RPM is higher than usual, and try to turn off the arm switch, which will not work.
1. To kill the quad, 
  - Wait for 19s then it'll stop the motor with a failsafe flag,
  - or, flick the roll/pitch stick hard,
  - or, turn off the transmitter forcefully.

Here's the video demonstration of the procedure and symptoms.
https://youtu.be/Ezm3wsmGP24

The motor will spin for 19s, with 17-18% motor power (cannot disarm by switch during this), which is exactly correspond to `RESCUE_DO_NOTHING` throttle level, which is `1275 - 100 = 1175`  (=17.5%). 
Related code I think: https://github.com/betaflight/betaflight/blob/dfef3bfb0e20a5a87e3df63e4425301ce19374d4/src/main/flight/gps_rescue.c#L244C9-L244C65 

The 19s period is probably related to the line https://github.com/betaflight/betaflight/blob/dfef3bfb0e20a5a87e3df63e4425301ce19374d4/src/main/flight/gps_rescue.c#L518

In the attched logs, which are captured during the symptom, Failsafe Phase is 6, which is `GPS_RESCUE` state. All the clues indicate an accidental activation of the gps rescue, somewhere on startup sequence.
[unintentional_gps_rescue_activation.zip](https://github.com/betaflight/betaflight/files/12282656/unintentional_gps_rescue_activation.zip)

Also here are the logs of the two situations: sticks are active during the (unintentional) gps rescue, and a big stick flick stopping the rescue.
[unintentional_gps_rescue_activation_2.zip](https://github.com/betaflight/betaflight/files/12283077/unintentional_gps_rescue_activation_2.zip)

### Expected behavior

The gps rescue mode should not be activated.

### Support ID

```txt
Id: "d0df103e-0415-486f-a8de-af6d0c2434bd"
```

### Flight controller

Aikon F7 3030 V1

### Other components

* BF v4.4.2
* GPS (Beitian BE-122)
* ELRS RX (BETAFPV Super-D) on v3.3
* ELRS TX (DIY Gemini TX)
* DJI O3

Also, here's the others' setup:

* Darren's
 - Mamba F7 stacks MK4
 - 3 slightly different ELRS x's being a Jumper EP1 rx & Happy model diversity RX with 3.2.1 firmware
 - Radio master ranger Micro 2.4 ghz TX in my X9D
 - Packet rate 50hz
 - Telem ratio Std (1:16)
 - Switch mode Wide
 - ELRS arm switch setup on AUX 1 and disarm set at ~1000 and arm at -2000 as recommended on a two position switch.

* case 2's
 - BF 4.4.1 and BF 4.4.2
 - T-motor stack and HGLRC stack
 - ELRS v3.2 TX-RX

### How are the different components wired up (including port information)

RX: UART 5
GPS: UART 1 (baud 57600)
O3: UART 3
(although I guess the port setup doesn't matter)

### Add any other context about the problem that you think might be relevant here

Setting up sanity check `On` prevents the problem, but not completly. When it's set `On` and with the described procedure, the motor spins just a fraction of second, then stops immediatly.

Disabling the "Arming without fix" option completely prevents the problem.

nerdCopter, 2023-08-08:
@ctzsnooze

ctzsnooze, 2023-08-14:
Thanks for the report.

Arming without fix is not recommended for users of GPS rescue, since initiating GPS rescue will cause an immediate disarm in this case.  Of course, if the user only intends to see the current position in the OSD or in a log, then they can take off sooner by enabling arming without fix, so it does have a reasonable application in some user cases.

The behaviour is not profoundly unsafe, unless the idle value was set too high, in which case the quad could lift off.  Otherwise, the user intended to arm the quad, and it armed.  The only inconvenience was the absence of control and the inability to disarm while waiting out the 20s timeout.  The landing phase should respond to impact detection, so a bump on the quad should stop it also.

The factors involved in causing the problem was that:
- the user set failsafe to GPS rescue while permitted arming without a home fix
- by permitting arming without a home fix, the GPS Rescue failsafe system activated very soon after arming, rather than waiting for the home fix to occur
- the user armed before a radio link existed, rather than waiting for the link to be established before arming (never a good idea)
- the quad entered failsafe because failsafe was set to GPS rescue, failsafe was active, and a failsafe condition existed
- failsafe for this kind of events includes 20s of landing mode with auto-disarm on impact, rather than immediate disarm.

If the user intends to use GPS Rescue, they should not, as indicated previously, enable arming without a fix, since if they enter failsafe, having taken off without a fix, the quad will not return home.

If the user only wants to see GPS position in the OSD or the Log, and want to take off without a fix, they should not set failsafe to GPS rescue, but to some other failsafe mode.

In 4.5 there are some changes which might help.  I would be grateful if they could be tested in the same test situation.  The intent is that if the quad enters failsafe within 5m of the arming point, it just disarms.  However, I think we will find that the presence of a home point is needed so that the quad knows it is so close to home that it must disarm rather than enter failsafe.  And in the user case described here, that would not solve this problem, since the quad was being armed without a home point, and it would not know whether it was 'too close'.

I'm not entirely sure how best to deal with this.

ctzsnooze, 2023-08-14:
I guess, in this situation of quick arming before the Rx link is established, you would not want GPS Rescue to cause an immediate disarm, any more than you would want it to enter 'do_nothing' mode.  So I don't think the solution lies in changing the behaviour of the GPS rescue code, but perhaps there should be a delay activating failsafe itself.

Could you please confirm that even if GPS rescue is not chosen as the intended failsafe mode, but instead, if failsafe is set to say 'drop', then the sequence you describe results in an immediate disarm?

If so, we need to consider delaying enabling failsafe until some period of time has elapsed after arming.

In the meantime, do not attempt to arm until the radio link is properly established.

ctzsnooze, 2023-08-14:
Can I clarify if the stage 1 failsafe time was modified?  Usually that time prevents initiation of failsafe unless the Rx loss exceeds the stage 1 failsafe time.  It seems very curious that ELRS can have a sufficient connection to permit arming, yet even after the stage 1 failsafe time expires, the failsafe system still thinks that no Rx link is present.

ctzsnooze, 2023-08-14:
Can I also clarify if the `pwr_on_arm_grace` period was modified?
By default, this is 5s.
The intent is to give enough time for the Rx link and for the CPU spikes associated with boot up, GPS and OSD connectivity to stabilise, before which arming is permitted.

ctzsnooze, 2023-08-14:
I've checked our code, and I think that the problem here is not unique to GPS Rescue.

Betaflight requires the Radio link to be solid and good for 1.0s before considering it to be 'up'.  Any attempt to arm while the link is down is considered a failsafe.  Hence any attempt to arm the quad within 1.0s of the link being 'up' will result in a failsafe.

Prior to ELRS, most Rx links connected quickly, typically stabilising well before the `pwr_on_arm_grace` period had elapsed.  So that even if the user powered up and tried to arm really quickly, they would never failsafe.

I have noticed that my ELRS receivers can take a long time to connect to the Tx, sometimes many seconds after powering up the quad. If, after finally getting the link up, I was to try and arm *within the 1.0s Rx stabilisation period*, I think this would trigger a failsafe. 

Mostly this failsafe would be 'ignored' since if the failsafe mode was 'drop', the quad would just disarm.  Likewise if GPS Rescue was enabled, and user required a home point, mostly the Rx link would be up for a long time before the home point was set, and there would be no problem; and in 4.5, if they did get a home point really fast, before the Rx link established itself, an early arm request would lead to an immediate disarm for being too close to home. However, if GPS Rescue is enabled, and the user doesn't require a home point, then the scenario reported here could still occur.

So although I think this event is quite common with ELRS, most users are unaware of it, noticing only a transient 'failsafe' indication in the OSD perhaps.  Only in the specific situation of GPS rescue being configured as failsafe, and the home point being not required before arming, and a certain timing of arming happening in relation to the establishment of the Rx link - all those things must coincide to even be aware of the issue.

ctzsnooze, 2023-08-14:
So at present ELRS users need to know they should not try to arm until at least 1.0s after the link is established.  This is a good habit to acquire for any Rx link.

ctzsnooze, 2023-08-14:
Please flash PR #13033 and confirm that it solves this problem, without causing other issues.  Thanks.

SunjunKim, 2023-08-14:
> Please flash PR #13033 and confirm that it solves this problem, without causing other issues. Thanks.

Thanks for all the comments. I'll take a look, test the PR, and reply back.

PS.
> So at present ELRS users need to know they should not try to arm until at least 1.0s after the link is established. This is a good habit to acquire for any Rx link.

I'm 1000% agree...

SunjunKim, 2023-08-14:
> Could you please confirm that even if GPS rescue is not chosen as the intended failsafe mode, but instead, if failsafe is set to say 'drop', then the sequence you describe results in an immediate disarm?

If it's set to "drop", the problem situation doesn't occur.

SunjunKim, 2023-08-14:
> thanks @SunjunKim - looking forward to finding out if PR 13033 solves the problem.

Nah.. I flashed #13033 and the problem still persists with the same procedure. Do you want me to have any test, or collect any log for you?

Here's my build log: https://[build.betaflight.com/api/builds/887d80e0cbf90a94dc7be350c4add650/log](https://build.betaflight.com/api/builds/887d80e0cbf90a94dc7be350c4add650/log) FYI to double check that I correctly flashed the PR's firmware.

ot0tot, 2023-08-15:
> 

FWIW, an ELRS receiver will connect nearly instantly if:
1) the TX is powered on before the RX, which is typically standard practice; and
2) if the receiver's init rate is set to the packet rate being used (see: https://github.com/ExpressLRS/ExpressLRS/pull/1757 for more)

SunjunKim, 2023-08-15:
> > 
> 
> FWIW, an ELRS receiver will connect nearly instantly if:
> 
> 1. the TX is powered on before the RX, which is typically standard practice; and
> 2. if the receiver's init rate is set to the packet rate being used (see: [Allow config of RX initial packet rate ExpressLRS/ExpressLRS#1757](https://github.com/ExpressLRS/ExpressLRS/pull/1757) for more)

I know it does help (and also I tested this trick works already). But most of our users just wait until the connection is established instead of set the intial packet rate... 😞

ctzsnooze, 2023-08-15:
@SunjunKim - I am grateful that you could test PR 13033.  And equally I'm disappointed that I didn't solve the problem!  :-)

As it happens, there was a second place in the failsafe code where the block on arming during signal loss was cleared immediately any Rx signal arrived.  

The reason, strangely enough, was an OSD message, `RXLOSS`.

This is a message which actually means, "arming is prevented because the link is considered bad".  It is shown on the OSD at 100ms of no signal (at the start of Stage 1 failsafe).  Previously it would go away immediately we received some valid Rx data (without any delay), because of a specific line of code that cleared it.  The sole purpose of the line of code wasn't to enable fast arming, it was because people complained about how long `RXLOSS` hung around for on the screen, even after just a transient loss of Rx signal.

The downside of clearing that message as soon as some new data arrived was that arming became possible immediately new data arrived.  That is perhaps good (in some situations, where you really want to arm it the instant signal arrives), and bad in other situations (where the Rx link is unreliable or weak and subsequent signals could be misinterpreted).

The consensus was to clear the message quickly, even though it permitted arming immediately the signal was restored - but that caused the problem that you identified in this issue.

In PR 13033 I've now removed that fast clearing behaviour.  `RXLOSS` will now stay on the OSD screen, and arming will be prevented, for the period of time set by `failsafe_recovery_delay` (which defaults to 10, meaning 1.0s).  

I am confident that this will definitely solve your issue, but the downside is that any link loss for more than 100ms will result in RXLOSS showing up on-screen for the entire 1s `failsafe_recovery_delay` period.

In a sense this is 'good', because it means that you should no arm now, or you may cause a failsafe; as soon as the message is gone, you can arm again.

If your preference is to arm very quickly after the Rx link arrives, and for the `RXLOSS` signal to only be very brief, try a value of 2, or 200ms, for the `failsafe_recovery_delay` period.  This will also restore pilot control after entering Stage 1 when only 200ms of continuous signal has been received, much more quickly than before.

Please let me know if this solves the problem, and experiment with the `failsafe_recovery_delay` period.

PS I also re-named the 'BADRX' message to 'NOT_DISARMED`.  That message comes on if the ARM switch is in the ARMED position when the signal is restored, and blocks re-arming until it is in the disarmed position.  I never understood what `BADRX` means

SunjunKim, 2023-08-16:
> Please let me know if this solves the problem, and experiment with the `failsafe_recovery_delay` period.

I confirmed #13033 fixes the problem, Solved! Good job.

closed 2023-09-01 by haslinghuis