betaflight/betaflight #13023: HGLRCF722 random freeze on 4.4.2 but ok on 4.3.0

attilafustos, 2023-08-10:
### Describe the bug

I updated my Rekon 5 quad to BF 4.4.2 HGLRCF722.
While working in the configurator (but also when flying) the FC randomly freezes.
How do I know it freezes ? The blue led is not blinking anymore and the configurator cannot communicate anymore with it.
Tested on 2 windows 10 laptops with 2 USB cables.
When flying, I pressed the arming button and nothing happened. Inspecting the FC showed the same behaviour: the led stopped blinking , FC dead. If I power cycle the FC then it works again.
I reverted to BF 4.3 and I could not reproduce it anymore.

### To Reproduce

Flash 4.4.2 on HGLRCF722 (Rekon 5).
Provided diff is prepared for step response logging.

### Expected behavior

Should not freeze or blue screen of death.

### Support ID

```txt
# 

# diff -all

# version
# Betaflight / STM32F7X2 (S7X2) 4.4.2 Jun  1 2023 / 04:48:19 (23d066d08) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name HGLRCF722
manufacturer_id HGLR

# name: Rekon 5

# feature
feature GPS
feature OSD

# serial
serial 1 2 115200 57600 0 115200
serial 3 1 115200 57600 0 115200

# mixer
mmix 0  1.000 -0.811  0.732 -1.000
mmix 1  1.000 -1.000 -0.732  1.000
mmix 2  1.000  0.811  0.732  1.000
mmix 3  1.000  1.000 -0.732 -1.000

# aux
aux 0 0 1 1700 2100 0 0
aux 1 1 0 1300 1700 0 0
aux 2 13 3 1700 2100 0 0
aux 3 26 3 1300 1700 0 0
aux 4 35 0 1700 2100 0 0

# rxfail
rxfail 3 s 1500
rxfail 4 s 1500

# master
set acc_calibration = 16,2,29,1
set baro_hardware = NONE
set blackbox_sample_rate = 1/2
set dshot_bidir = ON
set failsafe_procedure = GPS-RESCUE
set small_angle = 180
set gps_sbas_mode = EGNOS
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set gps_rescue_allow_arming_without_fix = ON
set deadband = 3
set osd_rssi_alarm = 30
set osd_cap_alarm = 1300
set osd_alt_alarm = 1000
set osd_tim2 = 1281
set osd_vbat_pos = 385
set osd_rssi_pos = 396
set osd_link_quality_pos = 2412
set osd_rssi_dbm_pos = 11
set osd_tim_1_pos = 54
set osd_tim_2_pos = 2433
set osd_remaining_time_estimate_pos = 354
set osd_flymode_pos = 409
set osd_throttle_pos = 385
set osd_vtx_channel_pos = 377
set osd_ah_pos = 200
set osd_current_pos = 406
set osd_mah_drawn_pos = 376
set osd_craft_name_pos = 362
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 2097
set osd_gps_lat_pos = 2065
set osd_gps_sats_pos = 2091
set osd_home_dir_pos = 2254
set osd_home_dist_pos = 2317
set osd_flight_dist_pos = 2456
set osd_compass_bar_pos = 353
set osd_altitude_pos = 18433
set osd_pid_roll_pos = 391
set osd_pid_pitch_pos = 391
set osd_pid_yaw_pos = 391
set osd_debug_pos = 289
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 2345
set osd_avg_cell_voltage_pos = 2444
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 41
set osd_disarmed_pos = 2081
set osd_nheading_pos = 311
set osd_esc_tmp_pos = 194
set osd_esc_rpm_pos = 83
set osd_esc_rpm_freq_pos = 312
set osd_core_temp_pos = 88
set osd_log_status_pos = 2060
set osd_stat_bitmask = 2373420
set osd_gps_sats_show_hdop = ON
set osd_displayport_device = MSP
set osd_canvas_width = 30
set osd_canvas_height = 13
set debug_mode = GYRO_SCALED
set craft_name = Rekon 5
set pilot_name = Attila Fustos

profile 0

# profile 0
set i_pitch = 25
set d_pitch = 20
set f_pitch = 0
set i_roll = 23
set d_roll = 17
set f_roll = 0
set i_yaw = 23
set f_yaw = 0
set angle_level_strength = 100
set d_min_roll = 17
set d_min_pitch = 20
set dyn_idle_min_rpm = 25
set simplified_i_gain = 30
set simplified_d_gain = 60
set simplified_dmax_gain = 0
set simplified_feedforward_gain = 0

rateprofile 0

# rateprofile 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 100
set pitch_rc_rate = 100
set yaw_rc_rate = 100
set roll_expo = 20
set pitch_expo = 20
set yaw_expo = 20
set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 80

# end the command batch
batch end

#
```

### Flight controller

HGLRCF722

### Other components

Default Rekon 5 from HGLRC (DJI)

haslinghuis, 2023-08-10:
Please flash using [#13022](https://github.com/betaflight/betaflight/pull/13022) in the select commit field

Why is baro disabled?

haslinghuis, 2023-08-10:
Duplicate of #12924

haslinghuis, 2023-08-10:
Made reference of this issue. Please continue in #12924

closed 2023-08-10 by haslinghuis