betaflight/betaflight #13024: RPM fails to start due to replacement of W25N01GVZEIG flash

ArtcocoX, 2023-08-10:
### Describe the bug

I replaced the flash of the mamba F722 APP MK4 flight controller with W25N01GVZEIG, which can correctly identify the capacity of 128mb, but in the recorded log, I found that the RPM filter was not enabled. After replacing it with W25Q128, the RPM filter was started correctly.

### To Reproduce

/

### Expected behavior

/

### Support ID

```txt
/
```

### Flight controller

mamba f722 app mk4

haslinghuis, 2023-08-10:
IIRC @SteveCEvans found the W25N01G [driver] having [some] problems.

Prajwal-Koirala, 2023-12-27:
@haslinghuis 
@SteveCEvans 

A huge thank you to everyone who smashed commits this year! Your dedication and hard work are truly appreciated.

haslinghuis, 2024-05-12:
Should be fixed in 4.5 with improved driver code.

closed 2024-05-12 by haslinghuis