betaflight/betaflight #13036: 4.5 nightly build 8/12/23 OSD elements disappear on punch outs

lemonyleprosy, 2023-08-15:
### Describe the bug

Average cell voltage and altitude disappear from osd on hard throttle punches.
This is with a NeutronRC AT32F435 35A aio, Walksnail 1s light vtx, Walksnail Pro cam, 4s, 1506 4300kv motors.

### To Reproduce

With a v1 NeutronRC AT32F435 35A aio, hit the throttle hard on 4s.

### Expected behavior

I expect altitude and average cell voltage to be visible on osd regardless of throttle position.

### Support ID

```txt
# # Problem description

# # Altitude and average cell voltage disappear from osd on throttle punch outs

# ###

# version
# Betaflight / AT32F435G (A435) 4.5.0 Aug 13 2023 / 05:29:01 (a5791814c) MSP API: 1.46
# config rev: fa51239
# board: manufacturer_id: NERC, board_name: NEUTRONRCF435MINI

# status
MCU AT32F435 Clock=288MHz, Vref=3.31V, Core temp=50degC
Stack size: 2048, Stack address: 0x2002fff0
Configuration: CONFIGURED, size: 3384, max available: 16384
Devices detected: SPI:1, I2C:1
Gyros detected: gyro 1 locked dma
GYRO=BMI270, ACC=BMI270, BARO=DPS310
OSD: MSP (30 x 13)
BUILD KEY: 8e4337b361c963b79c7e6529b4269d17 (4.5.0-zulu)
System Uptime: 92 seconds, Current Time: 2023-08-15T04:34:42.672+00:00
CPU:14%, cycle time: 318, GYRO rate: 3144, RX rate: 15, System rate: 9
Voltage: 5 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
FLASH: JEDEC ID=0x00ef4018 16M
GPS: NOT ENABLED
Arming disable flags: RXLOSS CLI MSP RPMFILTER

# flash_info
Flash sectors=256, sectorSize=65536, pagesPerSector=256, pageSize=256, totalSize=16777216 JEDEC ID=0x00ef4018
Partitions:
  0: FLASHFS   0 255
FlashFS size=16777216, usedSize=16777216

# dump hardware

# version
# Betaflight / AT32F435G (A435) 4.5.0 Aug 13 2023 / 05:29:01 (a5791814c) MSP API: 1.46
# config rev: fa51239

# start the command batch
batch start

board_name NEUTRONRCF435MINI
manufacturer_id NERC

# resources
resource BEEPER 1 C15
resource MOTOR 1 B06
resource MOTOR 2 B07
resource MOTOR 3 A03
resource MOTOR 4 B01
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B11
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 B09
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 B04
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 B00
resource SERIAL_RX 3 B10
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 B08
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 B03
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource I2C_SCL 1 NONE
resource I2C_SCL 2 H02
resource I2C_SCL 3 NONE
resource I2C_SCL 4 NONE
resource I2C_SDA 1 NONE
resource I2C_SDA 2 H03
resource I2C_SDA 3 NONE
resource I2C_SDA 4 NONE
resource LED 1 C13
resource LED 2 C14
resource LED 3 NONE
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 NONE
resource SPI_SCK 4 NONE
resource SPI_SDI 1 A06
resource SPI_SDI 2 B14
resource SPI_SDI 3 NONE
resource SPI_SDI 4 NONE
resource SPI_SDO 1 A07
resource SPI_SDO 2 B15
resource SPI_SDO 3 NONE
resource SPI_SDO 4 NONE
resource ADC_BATT 1 A00
resource ADC_RSSI 1 NONE
resource ADC_CURR 1 A01
resource ADC_EXT 1 NONE
resource BARO_CS 1 NONE
resource BARO_EOC 1 NONE
resource BARO_XCLR 1 NONE
resource COMPASS_CS 1 NONE
resource COMPASS_EXTI 1 NONE
resource PINIO 1 NONE
resource PINIO 2 NONE
resource PINIO 3 NONE
resource PINIO 4 NONE
resource USB_MSC_PIN 1 NONE
resource FLASH_CS 1 B05
resource OSD_CS 1 B12
resource GYRO_EXTI 1 A15
resource GYRO_EXTI 2 NONE
resource GYRO_CS 1 A04
resource GYRO_CS 2 NONE
resource USB_DETECT 1 NONE
resource PULLUP 1 NONE
resource PULLUP 2 NONE
resource PULLUP 3 NONE
resource PULLUP 4 NONE
resource PULLDOWN 1 NONE
resource PULLDOWN 2 NONE
resource PULLDOWN 3 NONE
resource PULLDOWN 4 NONE

# timer
timer A08 AF1
# pin A08: TIM1 CH1 (AF1)
timer B06 AF2
# pin B06: TIM4 CH1 (AF2)
timer B07 AF2
# pin B07: TIM4 CH2 (AF2)
timer A03 AF1
# pin A03: TIM2 CH4 (AF1)
timer B01 AF2
# pin B01: TIM3 CH4 (AF2)

# dma
dma SPI_SDO 1 NONE
dma SPI_SDO 2 NONE
dma SPI_SDO 3 NONE
dma SPI_SDO 4 NONE
dma SPI_SDI 1 NONE
dma SPI_SDI 2 NONE
dma SPI_SDI 3 NONE
dma SPI_SDI 4 NONE
dma SPI_TX 1 NONE
dma SPI_TX 2 NONE
dma SPI_TX 3 NONE
dma SPI_TX 4 NONE
dma SPI_RX 1 NONE
dma SPI_RX 2 NONE
dma SPI_RX 3 NONE
dma SPI_RX 4 NONE
dma ADC 1 11
# ADC 1: DMA2 Channel 0 Channel 5
dma ADC 2 NONE
dma ADC 3 NONE
dma UART_TX 1 NONE
dma UART_TX 2 NONE
dma UART_TX 3 NONE
dma UART_TX 4 NONE
dma UART_TX 5 NONE
dma UART_TX 6 NONE
dma UART_TX 7 NONE
dma UART_TX 8 NONE
dma UART_TX 9 NONE
dma UART_TX 10 NONE
dma UART_TX 11 NONE
dma UART_RX 1 NONE
dma UART_RX 2 NONE
dma UART_RX 3 NONE
dma UART_RX 4 NONE
dma UART_RX 5 NONE
dma UART_RX 6 NONE
dma UART_RX 7 NONE
dma UART_RX 8 NONE
dma UART_RX 9 NONE
dma UART_RX 10 NONE
dma UART_RX 11 NONE
dma pin A08 7
# pin A08: DMA2 Channel 0 Channel 42
dma pin B06 0
# pin B06: DMA1 Channel 0 Channel 67
dma pin B07 1
# pin B07: DMA1 Channel 0 Channel 68
dma pin A03 2
# pin A03: DMA1 Channel 0 Channel 59
dma pin B01 3
# pin B01: DMA1 Channel 0 Channel 64

# feature
feature -RX_PPM
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -RANGEFINDER
feature -TELEMETRY
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -OSD
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -RX_SPI
feature -ESC_SENSOR
feature -ANTI_GRAVITY
feature RX_SERIAL
feature TELEMETRY
feature LED_STRIP
feature OSD
feature AIRMODE
feature ANTI_GRAVITY

# serial
serial 20 1 115200 57600 0 115200
serial 0 64 115200 57600 0 115200
serial 1 131073 115200 57600 0 115200
serial 2 0 115200 57600 0 115200
serial 4 0 115200 57600 0 115200
serial 6 8192 115200 57600 0 115200

# master
set gyro_to_use = FIRST
set align_mag = DEFAULT
set mag_align_roll = 0
set mag_align_pitch = 0
set mag_align_yaw = 0
set mag_bustype = I2C
set mag_i2c_device = 0
set mag_i2c_address = 0
set mag_spi_device = 0
set baro_bustype = I2C
set baro_spi_device = 0
set baro_i2c_device = 2
set baro_i2c_address = 0
set adc_device = 1
set blackbox_device = SPIFLASH
set motor_kv = 1960
set dshot_bitbang = AUTO
set dshot_bitbang_timer = AUTO
set dshot_telemetry_start_margin = 10
set current_meter = ADC
set battery_meter = ADC
set vbat_scale = 110
set ibata_scale = 100
set beeper_inversion = ON
set beeper_od = OFF
set beeper_frequency = 0
set max7456_clock = NOMINAL
set max7456_spi_bus = 2
set max7456_preinit_opu = OFF
set led_inversion = 3
set pinio_config = 1,1,1,1
set pinio_box = 255,255,255,255
set usb_msc_pin_pullup = ON
set flash_spi_bus = 2
set gyro_1_bustype = SPI
set gyro_1_spibus = 1
set gyro_1_i2cBus = 0
set gyro_1_i2c_address = 0
set gyro_1_sensor_align = CW0
set gyro_1_align_roll = 0
set gyro_1_align_pitch = 0
set gyro_1_align_yaw = 0
set gyro_2_bustype = NONE
set gyro_2_spibus = 0
set gyro_2_i2cBus = 0
set gyro_2_i2c_address = 0
set gyro_2_sensor_align = DEFAULT
set gyro_2_align_roll = 0
set gyro_2_align_pitch = 0
set gyro_2_align_yaw = 0
set scheduler_relax_rx = 25
set scheduler_relax_osd = 25
set box_user_1_name = -
set box_user_2_name = -
set box_user_3_name = -
set box_user_4_name = -

# end the command batch
batch end

*** Support data submitted *** 
Id: "563cb0ce-f102-45cb-a974-21f5d1c7f288"

# copy ID and provide to the betaflight team.
```

### Flight controller

NeutronRC AT32F435 25A aio

### Other components

Walksnail 1s lite vtx, Walksnail Pro cam, ELRS rx v3

### How are the different components wired up (including port information)

RX- uart1
VTX- uart2

SteveCEvans, 2023-08-23:
How are you powering the VTX? I suggesting sticking a large cap on it's power input.

lemonyleprosy, 2023-08-23:
It is a 1s Walksnail vtx, being powered by the built in 5v 2.5A bec on the AT32F435 35A aio.

haslinghuis, 2023-08-23:
> AT32F435 35A aio

which board?

lemonyleprosy, 2023-08-23:
NeutronRC AT32F435 35A aio, v1aliexpress.usWalksnail vtx is a v2 1s lite.

Lckythr33, 2023-09-08:
Connect vtx ground to camera ground to fc ground. The problem is that if you use a filtered output for camera power, the ground won’t be the same as the FC ground, so the osd might not detect a video signal.

lemonyleprosy, 2023-09-08:
…thanks, but, this is a 1s Walksnail Avatar v2 VTX and a Walksnail Avatar HD Pro camera.Not analog.They are connected via a mipi cable. I mean, if necessary, I can open up the camera, find a ground pad, and connect it to the FC where the vtx ground is connected, but, that seems like something a typical user wouldn’t be capable of doing?We’re not talking about getting rid of video noise. The digital vtx is powered via the 5v bec. The digital vtx powers the cam via the mipi cable.Osd in this case is handled via msp over uart, no?

haslinghuis, 2023-09-08:
Has WS been updated to latest beta firmware as we have new defaults for OSD fonts?

Details: https://github.com/betaflight/betaflight/pull/13005

haslinghuis, 2023-09-08:
If using stable release you can check with

    set ‎displayport_msp_fonts = 0,0,0,0
    save

lemonyleprosy, 2023-09-10:
Oh, snap.

Running the cli command:

set ‎displayport_msp_fonts = 0,0,0,0

Fixed it.

Average cell voltage and altitude are no longer disappearing from osd no matter how hard I hit the throttle or how low I run my battery down. 

Thanks folks. Apparently this wasn’t a bug and was just a user lack of knowledge. You can close this out.

> On Sep 8, 2023, at 10:35 AM, Mark Haslinghuis ***@***.***> wrote:
> 
> set ‎displayport_msp_fonts = 0,0,0,0

closed 2023-09-10 by haslinghuis