betaflight/betaflight #13037: Kakute H7 HDV Stack ESC does take long to start up using Betaflight 4.4.2

Mane660, 2023-08-15:
### Describe the bug

i have following problem.

My drone does take a long time to start up the ESC.
Sometimes it takes longer than 20s and sometimes much more longer.

But when the FC is plugged and connected to Betaflight Configurator, and i plug in the battery the ESC start up directly.

When disconnected I try to arm the drone, the OSD says "ARMED" and it should be ready.
But the ESC is not started up and so the Motors does not spin.

### To Reproduce

I did not have this problem on older betaflight versions. But I did an update to get the full funcionalitys like OSD (HD), and some more things.

Did also try to flash 4.3 but there was no difference.

### Expected behavior

plugging in the drone to the battery.
Normal beeper beeps, afterwards after some long time, the FC/ESC seems to reboot and there is the typical ESC beeping

Cloud-Build Details: https://build.betaflight.com/api/builds/fa7f2d1d3d226869ab329d7460b6f3f8/log
[image]
[image]

### Support ID

```txt
e5e34d82-b3ca-44cc-b81d-729e62a054a3
```

### Flight controller

Holybro Kakute F7 HDV

### How are the different components wired up (including port information)

GPS, CRSF, Vista, Beeper/LED

Mane660, 2023-08-15:
https://github.com/betaflight/betaflight/issues/7706

found also this issue maybe the same problem but on newer versions?

SteveCEvans, 2023-08-15:
What ESC firmware are you running?

Mane660, 2023-08-15:
[image]
updated to Rev. 32.9 before it was 32.7 but this did not fix the problem

Mane660, 2023-08-19:
So new information:

Tried to flash BF 4.2.0 onto the flight controller.
Disconnected from USB connected to battery. Direct boot up of the ESC hearing the 2 beeps at the end.

With BF 4.4.x it does not work. Would like to use the newest 4.4.x because of the HD OSD option for DJI. Hope someone can help.

haslinghuis, 2023-08-19:
Did you already try testing with 4.4.3-zulu as it contains some [fixes](https://github.com/betaflight/betaflight/pulls?q=is%3Apr+milestone%3A4.4.3+is%3Aclosed)

Mane660, 2023-08-19:
> Did you already try testing with 4.4.3-zulu as it contains some [fixes](https://github.com/betaflight/betaflight/pulls?q=is%3Apr+milestone%3A4.4.3+is%3Aclosed)

does not work since BF > 4.3.x 
BF 4.2.11 is the last one working

haslinghuis, 2023-08-19:
Works on SPRacing H7EXTREME (4.5-zulu)

[image]

haslinghuis, 2023-08-19:
@Mane660 Did you try a lower PWM frequency?

Mane660, 2023-08-19:
> Works on SPRacing H7EXTREME (4.5-zulu)
> 
> [image]
> 
> [image]

But am I able to flash this on my Kakute H7 HDV? because it seems to be a different board. 
[image]

haslinghuis, 2023-08-19:
Asking for you try to flash the ESC with 24 or 48Khz PWM frequency.

[image]

Mane660, 2023-08-19:
[image]
i activated the Bidirectional DShot as in your screenshot now the ESC boots up in 4.4.2

closed 2023-09-27 by github-actions[bot]