betaflight/betaflight #13042: Backtracking as an option for stage 2 of Failsafe

pegurov, 2023-08-17:
### Is your feature request related to a problem? Please describe

GPS Rescue is awesome, but in these scenarios it is not going to return the quad home:
1. Flying behind a tall structure, like a skyscraper. After signal loss the quad is going to gain some altitude, rotate and then try to get back to home. And crash into the structure, as it is taller than the gained altitude.
2. Flying under something, like a bridge. When trying to gain altitude, it is going to crash into the bridge.

### Describe the solution you'd like

I would like to see a 'Backtracking' feature as an alternative to GPS Rescue in stage 2 of Failsafe.

I suppose it should be possible to track 'breadcrumbs' during the flight, every N milliseconds, like a combo of altitude + GPS coords. And in case of a Failsafe event try to fly backward on a path, very close to the path in 3D space, that was taken to get to the point, where Failsafe happened.

This feature would probably result in safer return path, and quick signal recovery, as the quad is going to go back to the point where we still had signal, which should be pretty close to the point, where failsafe happened.

### Describe alternatives you've considered

None

### Other information

I have extensive iOS development experience (10 years), I could probably try to implement this feature myself)

ZzyzxTek, 2023-08-17:
Logging the breadcrumbs might be possible. Betaflight already does blackbox logging of GPS data, maybe that could be leveraged for this. But many FCs have limited flash (or no flash) to use for logging.

The challenge is that Betaflight does not do autonomous flight. GPS rescue is a very specialized type of autonomous flight that heads back to a single home point. This would require a more generalized waypoint flight to follow the breadcrumbs.

If waypoint flight ever does get implemented, it's a ways off in the future, there are lots of other support features needed first on the way to that. At this point, we're just getting the GPS driver more solid, and working on better compass integration.

pegurov, 2023-08-18:
Thoughts on limited flash memory:
1. Maybe it is possible to allow this feature only when it is known that there is enough memory to do that? Is there a way to figure it out of the code, running on FC? Or just add a warning in Betaflight configurator, that this feature is only available when the FC has enough flash storage.
2. Maybe we could implement a combo of bactracking and GPS Rescue, where the craft follows breadcrumbs back for some N meters, and then goes into GPS Rescue mode. That way we would only need to store a limited amount of breadcrumbs for the last N meters. Maybe that could just be an option for GPS rescue, like a switch that enables backtracking + an input, that allows you to specify the N.

Thoughts on waypoint flight:
1. Could you please be a little more specific, as to what exactly is needed for waypoint flying? I'm guessing there needs to be some logic, that is able to get from A to B in 3D space, first in ideal conditions (no wind), and then being able to correct itself against wind. Sounds like a challenge, but totally doable.
2. I do not really see, how getting GPS drivers more solid conflicts with waypoint flying, looks like more solid GPS performance only enhances the overall quality of GPS rescue and backtracking. 

As I said above, I would be willing to try and implement this myself, just need some kind of approval from the repo maintainers) I have tried to figure out the process of adding new features in the repo docs, but it is rather focused on code style and testing.

haslinghuis, 2023-08-18:
Right now we are in feature freeze. Some requirements:

- Integration of barometer with GPS: https://github.com/betaflight/betaflight/pull/11958
- Improvements for compass: https://github.com/betaflight/betaflight/pull/12998
- Position hold: https://github.com/betaflight/betaflight/pull/11342
- Wind estimation. See also https://github.com/betaflight/betaflight/pull/12792
- Way points. https://github.com/betaflight/betaflight/pull/8413
- Backtracking.

ledvinap, 2023-08-18:
IMO only reasonable way to handle this is to port it from iNav. Either iNav already supports it, or implement it in iNav first.

haslinghuis, 2023-08-19:
https://link.springer.com/content/pdf/10.1007/s00521-020-05097-x.pdf

haslinghuis, 2023-08-19:
[s00521-020-05097-x.pdf](https://github.com/betaflight/betaflight/files/12386694/s00521-020-05097-x.pdf)

sudoTheBotFather, 2023-08-21:
Just use Inav, it handles this beautifully right now and it based for autonomous flight / advanced rescue modes.

Prajwal-Koirala, 2023-12-29:
This GitHub issue proposes a new feature for Betaflight called "Backtracking," which would be an alternative to the current Failsafe rescue mode. Instead of flying the quadcopter back to its home point in a straight line, Backtracking would retrace the quad's path back to the Failsafe event location, making it potentially safer near obstacles.

The author suggests recording the quad's position at regular intervals during flight and then following that path back in case of a Failsafe. However, concerns exist about feasibility due to limited flash memory and the lack of waypoint flight implementation in Betaflight. The author proposes making Backtracking optional or limiting its distance to address these concerns.

Ultimately, the decision rests with the Betaflight maintainers. While they're currently focused on improving GPS and compass performance, they're open to Backtracking and might consider it in the future.

I hope this summary is helpful! Let me know if you have any other questions.

ctzsnooze, 2024-04-06:
Closing, we don't do this at the moment.  Use iNav for this.

AirCombat, 2025-04-15:
This could be made a lot simpler to just use dead-reckoning rather than some precision navigation... turn 180, trevel for 5 seconds with a little extra throttle, if signal does not return failsafe. As a heli pilot, if we fly into a cloud we will do a 180 and use dead reckoning. In smaller aircraft: just a artificial horizon and altimiter.

closed 2026-09-26 by blckmn