betaflight/betaflight #13043: Battery Capacity is not reset when FC is powered from the USB and we change batteries

cptX, 2023-08-19:
### Describe the bug

When connected to usb (I use power bank between battery changes to preserve GPS satellites) changing battery doesn't reset the capacity (mAh). This leads to making the capacity measurement for the next batteries unusable!!
I have connected the GPS Power supply to the 5V of the FC, and in the field when I change batteries I connect a power bank to the USB to preserve the satellites. This bug reported here doesn't reset the capacity, thus when a new battery is connected the old capacity is already shown, making the measurement unusable...

### To Reproduce

1. Connect the USB to a computer and open Betaflight Configurator
2. Connect a battery to the drone.
3. Let the drone drain some battery, note the mAh used.
4. Disconnect the battery
5. Wait some time (a couple of minutes)
6. Reconnect the battery. The mAh are remembered and counting from the last value before disconnection.

### Expected behavior

Capacity (mAh) should be reset after the FC detects that no battery is connected. With the current implementation that the capacity is remembered, connecting a new battery shows the capacity used from the previous battery, thus making the measurement useless.

### Support ID

```txt
fb283029-6d04-49c2-8b4a-9128adff0e58
```

### Flight controller

Omnibus F4 v3

### How are the different components wired up (including port information)

GPS is connected to the 5V of the FC, so when FC is powered from the USB the GPS is also powered. I'm using a powerbank at the field to preserve the Satellites when changing batteries.

klutvott123, 2023-09-01:
There are probably not many people doing the same procedure as you to preserve satellite count, so using complete power loss to reset consumed mAh is fine for most, but detecting a battery disconnect could be a good idea. Feel free to make a PR if you want 🙂

ExplodingFist, 2023-09-11:
I use a lipo to 5v USB step down for the purpose of obtaining and maintaining sat locks on Naked Vista/O3 boards where sitting idle would cause an overheat. This would be an improvement in this situation as it would allow you to plugin USB power, unplug/plug lipo, disconnect USB power and take off right away without worrying about losses in locks at all. Given the improvements on Betaflight return to home, and the importance of locks to establish the home position at time of arm, this should be considered.

Prajwal-Koirala, 2023-12-29:
This is an issue about the Betaflight flight controller software not resetting the battery capacity when the battery is changed while the FC is powered from USB. This can make the capacity measurement for the next batteries unusable. The user suggests that the software should be changed to reset the battery capacity when the FC detects that no battery is connected. Some other users have also reported this issue and suggested that it would be a useful improvement.

I hope this is helpful!