betaflight/betaflight #13049: No telemetry after upgrade to 4.4.2 on IFLIGHT_SUCCEX_E_F4

tihonovS, 2023-08-25:
### Describe the bug

Hi, having upgraded from 4.3.2 to 4.4.2, I have lost telemetry in my Tango2. Core only mod did not help.
I deleted all the sensors on my tango 2 and tried to search again, but it didn't do anything.
After downgrate to version 4.3.2, everything works fine.

### To Reproduce

upgraded from 4.3.2 to 4.4.2 on IFLIGHT_SUCCEX_E_F4

### Expected behavior

betaflight will broadcast telemetry

### Support ID

```txt
d919ee56-0af8-4091-97f1-92cbfba7ae29
```

### Flight controller

IFLIGHT_SUCCEX_E_F4

klutvott123, 2023-09-01:
Could be an old configurator bug that disabled telemetry when making changes in the ports tab.  Check the receiver tab to se if telemetry is still enabled. If not, just enable it again

tihonovS, 2023-09-01:
I checked the receiver tab telemetry is on. Turned off, then on, still no telemetry. Do you have any idea what else could be the problem?

klutvott123, 2023-09-01:
No. We'll have to look into it. I'll see if I can reproduce when I have the time

haslinghuis, 2023-09-01:
Check wiring for ch 2

[image]

klutvott123, 2023-09-02:
and post a diff please

tihonovS, 2023-09-03:
ch 2 is ok. 
`# diff all

# version
# Betaflight / STM32F405 (S405) 4.4.2 Jun 18 2023 / 15:07:25 (23d066d08) MSP API: 1.45

# config: YES

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name IFLIGHT_SUCCEX_E_F4
manufacturer_id IFRC
mcu_id 003700395853500620303758
signature 

# name: XL5 V5

# feature
feature -LED_STRIP
feature -DISPLAY
feature SOFTSERIAL
feature GPS
feature AIRMODE

# serial
serial 0 131073 115200 57600 0 115200
serial 3 2 115200 115200 0 115200

# beeper
beeper -GYRO_CALIBRATED
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# beacon
beacon RX_SET

# map
map TAER1234

# aux
aux 0 0 0 1650 2100 0 0
aux 1 46 3 1600 2100 0 0
aux 2 13 5 1675 2100 0 0
aux 3 26 1 1675 2100 0 0
aux 4 35 2 1750 2100 0 0

# rxfail
rxfail 6 s 1800

# master
set gyro_lpf1_static_hz = 0
set gyro_lpf2_static_hz = 0
set dyn_notch_min_hz = 90
set dyn_notch_max_hz = 800
set gyro_lpf1_dyn_min_hz = 0
set gyro_lpf1_dyn_max_hz = 0
set acc_lpf_hz = 10
set acc_calibration = 4,15,206,1
set min_check = 1000
set rc_smoothing_auto_factor = 175
set rc_smoothing_auto_factor_throttle = 100
set rc_smoothing_setpoint_cutoff = 12
set rc_smoothing_feedforward_cutoff = 12
set rc_smoothing_throttle_cutoff = 20
set blackbox_sample_rate = 1/2
set failsafe_switch_mode = STAGE2
set failsafe_procedure = GPS-RESCUE
set bat_capacity = 1500
set vbat_max_cell_voltage = 436
set vbat_scale = 112
set beeper_dshot_beacon_tone = 2
set yaw_motors_reversed = ON
set small_angle = 180
set gps_sbas_mode = AUTO
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set gps_rescue_min_start_dist = 20
set gps_rescue_alt_mode = FIXED_ALT
set gps_rescue_return_alt = 60
set gps_rescue_max_angle = 40
set gps_rescue_min_sats = 5
set yaw_deadband = 5
set pid_process_denom = 2
set simplified_gyro_filter_multiplier = 120
set osd_warn_bitmask = 49151
set osd_vbat_pos = 1517
set osd_rssi_pos = 2498
set osd_rssi_dbm_pos = 2466
set osd_flymode_pos = 2581
set osd_crosshairs_pos = 185
set osd_ah_sbar_pos = 207
set osd_ah_pos = 90
set osd_current_pos = 3629
set osd_mah_drawn_pos = 3469
set osd_wh_drawn_pos = 267
set osd_gps_speed_pos = 2076
set osd_gps_lon_pos = 3080
set osd_gps_lat_pos = 3112
set osd_gps_sats_pos = 2062
set osd_home_dir_pos = 2073
set osd_home_dist_pos = 2067
set osd_altitude_pos = 2562
set osd_warnings_pos = 14421
set osd_avg_cell_voltage_pos = 3533
set osd_battery_usage_pos = 36205
set osd_disarmed_pos = 2613
set osd_core_temp_pos = 1128
set osd_displayport_device = MSP
set debug_mode = GYRO_SCALED
set vcd_video_system = HD
set gyro_1_align_yaw = 1800
set rpm_filter_q = 1000
set rpm_filter_fade_range_hz = 100
set craft_name = XL5 V5
set pilot_name = Sergey Tihonov

profile 0

# profile 0
set dterm_lpf1_dyn_min_hz = 80
set dterm_lpf1_dyn_max_hz = 110
set dterm_lpf1_static_hz = 0
set dterm_lpf2_static_hz = 0
set anti_gravity_gain = 100
set yaw_lowpass_hz = 0
set p_pitch = 77
set i_pitch = 117
set d_pitch = 39
set f_pitch = 168
set p_roll = 64
set i_roll = 97
set d_roll = 35
set f_roll = 140
set p_yaw = 64
set i_yaw = 97
set f_yaw = 140
set d_min_roll = 35
set d_min_pitch = 39
set motor_output_limit = 75
set feedforward_smooth_factor = 35
set feedforward_jitter_factor = 12
set simplified_master_multiplier = 130
set simplified_i_gain = 85
set simplified_d_gain = 90
set simplified_pi_gain = 110
set simplified_dmax_gain = 0
set simplified_feedforward_gain = 90
set simplified_pitch_pi_gain = 115
set simplified_dterm_filter = OFF
set simplified_dterm_filter_multiplier = 140

profile 1

profile 2

profile 3

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set thr_mid = 0
set roll_rc_rate = 2
set pitch_rc_rate = 2
set yaw_rc_rate = 2
set roll_expo = 40
set pitch_expo = 40
set yaw_expo = 40
set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 60

rateprofile 1

rateprofile 2

rateprofile 3

# restore original rateprofile selection
rateprofile 0

# save configuration`

klutvott123, 2023-09-03:
can you show us a screenshot of the ports tab? serial setup looks strange

tihonovS, 2023-09-03:
[image]

klutvott123, 2023-09-03:
@tihonovS Looks good to me. Can you please try 4.4.1?

tihonovS, 2023-09-04:
Can you please help me with the installation of 4.4.1?
[image]

haslinghuis, 2023-09-04:
4.4.1 has been withdrawn.
You can test 4.4.0 or 4.4.3-zulu and 4.5-zulu

tihonovS, 2023-09-04:
4.3.2 - ok
4.4.0 - ok
4.4.2 - not ok
4.4.3-zulu - not ok
4.5-zulu - not ok

Something's broken in 4.4.2

klutvott123, 2023-09-04:
@tihonovS Thanks for testing 👍

tihonovS, 2023-09-04:
Is there anything else I can check to help you?

klutvott123, 2023-09-04:
@tihonovS If you have a multimeter you could try measuring the voltage between ground and the TX pin your receiver is connected to on the FC

tihonovS, 2023-09-06:
@klutvott123  sorry for the long answer. 
ground -> TX pin: 1.2V
ground -> RX pin: 3.3V

haslinghuis, 2023-09-06:
TBS Sixty9 works on 4.5

However if I move TBS TX module from Taranis X9D+ (OpenTx) to Radiomaster TX16S (EdgeTx) there is no stick input. Binding is no problem.

tmunsch, 2023-09-09:
Any word on this? I find this very intriguing as I will be updating a buddies quad to a similar configuration and would like to avoid any headaches if possible.

haslinghuis, 2023-09-09:
What RF profile is being used? 50Hz, 150Hz or dynamic power? Dynamic power is not recommended.

klutvott123, 2023-09-09:
1.2V could be floating or just an average of communication. Hard to say without a scope. This issue could be caused by the newly introduced uart TX monitoring.

tihonovS, 2023-09-12:
Strange, I can connect to FC and everything works except telemetry which I don't get on my tango 2.

dang-itt, 2023-09-12:
> Strange, I can connect to FC and everything works except telemetry which I don't get on my tango 2.

Maybe uart issues or something. I quickly realized I didn't need the tx power in OSD XD I'll stick with 4.2

howels, 2023-09-19:
> ch 2 is ok. Strange that the "# feature" section has no telemetry flag
> 
> ```
> # diff all
> 
> # version
> # Betaflight / STM32F405 (S405) 4.4.2 Jun 18 2023 / 15:07:25 (23d066d08) MSP API: 1.45
> 
> # config: YES
> 
> # start the command batch
> batch start
> 
> # reset configuration to default settings
> defaults nosave
> 
> board_name IFLIGHT_SUCCEX_E_F4
> manufacturer_id IFRC
> mcu_id 003700395853500620303758
> signature 
> 
> # name: XL5 V5
> 
> # feature
> feature -LED_STRIP
> feature -DISPLAY
> feature SOFTSERIAL
> feature GPS
> feature AIRMODE
> 
> # serial
> serial 0 131073 115200 57600 0 115200
> serial 3 2 115200 115200 0 115200
> 
> # beeper
> beeper -GYRO_CALIBRATED
> beeper -ARMING_GPS_FIX
> beeper -GPS_STATUS
> beeper -RX_SET
> beeper -ACC_CALIBRATION
> beeper -ACC_CALIBRATION_FAIL
> beeper -READY_BEEP
> beeper -DISARM_REPEAT
> beeper -ARMED
> beeper -SYSTEM_INIT
> beeper -ON_USB
> beeper -BLACKBOX_ERASE
> beeper -CRASH_FLIP
> beeper -CAM_CONNECTION_OPEN
> beeper -CAM_CONNECTION_CLOSE
> beeper -RC_SMOOTHING_INIT_FAIL
> 
> # beacon
> beacon RX_SET
> 
> # map
> map TAER1234
> 
> # aux
> aux 0 0 0 1650 2100 0 0
> aux 1 46 3 1600 2100 0 0
> aux 2 13 5 1675 2100 0 0
> aux 3 26 1 1675 2100 0 0
> aux 4 35 2 1750 2100 0 0
> 
> # rxfail
> rxfail 6 s 1800
> 
> # master
> set gyro_lpf1_static_hz = 0
> set gyro_lpf2_static_hz = 0
> set dyn_notch_min_hz = 90
> set dyn_notch_max_hz = 800
> set gyro_lpf1_dyn_min_hz = 0
> set gyro_lpf1_dyn_max_hz = 0
> set acc_lpf_hz = 10
> set acc_calibration = 4,15,206,1
> set min_check = 1000
> set rc_smoothing_auto_factor = 175
> set rc_smoothing_auto_factor_throttle = 100
> set rc_smoothing_setpoint_cutoff = 12
> set rc_smoothing_feedforward_cutoff = 12
> set rc_smoothing_throttle_cutoff = 20
> set blackbox_sample_rate = 1/2
> set failsafe_switch_mode = STAGE2
> set failsafe_procedure = GPS-RESCUE
> set bat_capacity = 1500
> set vbat_max_cell_voltage = 436
> set vbat_scale = 112
> set beeper_dshot_beacon_tone = 2
> set yaw_motors_reversed = ON
> set small_angle = 180
> set gps_sbas_mode = AUTO
> set gps_auto_baud = ON
> set gps_ublox_use_galileo = ON
> set gps_rescue_min_start_dist = 20
> set gps_rescue_alt_mode = FIXED_ALT
> set gps_rescue_return_alt = 60
> set gps_rescue_max_angle = 40
> set gps_rescue_min_sats = 5
> set yaw_deadband = 5
> set pid_process_denom = 2
> set simplified_gyro_filter_multiplier = 120
> set osd_warn_bitmask = 49151
> set osd_vbat_pos = 1517
> set osd_rssi_pos = 2498
> set osd_rssi_dbm_pos = 2466
> set osd_flymode_pos = 2581
> set osd_crosshairs_pos = 185
> set osd_ah_sbar_pos = 207
> set osd_ah_pos = 90
> set osd_current_pos = 3629
> set osd_mah_drawn_pos = 3469
> set osd_wh_drawn_pos = 267
> set osd_gps_speed_pos = 2076
> set osd_gps_lon_pos = 3080
> set osd_gps_lat_pos = 3112
> set osd_gps_sats_pos = 2062
> set osd_home_dir_pos = 2073
> set osd_home_dist_pos = 2067
> set osd_altitude_pos = 2562
> set osd_warnings_pos = 14421
> set osd_avg_cell_voltage_pos = 3533
> set osd_battery_usage_pos = 36205
> set osd_disarmed_pos = 2613
> set osd_core_temp_pos = 1128
> set osd_displayport_device = MSP
> set debug_mode = GYRO_SCALED
> set vcd_video_system = HD
> set gyro_1_align_yaw = 1800
> set rpm_filter_q = 1000
> set rpm_filter_fade_range_hz = 100
> set craft_name = XL5 V5
> set pilot_name = Sergey Tihonov
> 
> profile 0
> 
> # profile 0
> set dterm_lpf1_dyn_min_hz = 80
> set dterm_lpf1_dyn_max_hz = 110
> set dterm_lpf1_static_hz = 0
> set dterm_lpf2_static_hz = 0
> set anti_gravity_gain = 100
> set yaw_lowpass_hz = 0
> set p_pitch = 77
> set i_pitch = 117
> set d_pitch = 39
> set f_pitch = 168
> set p_roll = 64
> set i_roll = 97
> set d_roll = 35
> set f_roll = 140
> set p_yaw = 64
> set i_yaw = 97
> set f_yaw = 140
> set d_min_roll = 35
> set d_min_pitch = 39
> set motor_output_limit = 75
> set feedforward_smooth_factor = 35
> set feedforward_jitter_factor = 12
> set simplified_master_multiplier = 130
> set simplified_i_gain = 85
> set simplified_d_gain = 90
> set simplified_pi_gain = 110
> set simplified_dmax_gain = 0
> set simplified_feedforward_gain = 90
> set simplified_pitch_pi_gain = 115
> set simplified_dterm_filter = OFF
> set simplified_dterm_filter_multiplier = 140
> 
> profile 1
> 
> profile 2
> 
> profile 3
> 
> # restore original profile selection
> profile 0
> 
> rateprofile 0
> 
> # rateprofile 0
> set thr_mid = 0
> set roll_rc_rate = 2
> set pitch_rc_rate = 2
> set yaw_rc_rate = 2
> set roll_expo = 40
> set pitch_expo = 40
> set yaw_expo = 40
> set roll_srate = 80
> set pitch_srate = 80
> set yaw_srate = 60
> 
> rateprofile 1
> 
> rateprofile 2
> 
> rateprofile 3
> 
> # restore original rateprofile selection
> rateprofile 0
> 
> # save configuration
> ```

Please enable `feature TELEMETRY` this is no enabled by default but it used to be.

haslinghuis, 2023-09-19:
Can you enable telemetry on the receiver tab and save and reboot? (As saving on ports tab in confgurator 10.9 disables it - fixed in 10.10 (nightly))

tihonovS, 2023-09-19:
telemetry was already enabled on the receiver tab. But I turned it off, saved it, turned it back on, saved it. Nothing changed.
[image]

haslinghuis, 2023-09-19:
Using CRSF myself cannot confirm the telemetry issue. Found that using dynamic power profile on some builds did not get stick movements. Bought 3 more receivers to confirm. Using Radiomaster TX16S with EdgeTX 2.8 flying dutchman.
One build had issues with telemetry lost - but replacing antenna worked out.

klutvott123, 2023-09-19:
@haslinghuis can you provide a 4.4.2 build for the target in question with the tx monitoring PRs reverted?

haslinghuis, 2023-09-19:
Getting more difficult in time. A buildkey or support ID would help as I have to setup maintenance and revert some PR's.

tihonovS, 2023-09-19:
Something is broken, no UART in the ports tab
[image]

haslinghuis, 2023-09-19:
Please build 4.4.3-zulu using #13084 in select commit field.

tihonovS, 2023-09-20:
everything works. thank you) 
Is this build stable or should I wait until these fixes are available?

haslinghuis, 2023-09-20:
Almost there: https://github.com/betaflight/betaflight/pulls?q=is%3Aopen+is%3Apr+milestone%3A4.4.3

closed 2023-10-23 by haslinghuis