betaflight/betaflight #13059: No I2C BMP280 barometer support for Xilo F4 FC on 4.4.2

AdamVicknair, 2023-08-31:
### Describe the bug

BMP280 connected via I2C is not detected.

### To Reproduce

Connect BMP280 to Xilo Stax F4 flight controller (V1) via I2C. Flash BF 4.4.2 including BARO_BMP280 in the custom defines input box. Enable barometer from the configuration tab.

### Expected behavior

BMP280 barometer detected and configurable through CLI.

### Support ID

```txt
a036dcdd-421f-4340-81c3-f7eb1023cdd3
```

### Flight controller

Xilo STAX F4 (V1), bought from GetFPV as a bundle with the ESC a few years ago.

### Other components

BMP280 barometer (generic breakout board), GEPRC M100 GPS, Crossfire Nano Reciever

### How are the different components wired up (including port information)

Crossfire to UART 3, GPS to UART4 and Barometer/GPS Magnetometer to I2C bus (barometer closest to the FC)

### Add any other context about the problem that you think might be relevant here

The GPS, Magnetometer and the BMP280 Barometer were working with BF 4.2.8. Here is a dump of the settings I had configured on 4.2.8 when everything was working.

[BTFL_cli_dump_SourceOne_GPS+Baro.txt](https://github.com/betaflight/betaflight/files/12488707/BTFL_cli_dump_SourceOne_GPS%2BBaro.txt)

I tried pasting all of the baro types into the custom defines, just the one I have, and none at all. I do not get any barometer configuration options in the CLI (bustype, address, etc.) but the mag is there and recognized (also on the same I2C bus, after the baro). And of course when toggling the barometer switch on the config tab it does not remain persistent when saved.

haslinghuis, 2023-08-31:
For external baro should be able to fix with reflashing:

- Enable show release candidates
- Enable expert mode
- Add BARO BARO_BMP280 to custom defines.

AdamVicknair, 2023-09-01:
This fixed it! Thank you so much!! I guess what I did wrong was not include the first "BARO" in front of "BARO BARO_BMP280" when I added it to the custom defines.

closed 2023-09-01 by AdamVicknair