betaflight/betaflight #13060: OSD gone above Betaflight 4.3.2

Tinygo1, 2023-08-31:
### Describe the bug

I have a drone with a GEPRC TAKER F411 8Bit 12A AIO flight controller. The OSD works fine on Betaflight 4.3.2 but if i pick a version above that my OSD stops working and i cant seem to find the problem. OSD is on in settings, i connected a battary (even when never needed before), tried to use my backup from working OSD settings. I then get a few error lines shown in picture. Anyone knows a fix?
[image]

### To Reproduce

Take a GEPRC TAKER F411 8Bit 12A AIO flight controller with working settings (backup) and flash it above 4.3.2

### Expected behavior

OSD not showing in goggles anymore.

### Support ID

```txt
1a21a3f9-1614-49d1-8c29-01b9a6a43496
```

### Flight controller

GEPRC TAKER F411 8Bit 12A AIO with on board ELRS

### Other components

Geprc Rad Whoop 5.8G Vtx 32CH Pitmode

haslinghuis, 2023-08-31:
This is another example where sharing SPI no longer works.

Tinygo1, 2023-08-31:
Guess i will have to fly without OSD for now then.

nerdCopter, 2023-08-31:
> Guess i will have to fly without OSD for now then.

or keep flying 4.3.x ; nothing wrong with such.

Tinygo1, 2023-08-31:
The problem is i can't connect to the SPI than. So im choosing between no OSD or no controlls lol. I tried the bind command in CLI with version 4.3.2 but that did not work.

SteveCEvans, 2023-08-31:
Whilst sharing SPI busses isn't ideal, the only case where is simply won't work is where the bus for the SD card is shared. Such sub-optimal device assignment doesn't explain the invalid resource names.

From https://build.betaflight.com/api/support/1a21a3f9-1614-49d1-8c29-01b9a6a43496 I see that OSD is disabled. The following are set.

```
resource FLASH_CS 1 A08
resource OSD_CS 1 B12
```

The fact that resetting to defaults produces these errors suggests that FLASH and OSD support are excluded from the build.

https://build.betaflight.com/api/builds/5cdf41dd7f882ec1d70965625d16fb92/log shows that `-DCORE_BUILD` is defined.

@Tinygo1 Could you please try flashing again using an online build without using the **Core Only** switch.

Tinygo1, 2023-08-31:
5003f807-91ff-4730-b939-053f2d2236fd

Thisone is with the settings i usually run.

haslinghuis, 2023-08-31:
[support](https://build.betaflight.com/api/support/5003f807-91ff-4730-b939-053f2d2236fd)
[build](https://build.betaflight.com/api/builds/a009b9623088574c0495c57a69504f72/log)

haslinghuis, 2023-08-31:
Is it solved?

Tinygo1, 2023-08-31:
No i gave up for now. Will try to figure it out later, or just steal the kitchen timer from the lady and fly without OSD.

nerdCopter, 2023-09-01:
this unified target seems to not have any USE_FLASH defined. https://github.com/betaflight/unified-targets/blob/master/configs/default/GEPR-GEPRCF411SX1280.config , yet as Steve pointed out does indeed have a flash resource set.

OSD is enabled correct?  just asking because it's not set as a enabled by default in unified target.
[image]

Tinygo1, 2023-09-01:
@NerdCopter 

Yes i have it enabled in the settings.
[image]

[image]

And it works right away with this settings when i go back to Betaflight version 4.3.2 Also i am aware of the new (Other options) to leave stuff off when Flashing. So i don't think that is the problem neither.
[image]

Thanks for still looking into it. You guys are great.

haslinghuis, 2023-09-01:
When enabling show custom release + enable expert mode you can add `FLASH FLASH_W25Q128FV` to Custom Defines.

Should not be needed but can you also add `OSD (HD)` to Other Options

Tinygo1, 2023-09-01:
@haslinghuis

Interesting, i tried the things you advised:
[image]

It did not work yet but i noticed the five error messages i got when loading my settings the first time now turned into two.
[image]

Also when i run Core only flash version 4.4.2 it seems to work fine.

haslinghuis, 2023-09-01:
Also add `MAX7456` to custom define and it will work 😏 - as it's missing in configuration. Does blackbox work now?