betaflight/betaflight #13071: BF 4.5.0 cannot enable  external barometer BME/BMP280 i2c

hayorov, 2023-09-08:
### Describe the bug

# version
# Betaflight / STM32F745 (S745) 4.5.0 Sep  8 2023 / 03:24:34 (5cd2ab50e) MSP API: 1.46
# config rev: 09f53ae
# board: manufacturer_id: FOXE, board_name: FOXEERF745V4_AIO

CU F745 Clock=216MHz, Vref=3.29V, Core temp=50degC
Stack size: 2048, Stack address: 0x20010000
Configuration: CONFIGURED, size: 3838, max available: 32768
Devices detected: SPI:1, I2C:0
Gyros detected: gyro 1 locked dma
GYRO=MPU6000, ACC=MPU6000
OSD: MSP (53 x 20)
BUILD KEY: 17a4f7eea6690d1c569f7532630262c8 (4.5.0-zulu)
System Uptime: 429 seconds, Current Time: 2023-09-08T04:27:42.865+00:00
CPU:40%, cycle time: 125, GYRO rate: 8000, RX rate: 15, System rate: 9
Voltage: 63 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
FLASH: JEDEC ID=0x00ef4018 16M
GPS: connected, UART7 57600 (set to AUTO), configured, version = M10

### To Reproduce

# get baro
baro_bustype = I2C
Allowed values: NONE, I2C, SPI, SLAVE

baro_spi_device = 0
Allowed range: 0 - 5

baro_i2c_device = 2
Allowed range: 0 - 5
Default value: 1

baro_i2c_address = 118
Allowed range: 0 - 119
Default value: 0

baro_hardware = BMP280
Allowed values: AUTO, NONE, BMP085, MS5611, BMP280, LPS, QMP6988, BMP388, DPS310, 2SMPB_02B, VIRTUAL
Default value: AUTO

altitude_prefer_baro = 100
Allowed range: 0 - 100

### Expected behavior

barometer detection

### Support ID

```txt
fb78882f-3581-4d41-b54e-d06c895f519c
```

### Flight controller

FOXEERF745V4_AIO

### Other components

External barometer via I2c (SCl/SDA) BME/BMP280

### How are the different components wired up (including port information)

I2C SCL->SCL, SDA-> SDA

resource I2C_SCL 2 B08
resource I2C_SDA 2 B09

based on https://github.com/betaflight/config/blob/master/configs/FOXEERF745V4_AIO/config.h

#define I2C1_SCL_PIN         PB8
#define I2C1_SDA_PIN         PB9

### Add any other context about the problem that you think might be relevant here

BF build 4.5.0 no custom targets like baro or baro_bmp280 was used

haslinghuis, 2023-09-08:
Why use external baro when Foxeer F745AIO V4 comes with onboard baro?

Just finished build with Foxeer F745AIO (V1) with crossfire nano, Walksnail VTX and Walksnail GPS with external baro and baro and mag are detected on 4.5.0. However I need to enable magnetometer again after cold boot while it sticks on save and reboot.

hayorov, 2023-09-08:
>> Why use external baro when Foxeer F745AIO V4 comes with onboard baro?
https://www.foxeer.com/foxeer-reaper-aio-v4-f745-mpu6000-45a-bluejay-bls-2-6s-g-542

No, it doesn't have baro on AIO board.

haslinghuis, 2023-09-08:
You don't have to reassign the resources for I2C as the driver (define) is already applied I don't see why it doesn't work. Just to be sure - did you enable barometer on configuration tab?

hayorov, 2023-09-08:
>> Have you activated the barometer in the configuration tab?
Yes, I have.

In the meantime, I've noticed that when I set the baro_hardware to AUTO, the barometer switch in the configuration tab automatically disables itself after I apply or restart. However, when I set baro_hardware to BMP280, I can enable the barometer in the configuration tab, but the baro icon doesn't appear.

I've consistently had zero I2C errors, and I've double-checked that the battery is properly connected; for instance, when I use a multimeter, I can confirm that there's a +5V reading on the baro module.

@haslinghuis , Do you think it would be worthwhile to check the data line with an oscilloscope? (I already have one available for testing.)

hayorov, 2023-09-08:
Oki-doki, i've figure it out. Down below is the correct config for my board: FOXEERF745V4_AIO

```
# FOXEERF745V4_AIO external i2c baro
# default resources 
# B08: I2C_SCL 1
# B09: I2C_SDA 1

# get baro
baro_bustype = I2C
Allowed values: NONE, I2C, SPI, SLAVE

baro_spi_device = 0
Allowed range: 0 - 5

baro_i2c_device = 1
Allowed range: 0 - 5

baro_i2c_address = 0
Allowed range: 0 - 119

baro_hardware = AUTO
Allowed values: AUTO, NONE, BMP085, MS5611, BMP280, LPS, QMP6988, BMP388, DPS310, 2SMPB_02B, VIRTUAL

altitude_prefer_baro = 100
Allowed range: 0 - 100
```

closed 2023-09-08 by hayorov

haslinghuis, 2023-09-08:
baro_i2c_address = 0

Was going to ask to reset the address before. So that was it?

hayorov, 2023-09-09:
> baro_i2c_address = 0
> Was going to ask to reset the address before. So that was it?

You were right. It took me some time to realize that I didn't need to remap the I2C address. Thanks a lot.