betaflight/betaflight #13076: Servo limits cannot be extended in betaflight.

Nilssonfpv, 2023-09-15:
### Describe the bug

Setting servo endpoints to 800/2200 still only gives 1000-2000 as servo output.

Rate cannot be set above 100% through the configurator or the cli.

### To Reproduce

No option to set rate above 100.

### Expected behavior

The ability to set rate above 100% so extended limits on a servo can be used.

### Support ID

```txt
-----------
```

### Flight controller

Speedbee F7 V3

sartor, 2023-10-17:
I have same issue. It just change anything

haslinghuis, 2023-11-25:
[image]

sartor, 2023-11-27:
If I understood right this picture I need to do both settings:
1) Extend limit in Betaflight for particular servo
2) Extend range in EdgeTX transmitter for this aux channel

Then it will work?
I've tried to do both of this, but not together

Prajwal-Koirala, 2023-12-29:
This is an issue about extending servo limits in Betaflight. It discusses that setting servo endpoints to 800/2200 still only gives 1000-2000 as servo output. The rate cannot be set above 100% through the configurator or the CLI. So the user is asking for the ability to set the rate above 100% so extended limits on a servo can be used. Others have chimed in that they have the same issue. The issue was closed as inactive but has been reopened.

I hope this is helpful!

haslinghuis, 2024-03-14:
Fixed in #13451

closed 2024-03-14 by haslinghuis