betaflight/betaflight #13077: Seperate motor output limit for turtle mode

hyp0dermik-code, 2023-09-16:
### Is your feature request related to a problem? Please describe

Throttle response in turtle mode sensitive.

Big jump from motor startup to normal output limit potentially no good in a semi-stalled situation

### Describe the solution you'd like

Separate motor output limit for turtle mode could help in this case

### Describe alternatives you've considered

LUA script in transmitter

Dcvelopement, 2023-09-20:
You could easily limit max throttle based on switch conditions (in this case turtle mode on switch) within EdgeTX/ OpenTX. I have a custom curve setup for turtle mode that provides expo, and could define max throttle. Joshua Bardwell just did a video on this: https://www.youtube.com/watch?v=22ueY-bvGnI

nerdCopter, 2023-10-05:
is this not already `set crashflip_power_percent = ` ?

closed 2023-10-05 by haslinghuis