betaflight/betaflight #13079: Unable to safely hand over from Switched GPS Resue to Manual Flight

bn880, 2023-09-17:
### Describe the bug

Once Switch triggered GPS rescue initiates , the only way to try and take over control from it mid flight is to disable GPS rescue via the switch, disarm and then try to re-arm mid air which will likely fail sanity checks, and the drone will drop somewhere.

Disabling switch initiated GPS rescue should be as easy as turning off the rescue switch, for safety reasons.  (Technically speaking, this should ALSO be possible for regular GPS rescue fail safe when signal was lost, if someone assigns a switch to GPS rescue and toggles it on and off DURING GPS rescue)

### To Reproduce

Configure all parameters of GPS rescue in betaflight as usual
Assign a controller switch to trigger GPS rescue mid flight
Take off in a manner that satisfies GPS rescue (sufficient satellites)
Fly out and initiate GPS rescue with the assigned switch
Ensure GPS rescue is working and flying back
Disable GPS rescue with the switch (notice drone will drop instead of handing back over to the controller)

### Expected behavior

Hand over control to the radio when the GPS rescue switch is switched off while the radio remains armed.  (Technically speaking, this should ALSO be possible for regular GPS rescue fail safe when signal was lost, if someone assigns a switch to GPS rescue and toggles it on and off DURING GPS rescue)

### Support ID

```txt
067262e3-839d-4eea-a43d-c49a1b56b53d
```

### Flight controller

Speedybee F405-V3

aowi7280, 2023-09-20:
move your sticks to regain control

bn880, 2023-09-20:
That didn't work, wouldn't have reported.

haslinghuis, 2023-09-20:
Needs at least 30% movement of the sticks

bn880, 2023-09-20:
Perhaps that's it, I will re-test next time I head out and update here.

bn880, 2023-09-27:
It works indeed, disable the rescue switch and then move both sticks at least 30%.  Handover is good!  Thank you.

bn880, 2023-09-27:
Handover works as intended:  disable the rescue (RTH) switch while remaining armed, and then move both sticks at least 30%.  Handover to ACRO mode without issue.

closed 2023-09-27 by bn880