betaflight/betaflight #13086: Uncorrect beeper behavior with BLITZ MINI F7 V1.1

ikherty, 2023-09-22:
### Describe the bug

I have Iflight Evoque F5D V2 O3 TBS that comes with Betaflight version 4.3.2 and it works fine on that firmware. I flashed 4.4.2 version with stock build configuration and after that the beeper started clicking constantly. I manually configured 4.4.2 and the problem didn't go away. I reset the settings and put in the diff from the last version, the behavior is similar. Also beeper does not react to changing the value in beeper mode.
I noticed that the resources are different with only one line: in 4.3.2 was "resource PPM 1 A03", 4.4.2 does not.
Dump of this FC after flashing 4.4.2 with full chip erase and apply defaults:
[stock dump 442.txt](https://github.com/betaflight/betaflight/files/12696828/stock.dump.442.txt)
Dump of this FC before flashing:
[dump from box 432.txt](https://github.com/betaflight/betaflight/files/12696865/dump.from.box.432.txt)
I flashed 4.4.2 with "Radio Protocol" = PPM option and the problem went away. But I need CRSF protocol for my radio :D

target: IFLIGHT_BLITZ_F722
settings out of the box:
[stock diff evoque o3 f5d v2.txt](https://github.com/betaflight/betaflight/files/12696789/stock.diff.evoque.o3.f5d.v2.txt)

### To Reproduce

Flash stosk Iflight Evoque F5D V2 O3 TBS to betaflight 4.4.2 with stock build configuration. Apply custom settings

### Expected behavior

Correct beeper work

### Support ID

```txt
Id: "3472555a-41ba-4cb4-81f0-b07ae6f3cc5b"
```

### Flight controller

BLITZ MINI F7 V1.1

ikherty, 2023-09-22:
sorry, it was my mistake, don't bug

closed 2023-09-22 by ikherty

athchris, 2024-09-19:
Hello, I have the same issue, what was the problem in your case?