betaflight/betaflight #13087: Corrupted blackbox logs via serial/OpenLager going from BF 4.3.2 to 4.4.2 on OMNIBUSF4SD target

krisdover, 2023-09-22:
### Describe the bug

Hi team, I just updated the first quad of my fleet from v4.3.2 to v4.4.2 and I'm seeing corrupted data in the blackbox logs on the new version:

# on Betaflight V4.4.2
```
 w9  % Á9  ÿP €0 †±ó¢Çßßšt ¾g{èõ}Ýpüuÿµy]ÝõýeýýX‡ö×ÍÍŸõŸ½ÿÿ½÷õÿýÑûIH Product:Blackbox flight data recorder by Nicholas Sherlock
H Data version:2
H Field I name:loopIteration,time,axisP[0],axisP[1],axisP[2],axisI[0],axisI[1],axisI[2],axisD[0],axisD[1],axisF[0],axisF[1],axisF[2],rcCommand[0],rcCommand[1],rcCommand[2],rcCommand[3],setpoint[0],setpoint[1],setpoint[2],setpoint[3],vbatLatest,amperageLatest,rssi,gyroADC[0],gyroADC[1],gyroADC[2],accSmooth[0],accSmooth[1],accSmooth[2],debug[0],debug[1],debug[2],debug[3],motor[0],motor[1],motor[2],motor[3]
)bŠbŠb‚bŠbŠbŠbŠb‚bŠb‚bŠbŠbŠbŠbŠbŠbŠbŠbŠbŠb‚b‚b‚b‚RH Field I predictor:0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,9,0,0,0,0,0,0,0,0,0,0,0,0,11,5,5,5
H Field I encoding:1,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,1,0,0,0,0,3,0,1,0,0,0,0,0,0,0,0,0,0,1,0,0,0
H Field P predictor:6,2,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,3,3,3,3,3,3,3,3,3,3,3,3,3,3
H Field P encoding:9,0,0,0,0,7,7,7,0,0,0,0,0,8,8,8,8,8,8,8,8,6,6,6,0,0,0,0,0,0,0,0,0,0,0,0,0,0
H Field S name:flightModeFlags,stateFlags,failsafePhase,rxSignalReceived,rxFlightChannelsValid
H Field S signed:0,0,0,0,0
H Field S predictor:0,0,0,0,0
H Field S encoding:1,1,7,7,7
H Firmware type:Cleanflight
H Firmware revision:Betaflight 4.4.2 (23d066d08) STM32F405
H Firmware date:Jun  1 2023 02:07:34
H Board information:AIRB OMNIBUSF4SD
H Log start datetime:0000-01-01T00:00:00.000+00:00
H Craft name:Chameleon
H I interval:128
H P interval:2
H P ratio:64
H minthrottle:1070
H maxthrottle:2000
H gyro_scale:0x3f800000
H motorOuÁÕÑé48,2047
H acc_1G:2048
H vbat_scale:110
H vbatcellvoltage:330,350,430
H vbatref:2012
H currentSensor:0,800
H looptime:125
H gyro_sync_denom:1
H pid_process_denom:2
H thr_mid:50
H thr_expo:0
H tpa_mode:1
H tpa_rate:65
H tpa_breakpoint:1350
H rc_rates:4,4,4
H rc_expo:0,0,0
H rates:9H rate_limits:1998,1998,1998
H rollPID:63,0,36
H pitchPID:65,0,41
H yawPID:63,0,0
H levelPID:50,50,75
H magPID:40
H d_min:27,30,0
H d_max_gain:37
H d_max_advance:20
H dterm_lpf1_type:0
H dterm_lpf1_static_hz:0
H dterm_lpf1_dyn_hz:75,150
H dterm_lpf2_type:0
Hødrlüf›Ñ…Ñ¥}¡éé150
H yaw_lowpass_hz:100
H dterm_notH dterm_notch_cutoff:0
H iterm_windup:85
H iterm_relax:1
H iterm_relax_type:1
H iterm_relax_cutoff:15
H pid_at_min_throttle:1
H anti_gravity_gain:80
H anti_gravity_cutoff_hz:5
H anti_gravity_p_gain:100
H abs_control_gain:0
[... 409 lines cut ...]
set pitch_srate = 91
set yaw_srate = 94

# end the command batch
batch end

#
```

### Flight controller

Omnibus F4 AIO OSD (from Banggood)

### Other components

- OpenLager
- Jumper Aion 2.4GHz Mini ELRS
- Eachine TX805 VTX

Board Image:
[image]

### How are the different components wired up (including port information)

I'm running the Jumper Aion 2.4GHz Mini ELRS receiver on UART3 (TX3/RX3) and the Smart Audio from my Eachine TX805 VTX is wired to UART6(TX6). The OpenLager is on UART1 (TX1).

haslinghuis, 2023-09-22:
For testing please try flashing development build:
1. 4.4.3-zulu 
2. 4.4.3-zulu using https://github.com/betaflight/betaflight/pull/13084 in select commit field.

isilcala, 2023-09-28:
> For testing please try flashing development build:
> 
> 1. 4.4.3-zulu
> 2. 4.4.3-zulu using [Remove uart mon #13084](https://github.com/betaflight/betaflight/pull/13084) in select commit field.

I'm facing the same issue. Tested:

1. 4.4.3-zulu --- didn't fix
2. 4.4.3-zulu using https://github.com/betaflight/betaflight/pull/13084 in select commit field. --- I couldn't find this commit.

haslinghuis, 2023-09-28:
You have to find the commit. Just add #13084 in commit field when flashing 4.4.3 zulu

isilcala, 2023-09-29:
> You have to find the commit. Just add #13084 in commit field when flashing 4.4.3 zulu

Thanks. 

So I put literally '#13084' in the commit field and flashed 4.4.3 zulu, but still no luck.

closed 2023-11-07 by github-actions[bot]

Zagitta, 2023-11-09:
I can confirm `4.4.3-zulu` fixed this corruption issue for me

sunjialinsun, 2024-03-31:
> I can confirm `4.4.3-zulu` fixed this corruption issue for me
Thanks, I am facing the same issue on 4.4.2. let me upgrade to 4.4.3 to test it.