betaflight/betaflight #13089: Betaflight 4.3-4.4 gyro fix

Serge574, 2023-09-23:
### Describe the bug

[image]
[image]
Hello dear friends, my copter pulls to the side. After calibrating the gyrocop or rebooting the power, this temporarily solves the problem. When GPS Rescue is turned on, the copter pulls to the side while gaining altitude, but when it rises to a given height it levels out and makes a smooth return home.

### To Reproduce

This happened shortly after Betaflight 4.3 firmware, the exact release is unknown, Betaflight 4.4.2 firmware did not solve the problem.

### Expected behavior

solutions to similar problems on Github did not help me solve this problem

### Support ID

```txt
version
# Betaflight / STM32F405 (S405) 4.4.2 Jun  1 2023 / 18:48:17 (23d066d08) MSP API: 1.45

# config: YES
# board: manufacturer_id: HBRO, board_name: KAKUTEF4V2

# status
MCU F40X Clock=168MHz (PLLP-HSE), Vref=3.30V, Core temp=43degC
Stack size: 2048, Stack address: 0x1000fff0
Configuration: CONFIGURED, size: 4167, max available: 16384
Devices detected: SPI:1, I2C:2
Gyros detected: gyro 1 locked
GYRO=ICM20689, ACC=ICM20689, BARO=BMP280, MAG=QMC5883
OSD: MAX7456 (30 x 13)
BUILD KEY: 77170922c9f94cdc1307ab6af4eb8d4f (4.4.2)
System Uptime: 3491 seconds, Current Time: 2023-09-23T09:33:35.174+00:00
CPU:53%, cycle time: 128, GYRO rate: 7812, RX rate: 15, System rate: 9
Voltage: 1634 * 0.01V (4S battery - OK)
I2C Errors: 0
SD card: Not configured
FLASH: JEDEC ID=0x00ef4018 16M
Arming disable flags: RXLOSS CLI MSP GPS RESCUE_SW

# flash_info
Flash sectors=256, sectorSize=65536, pagesPerSector=256, pageSize=256, totalSize=16777216 JEDEC ID=0x00ef4018
Partitions:
  0: FLASHFS   0 255
FlashFS size=16777216, usedSize=0

# dump hardware

# version
# Betaflight / STM32F405 (S405) 4.4.2 Jun  1 2023 / 18:48:17 (23d066d08) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name KAKUTEF4V2
manufacturer_id HBRO

# resources
resource BEEPER 1 C09
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 A03
resource MOTOR 4 A02
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
[... 342 lines cut ...]
07 - (             RX)     10      27      14    0.0%    0.0%       520      8  33597      12
08 - (         SERIAL)     99  613300       1 6071.6%    0.0%      1356      0 109779      22
09 - (       DISPATCH)    979       6       0    0.5%    0.0%       662     18 1096620       2
10 - (BATTERY_VOLTAGE)     50       6       1    0.0%    0.0%        79      2  55914       3
11 - (BATTERY_CURRENT)     50       7       1    0.0%    0.0%        84      2  55914       2
12 - ( BATTERY_ALERTS)      5       5       1    0.0%    0.0%         9      0   5604       2
13 - (         BEEPER)     99       8       2    0.0%    0.0%       263      0 109778       5
14 - (            GPS)    125      88      13    1.1%    0.1%      2117     20 144639      37
15 - (     GPS_RESCUE)    100      10       3    0.1%    0.0%       422      9 111470       4
16 - (        COMPASS)     10       9       2    0.0%    0.0%        92      0  33000       5
17 - (           BARO)     40      15       6    0.0%    0.0%       837      3 134383       7
19 - (       ALTITUDE)     99       7       1    0.0%    0.0%       157      8 109778       3
21 - (      TELEMETRY)    489      42       0    2.0%    0.0%      3019    132 547415       1
24 - (            OSD)     12     184      20    0.2%    0.0%      7542    734 430452      28
26 - (            CMS)     20       7       1    0.0%    0.0%        30     13  22387       3
27 - (        VTXCTRL)      5      58       0    0.0%    0.0%        49      2   5597      12
29 - (    ADCINTERNAL)      1       5       2    0.0%    0.0%         2      0   1122       3
31 - (SPEED_NEGOTIATION)     99       6       0    0.0%    0.0%        95    129 109778       3
RX Check Function                  12       5                      2303
Total (excluding SERIAL)                                51.9%
```

### Flight controller

KAKUTEF4V2(STM32F405)

closed 2023-10-31 by github-actions[bot]

haslinghuis, 2023-10-31:
Did you try 4.4.3-zulu or 4.5-zulu development build?

Serge574, 2023-11-10:
> Пробовали ли вы сборку разработки 4.4.3-zulu или 4.5-zulu?

> Пробовали ли вы сборку разработки 4.4.3-zulu или 4.5-zulu?

Serge574, 2023-11-10:
> Did you try 4.4.3-zulu or 4.5-zulu development build?

didn't try, I'm using development 4.4.2